TrunkRecorder/trunk-recorder #477: (request) Log Standardization

kcwebby, 2021-05-19:
Is it possible to establish a common format log file, so that the same data can be recorded, but that log-processing can be done (splunk/elasticsearch) to generate statistics and alerts based on log events?

Seems like there are several "types" of log messages, and none of them conform to a similar type of format.

Perhaps something like "timestamp" "log_level" "action (Starting|Stopping|Retuning|Decode Error|etc" "shortname" "tgid" "tg_name" "message"

leee, 2021-05-20:
I would love to see this happen and hope someone takes this issue up, and would like to note a few other issues (some of which you made too!) closely relate to this topic.

Would you be willing to close the following out and merge them into this issue?

- #106 
- #310 

I also think this topic is related to call statistics analysis discussion, #1 and #462.

robotastic, 2021-05-20:
I would be all for this if someone wants to tackle it. If someone can gather group consensus around what information should be logged and what events should generate logs, I will go through and make the changes.

Someone has built an interface for sending things to Splunk using the Status Server:
https://github.com/robotastic/trunk-recorder/issues/462#issuecomment-835384259

Ultimately, this would be something that would make a good plugin. I going to be focusing on that next.

kcwebby, 2021-05-24:
I'm not sure how to merge issues:  however this is related to:
#106 and #310 
may also related to call statistics analysis discussion, #1 and #462.

closed 2021-05-24 by kcwebby