TrunkRecorder/trunk-recorder #487: Log file format cleanup

jcummingsg2, 2021-06-04:
Creating issue to look at updating logging code to standardize format output to result in cleaner log file.  Tab formatting different due to TG number versus TG name returned by different logging code.

Example log output at info level:
`
[2021-06-04 08:46:09.131017]: [kcers]   TG: 3248        Freq: 856.687500 MHz    Broadcastify Upload Success - file size: 53460
[2021-06-04 08:46:10.702317]: [kcers]   TG:             Sea WtrShd1 (      6448)        Freq: 854.087500 MHz    Update Retuning - New Freq: 854.087500 MHz      Elapsed: 16s    Since update: 2s
[2021-06-04 08:46:12.253403]: [kcers]   TG:                  FTAC 2 (     22192)        Freq: 852.287500 MHz    Starting Recorder on Src: hackrf=26b468dc2e7d848f
[2021-06-04 08:46:13.502272]: [kcers]   TG:                   SPD E (      3344)        Freq: 856.762500 MHz    Starting Recorder on Src: hackrf=26b468dc2e7d848f
[2021-06-04 08:46:19.010710]: [kcers]   TG:                SFD Disp (      1744)        Freq: 851.962500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 41
[2021-06-04 08:46:20.010893]: [kcers]   TG:                  FTAC 1 (     22224)        Freq: 852.412500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 15
[2021-06-04 08:46:20.605997]: [kcers]   TG: 1744        Freq: 851.962500 MHz    Broadcastify Upload Success - file size: 450564
[2021-06-04 08:46:21.317065]: [kcers]   TG: 22224       Freq: 852.412500 MHz    Broadcastify Upload Success - file size: 137672
[2021-06-04 08:46:22.010113]: [kcers]   TG:              BotPD Disp (     20208)        Freq: 852.212500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 30
[2021-06-04 08:46:22.028135]: [kcers]   TG:                  KCSO N (     23152)        Freq: 851.162500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 26
[2021-06-04 08:46:22.181481]: [kcers]   TG:                  FTAC 1 (     22224)        Freq: 851.637500 MHz    Starting Recorder on Src: hackrf=26b468dc2e7d848f
[2021-06-04 08:46:23.171086]: [kcers]   TG: 23152       Freq: 851.162500 MHz    Broadcastify Upload Success - file size: 57324
`

Log created by call.cc
`
    BOOST_LOG_TRIVIAL(info) << "[" << sys->get_short_name() << "]\tTG: " << this->get_talkgroup_display() << "\tFreq: " << FormatFreq(get_freq()) << "\tEnding Recorded Call - Last Update: " << this->since_last_update() << "s\tCall Elapsed: " << this->elapsed();

[2021-06-03 13:30:19.012294]: [kcers]   TG:              BotPD Disp (     20208)        Freq: 852.162500 MHz    Ending Recorded Call - Last Update: 6s  Call Elapsed: 9
`

Log created by /uploaders/broadcastify_uploader.cc
`
    BOOST_LOG_TRIVIAL(info) << "[" << call->short_name << "]\tTG: " << call->talkgroup << "\tFreq: " << FormatFreq(call->freq) << "\tBroadcastify Upload Success - file size: " << file_info.st_size;

[2021-06-03 13:30:19.401844]: [kcers]   TG: 23152       Freq: 856.762500 MHz    Broadcastify Upload Success - file size: 100588
`

leee, 2021-06-04:
I think this issue is related to #106 and #310. Lets close this one and put this discussion into one of those.

leee, 2021-07-17:
@jcummingsg2  (or @robotastic ) this issue can be closed.

robotastic, 2021-07-20:
PS - I have tried to clean up the log formats in the v4.0-beta branch.

closed 2021-07-20 by robotastic