TrunkRecorder/trunk-recorder #489: Allow user to specify gain names.

jnse, 2021-06-16:
My USRP has gain levels not supported by the hardcoded gain names in trunk-recorder (BBG and GC1).
It would be preferable if the gain name could be user-specified.

robotastic, 2021-06-29:
Happy to check it out - what is the name of the Gain Stage for the USRP? I will have to check with the USRP code to see how the different types of gain are set... it has been a while since I messed with it.

jnse, 2021-07-08:
Mine has "GC1" and "BBG" as gains, but my point was you should be able to pass arbitrary names like the op25 rx.py lets you do via the -N argument (eg: ./rx.py -N 'GC1:48,BBG:3' ...) - so the op25 rx.py implementation might be useful as a reference.

closed 2024-12-14 by robotastic