TrunkRecorder/trunk-recorder #493: V4 make error

rpdale, 2021-07-14:
I git‘ed v4 into a new directory on my RPI4 and almost made it all the way :) 

Was I supposed to install anything ahead of running cmake/make that is different from v3?

[ 95%] Linking CXX executable recorder
/usr/bin/ld: CMakeFiles/recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o: undefined reference to symbol 'dlopen@@GLIBC_2.4'
/usr/bin/ld: //lib/arm-linux-gnueabihf/libdl.so.2: error adding symbols: DSO missing from command line

leee, 2021-07-17:
I'm going to add onto this thread so all V4 beta related messages can stay in one place, but if that's not desired I can break this out to a new issue.

I'm just blindly pasting my error here, I haven't started reading through the fantastic plugin work yet:

```
(info)   Setting up plugin -  Name: openmhz_uploader	 Library file: ./libopenmhz_uploader.so
(error)   Failed parsing Config: boost::dll::shared_library::load() failed (dlerror system message: ./libopenmhz_uploader.so: cannot open shared object file: No such file or directory): Bad file descriptor
```

leee, 2021-07-17:
I see, it looks like the shared objects do exist after build, and I'm able to use them successfully when they're in my working directory. @robotastic how would you want to handle users that run trunk-recorder in a different working directory than the build directory, and those that call trunk-recorder from their PATH, or directly from `/usr/local/bin/` or even the build directory?

kb2ear, 2021-07-17:
Easy enough to fix, just create a conf file in /etc/ld.so.conf.d/ with the folder that contains the .so files listed in it and run ldconfig. example: 
bash -c "echo '/usr/local/trunk-libs' > /etc/ld.so.conf.d/trunk-recorder.conf"
ldconfig

kb2ear, 2021-07-17:
> I git‘ed v4 into a new directory on my RPI4 and almost made it all the way :)
>
> Was I supposed to install anything ahead of running cmake/make that is different from v3?
>
> [ 95%] Linking CXX [...]

I believe this was fixed, do a git pull and re-run make

leee, 2021-07-17:
@kb2ear right now `libopenmhz_uploader.so` and `libbroadcastify_uploader.so` get called relative to `./`:

https://github.com/robotastic/trunk-recorder/blob/4.0-beta/trunk-recorder/main.cc#L549-L550

I'm not thoroughly read in yet, but I'm under the impression that boost:dll:import_alias that `plugin_manager/plugin_manager/setup_plugin` uses loads shared objects (or directories) by path, and has no awareness of the ld cache.

To expand upon what I was trying to say:

Some users run right in the build directory, which needs no further intervention.

Some users run in other working directories and call `build/recorder` by full path. How should they be handled? Copy over all objects to their working directory?

Some users use the install script, so should that be extended to also handle shared objects? Since `do_install()` copies to `/usr/local/bin`, maybe assume use of `/usr/local/bin/trunk-recorder/` which implies:

- copying over built libraries to that location,
- and setting plugin paths (both internal and in config) there?

So far a few libraries exist at the root of the build directory:

```
libbroadcastify_uploader.so
libgnuradio-op25_repeater.so
libopenmhz_uploader.so
libunit_script.so
```

so how do these special cases get handled?

rpdale, 2021-07-18:
I deleted the directory, pulled again, and same error.

robotastic, 2021-07-18:
@rpdale What OS are you using on the Pi? I haven't done a raw build on a Pi yet, but I think kb2ear found a fix for this that got merged in. https://github.com/robotastic/trunk-recorder/commit/b9e86de8a9eddf3f8f1d0f943e7907973fdafb18#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aR260

I wonder if it could have something to do with cmake versions?

robotastic, 2021-07-18:
@leee @kb2ear yea... this area needs some work. The plugins get compiled into shared libraries. Right now, it is hard coded that the uploader plugins are in the same directory as the recorder executable. 

I will go upgrade the install.sh script so that the plugin files get moved along with recorder.

I will also check to see how to make boost:dll:import_alias more flexible. Maybe there is a way to get the path of where the executable is running and use that instead of './'

I am sure there is someway to get cmake to handle the install and manage where the libraries go. I am not sure what the best practices are here.

robotastic, 2021-07-18:
It looks like some of the load_mode flags let you specify to also search the default library folders: https://www.boost.org/doc/libs/1_65_0/doc/html/boost/dll/load_mode/type.html

This could be an option if we have cmake install to the executable to /usr/local/bin and the plugins to /usr/local/lib

robotastic, 2021-07-18:
According to this highly opinionated doc, the proper place to store the plugin .so files would be: `/usr/local/lib/trunk-recorder/plugins`

https://tldp.org/HOWTO/HighQuality-Apps-HOWTO/plugin.html

rpdale, 2021-07-18:
Buster From: Luke BerndtSent: Sunday, July 18, 2021 7:14 AMTo: robotastic/trunk-recorderCc: rpdale; MentionSubject: Re: [robotastic/trunk-recorder] V4 make error ***@***.*** What OS are you using on the Pi? I haven't done a raw build on a Pi yet, but I think kb2ear found a fix for this that got merged in. b9e86de#diff-1e7de1ae2d059d21e1dd75d5812d5a34b0222cef273b7c3a2af62eb747f9d20aR260I wonder if it could have something to do with cmake versions?

robotastic, 2021-07-28:
OK! The 4.0-beta now supports doing a `make install`. This moves the plugins & op25 lib over to /usr/local/lib/trunk-recorder and **recorder** over to /usr/local/bin. You can now run **recorder** from any directory after you do the install. 

Big caveat - I have only tested on MacOS... I will give this a try on Linux tomorrow.

leee, 2021-07-28:
My caveats: I didn't use the `install.sh`/`rebuild.sh` scripts, I have the steps separated out as individual tasks and run an install play as its own playbook.

But the installation process works and things run just fine:
```
$ ansible-console -i ./inventory.yaml -l leee_kjfk11
Welcome to the ansible console.
Type help or ? to list commands.

leee@all (3)[f:5]$ become yes
leee@all (3)[f:5]# systemctl status recorder
kjfk11 r2-18602421ff94 (482-484 MHz) | CHANGED | rc=0 >>
● recorder.service - trunk-recorder
     Loaded: loaded (/etc/systemd/system/recorder.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-07-28 01:17:35 EDT; 2min 2s ago
   Main PID: 3188860 (recorder)
      Tasks: 1989 (limit: 9297)
     Memory: 327.6M
     CGroup: /system.slice/recorder.service
             └─3188860 /usr/local/bin/recorder

Jul 28 01:17:35 r2-18602421ff94 systemd[1]: Started trunk-recorder.
kjfk11 r2-040e3c18dae1 (476-478 MHz) | CHANGED | rc=0 >>
● recorder.service - trunk-recorder
     Loaded: loaded (/etc/systemd/system/recorder.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-07-28 01:06:50 EDT; 12min ago
   Main PID: 4187208 (recorder)
      Tasks: 626 (limit: 9296)
     Memory: 447.7M
     CGroup: /system.slice/recorder.service
             └─4187208 /usr/local/bin/recorder

Jul 28 01:06:50 r2-040e3c18dae1 systemd[1]: Started trunk-recorder.
kjfk11 r2-ace2d301a4d1 (470-472 MHz) | CHANGED | rc=0 >>
● recorder.service - trunk-recorder
     Loaded: loaded (/etc/systemd/system/recorder.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-07-28 00:58:53 EDT; 20min ago
   Main PID: 1285155 (recorder)
      Tasks: 479 (limit: 9296)
     Memory: 254.1M
     CGroup: /system.slice/recorder.service
             └─1285155 /usr/local/bin/recorder

Jul 28 00:58:53 r2-ace2d301a4d1 systemd[1]: Started trunk-recorder.
```

```
leee@all (3)[f:5]# systemctl status recorder
kjfk11 r2-18602421ff94 (482-484 MHz) | FAILED | rc=3 >>
● recorder.service - trunk-recorder
     Loaded: loaded (/etc/systemd/system/recorder.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Wed 2021-07-28 01:40:05 EDT; 27s ago
    Process: 3210491 ExecStart=/usr/local/bin/recorder (code=killed, signal=SEGV)
   Main PID: 3210491 (code=killed, signal=SEGV)

Jul 28 01:40:05 r2-18602421ff94 systemd[1]: recorder.service: Scheduled restart job, restart counter is at 8.
Jul 28 01:40:05 r2-18602421ff94 systemd[1]: Stopped trunk-recorder.
Jul 28 01:40:05 r2-18602421ff94 systemd[1]: recorder.service: Start request repeated too quickly.
Jul 28 01:40:05 r2-18602421ff94 systemd[1]: recorder.service: Failed with result 'signal'.
Jul 28 01:40:05 r2-18602421ff94 systemd[1]: Failed to start trunk-recorder.non-zero return code
```

Although looks like there's something that causes trunk-recorder to segv forcing systemd to restart as evidenced by the different active times, and the logs aren't helpful, other than some runs ending with `(error)   stop_recording() - stopping but recorder state is: 1` near the end of the logs.

Looks like it's only happening on a trunk-recorder instance with trunking systems, which might be a clue.

rpdale, 2021-07-31:
Using beta #3 I don't see any issues in cmake, but make finishes up with:

[ 95%] Linking CXX executable recorder

/usr/bin/ld: CMakeFiles/recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o: undefined reference to symbol 'dlopen@@GLIBC_2.4'

/usr/bin/ld: //lib/arm-linux-gnueabihf/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

make[2]: *** [CMakeFiles/recorder.dir/build.make:682: recorder] Error 1

make[1]: *** [CMakeFiles/Makefile2:222: CMakeFiles/recorder.dir/all] Error 2

make: *** [Makefile:130: all] Error 2

robotastic, 2021-07-31:
Thanks @rpdale - what OS / architecture are you using? It looks like it might be a Pi.

rpdale, 2021-07-31:
Correct - still Buster.

robotastic, 2021-08-01:
@rpdale I was just able to successfully compile on Buster on a Pi 4. Try doing a clean clone of the Trunk Recorder repo and then doing `git checkout -b origin/4.0-beta` 

I got a similar error as yours when I just did a `git checkout 4.0-beta`... I am not sure why though.

robotastic, 2021-09-15:
Please reopen if this is still an issue

closed 2021-09-15 by robotastic