TrunkRecorder/trunk-recorder #496: Where Should I Have the Squelch?

Dewey3, 2021-07-24:
Greetings all!  In the configuration file examples, I see the squelch setting in the [Sources] section, but in the explanations of the various settings for the configuration file, I see the squelch in the [Systems] section.  I have been placing the squelch (-60) in the [Systems] section.  A rebanded smartnet fsk4 system I monitor routinely catches 10-20 second long digital tail noice at the end of the analog transmissions.  Would moving the squelch to the [Sources] section help cut the squelch tail noise?

hayden-t, 2021-09-07:
squelch should be in systems, but only affects conventional/conventionalp25 systems, not sure if thats u or not. check your level settings are in systems too. you may have it wrong, as i found the level setting very powerful

canuckcam, 2023-03-28:
Has there been a solution to this? I'm running into the same problem. I'm trying to monitor a Smartnet system with mixed P25 and analog traffic. The P25 stuff is fine, but analog has a long tail. Does TR decode the Motorola end code? 

I've tried the Squelch setting in both sources and system. 

Attached is a recording. Thanks!

[36096-1680018189_142110000-call_7.zip](https://github.com/robotastic/trunk-recorder/files/11091344/36096-1680018189_142110000-call_7.zip)

Dewey3, 2023-03-28:
> Has there been a solution to this? I'm running into the same problem. I'm trying to monitor a Smartnet system with mixed P25 and analog traffic. The P25 stuff is fine, but analog has a long tail. [...]

Correct, the squelch settings go in "Systems" and the closer you go to zero, the tighter the squelch gets.  That was my problem, I was picturing zero like the the squelch on a radio or scanner, all the way open.  If you're getting long squelch tails, I would suggest that you experiment by reducing the negative number one or two from the current setting towards zero and listening to the results.  I find the best squelch for the mixed system that I monitor to be -60.

closed 2024-12-15 by robotastic