TrunkRecorder/trunk-recorder #497: v4.0-beta - Segmentation fault - Smartnet

jliedtke6014, 2021-07-26:
Not sure what you need, but I have two trunk-recorders running in 2 locations monitoring the same smartnet system.   Since installing 4.0 a few hours ago it has happened twice on each instance.

What can I do to help you debug this?

robotastic, 2021-07-26:
oh interesting! my Smartnet system has been pretty stable on it... but I do have it auto-reboot to switch the control channel between SDRs.

Are there any analog recorders on it? I haven't tested that out much.

if you could run `ulimit -c unlimited` this will save the core. 

You can then open it by doing `gdb recorder core`
that should start up the debugger. After that, if you could get me the output from running `bt` that would be awesome. it should help debug where the program is crashing.

jliedtke6014, 2021-07-26:
Do I do this now? Or after it crashes next time?

robotastic, 2021-07-26:
Sorry - can you run the ulimit command before starting it up and then run the gdb command after it crashes?

jliedtke6014, 2021-07-26:
Hello,

So it crashed overnight, ran the command this morning and only got this.

***@***.***:/home/administrator/trunk-build# gdb recorder core
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from recorder...
(No debugging symbols found in recorder)
/home/administrator/trunk-build/core: No such file or directory.
(gdb)

So I ran ulimit -c unlimited  again before restarting both instances and will try again if they crash.

Any other info on that gdb command I should know about?  I am not a programmer so debugging is not something I really know.

jliedtke6014, 2021-07-26:
(gdb) bt
No stack.
(gdb)
Was the rest of that paste I missed.

jliedtke6014, 2021-07-26:
Here is the output. Crashed 15mins after I restarted it.

Core was generated by `./recorder'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f7fa538ee24 in feof () from /lib/x86_64-linux-gnu/libc.so.6
[Current thread is 1 (Thread 0x7f7eedf83700 (LWP 23500))]
(gdb) bt
#0  0x00007f7fa538ee24 in feof () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x0000557fce655651 in gr::blocks::nonstop_wavfile_sink_impl::dowork(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#2  0x0000557fce6534f1 in gr::blocks::nonstop_wavfile_sink_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#3  0x00007f7fa6e2284b in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#4  0x00007f7fa6ddac03 in gr::block_executor::run_one_iteration() () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#5  0x00007f7fa6e2e59a in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#6  0x00007f7fa6e1e818 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#7  0x00007f7fa6e3d4e8 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#8  0x00007f7fa70f343b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#9  0x00007f7fa588e609 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#10 0x00007f7fa5423293 in clone () from /lib/x86_64-linux-gnu/libc.so.6

jliedtke6014, 2021-07-26:
Another one,  seems to be when there is a lot of traffic.   

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
--Type <RET> for more, q to quit, c to continue without paging--
Core was generated by `./recorder'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f7fa538ee24 in feof () from /lib/x86_64-linux-gnu/libc.so.6
[Current thread is 1 (Thread 0x7f7eedf83700 (LWP 23500))]
(gdb) bt
#0  0x00007f7fa538ee24 in feof () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x0000557fce655651 in gr::blocks::nonstop_wavfile_sink_impl::dowork(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#2  0x0000557fce6534f1 in gr::blocks::nonstop_wavfile_sink_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#3  0x00007f7fa6e2284b in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#4  0x00007f7fa6ddac03 in gr::block_executor::run_one_iteration() () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#5  0x00007f7fa6e2e59a in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#6  0x00007f7fa6e1e818 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#7  0x00007f7fa6e3d4e8 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#8  0x00007f7fa70f343b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#9  0x00007f7fa588e609 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#10 0x00007f7fa5423293 in clone () from /lib/x86_64-linux-gnu/libc.so.6

robotastic, 2021-07-28:
Thanks! I just made a change to remove that call to feof(). If you can pull down that latest version from github and try that, it would be awesome. You will need to delete the existing `core` file. The old one doesn't get over written when there is a crash.

I have a feeling it is a deeper problem though. I think one the pointer to the file maybe getting set to NULL while it is still being written to.

leee, 2021-07-28:
Turns out this affects my non-trunk trunk-recorders, although less frequently. Of my two conventional-only recorders, the more active one encounters segvs more often, another indication this is about call handling.

Looks like you still hit this check sometimes:
```
[/root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink_impl.cc] file i/o error
```

And my backtrace (above error message not from this):
```
(gdb) bt full 3
#0  _IO_ferror (fp=0x0) at ferror.c:35
        result = <optimized out>
        __self = <optimized out>
        __self = <optimized out>
        __value = <optimized out>
        __futex = <optimized out>
        __value = <optimized out>
        __futex = <optimized out>
        __private = <optimized out>
        __oldval = <optimized out>
        result = <optimized out>
        __ret = <optimized out>
        resultvar = <optimized out>
        __arg4 = <optimized out>
        __arg3 = <optimized out>
        __arg2 = <optimized out>
        __arg1 = <optimized out>
        _a4 = <optimized out>
        _a3 = <optimized out>
        _a2 = <optimized out>
        _a1 = <optimized out>
        __lll_private_flag_priv = <optimized out>
#1  0x000055cfdef4b824 in gr::blocks::nonstop_wavfile_sink_impl::dowork (
    this=0x55cfe6e47f30, noutput_items=853,
    input_items=std::vector of length 1, capacity 1 = {...},
    output_items=std::vector of length 0, capacity 0)
    at /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink_impl.cc:456
        chan = 0
        n_in_chans = 1
        sample_buf_s = -10914
        nwritten = 0
#2  0x000055cfdef4ab03 in gr::blocks::nonstop_wavfile_sink_impl::work (
    this=0x55cfe6e47f30, noutput_items=853,
    input_items=std::vector of length 1, capacity 1 = {...},
    output_items=std::vector of length 0, capacity 0)
    at /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink_impl.cc:339
        guard = {m = 0x55cfe6e481f0, is_locked = true}
        tags = std::vector of length 0, capacity 0
        this_key = {px = 0x55cfe550a0a0, pn = {pi_ = 0x55cfe549e410}}
        that_key = {px = 0x7ff0ac010850, pn = {pi_ = 0x7ff0ac010830}}
        squelch_key = {px = 0x7ff0ac0108b0, pn = {pi_ = 0x7ff0ac010890}}
        pos = 0
        nwritten = 226
(More stack frames follow...)
```

Looking at frame 2, indeed `d_fp` was `NULL`:
```
(gdb) frame 2
#2  0x000055cfdef4ab03 in gr::blocks::nonstop_wavfile_sink_impl::work (this=0x55cfe6e47f30, noutput_items=853, input_items=std::vector of length 1, capacity 1 = {...},
    output_items=std::vector of length 0, capacity 0) at /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink_impl.cc:339
339	  int nwritten = dowork(noutput_items, input_items, output_items);
(gdb) p d_fp
$1 = (FILE *) 0x0
```

Other variables in 2:
```
d_sample_count     = 0
d_fp               = (FILE *) 0x0
d_termination_flag = false
d_first_work       = true
curr_src_id        = 57608
state              = IDLE
guard              = {m = 0x55cfe6e481f0, is_locked = true}
d_current_call     = (Call *) 0x55cfe6da99b0
nwritten           = 226
```

In frame 1:
```
nwritten                  = 0
sample_buf_s              = -10914
d_fp                      = (FILE *) 0x0
chan                      = 0
n_in_chans                = 1
d_nchans                  = 1
noutput_items             = 853
current_base_filename     = "/root/logs/fdnyems/2021/7/28/7-1627461126_483281250", '\000' <repeats 203 times>
current_filename          = "/root/logs/fdnyems/2021/7/28/7-1627461126_483281250.wav", '\000' <repeats 199 times>
d_start_time              = 1627461126
record_more_transmissions = true
```

What I'm confused by is to have gotten into `dowork()`, and get to the following by frame 1:

https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L456

`d_fp` being null would've been implicitly converted to false, and therefore `!d_fp` be true. It's not modified after, at least what I can see. Why didn't this section get hit then?

https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L432-L436

Also weird is `d_first_work` is true in this same context. If it was `true`, then it would've been set false by:

https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L406-L430

But it remained true.

robotastic, 2021-07-28:
This is a great break down - thanks @leee I just checked and I don't think I am getting any SIGSEV. I am running a SmartNet and P25 system.... but I do not have any Analog Recorders enabled. Are you running analog recorders on the systems that are crashing?

I agree on you break down... those checks should have stopped a NULL pointer from making it to ferror() on that thread. work()/do_work() gets called on a gnuradio thread. Trunk Recorder is on a separate process/thread and can call stop_recorder():
https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L210

This then calls close_wav(), which sets the d_fp to NULL
https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L250-L255

I thought that the scoped_locks here:
https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L235
and here:
https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L280

would have stopped the threads from clobbering each other....

leee, 2021-07-28:
I can't debug further at the moment, but that's a good point. It's all conventional analog or P25 trunks, no more Type II on these hosts.

jliedtke6014, 2021-07-28:
Hi,  Regrabbed the beta this morning, ran it aain.  It ran for about 30mins and then died again.

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
--Type <RET> for more, q to quit, c to continue without paging--
Core was generated by `./recorder'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fbdcd27ef24 in ferror () from /lib/x86_64-linux-gnu/libc.so.6
[Current thread is 1 (Thread 0x7fbd1677c700 (LWP 57870))]
(gdb) bt
#0  0x00007fbdcd27ef24 in ferror () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x000055f7ad0e5671 in gr::blocks::nonstop_wavfile_sink_impl::dowork(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#2  0x000055f7ad0e3511 in gr::blocks::nonstop_wavfile_sink_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#3  0x00007fbdced1284b in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#4  0x00007fbdceccac03 in gr::block_executor::run_one_iteration() () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#5  0x00007fbdced1e59a in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#6  0x00007fbdced0e818 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#7  0x00007fbdced2d4e8 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.8.1
#8  0x00007fbdcefe343b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#9  0x00007fbdcd77e609 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#10 0x00007fbdcd313293 in clone () from /lib/x86_64-linux-gnu/libc

I'm honestly willing to give you access to this machine if you want to get into it to figure out what is going on.

robotastic, 2021-07-29:
Thanks @jliedtke6014 It looks like it is that block write there. I am not sure why it is able to write the sample in the function call above that line but then crash right below it. It might make sense to kill this whole block:
https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L456-L460

You can't actually call stop_recording() there because it will try to grab a scoped_lock and hang. Who knows, maybe something like this is happening. Let's see if this change fixes things, if not, having access might help because I am still not seeing it on my local systems.

robotastic, 2021-07-29:
OK! I think I found it... set_source() used to not have a lock at the beginning: https://github.com/robotastic/trunk-recorder/blob/f2d00d2c212bc615581a88e64d5cb366f5b7e362/lib/gr_blocks/nonstop_wavfile_sink_impl.cc#L191-L198

That means it is possible that end_transmission() could be called in there while the block was busy still writing to a file.

ALSO - set_source() is only called by the analog_recorder and I don't have any running on my system... so maybe?

The latest version should be pretty good

jliedtke6014, 2021-07-29:
Pulled the latest, started using it at 8:44am,  EMS traffic is heavy this morning and is all Analog smartnet so we shall see what happens.

jliedtke6014, 2021-07-29:
So far 11:30am and no crashes.

robotastic, 2021-07-29:
Sweet! I am going to keep my fingers crossed!

leee, 2021-07-29:
Great, works for me now for a few hours of testing.

It looks like there's some audio doubling though, [this is just one example](https://openmhz.com/system/fdnyems?call-id=61031651ae1ee7650baa5e38&time=1627592257001) but it happens up and down the call queue (I've reverted for now, really should get a box at that site just for testing).

@austein also pointed out the new behavior in the call concluder breaks existing expectations (like in the example uploadScript script and how it handles base names), I haven't given the changes so far the review it deserves so don't know how `-call.wav` files differ from `.wav` which don't appear anymore, as well as why `.m4a` files don't have the `-call` suffix in their basename.

robotastic, 2021-07-30:
Progress!! 

That is really weird - let go dig into why there is doubling of the call. Have you only noticed it on Analog calls or are you seeing it on Digital calls to?

Instead of having a single wav recording for the entire call - there is now a separate call for each transmission. After the call is finished, sox is called and it merges the wav files together into a combined one. 

My bet is that for some reason the wav file is getting added in twice.

robotastic, 2021-07-31:
@leee - I think I have both of these issues a little better addressed:
- I think I figured out the doubling up, and you example really helped. In it, the source changed with in the first second `0[0]4657[0.41]`. It was likely that there was just a slight delay in a trunk message with a source assigned coming in. Anyhow, previously it would have started a new transmission since the source id changed. However, the start time would have been the same since it was under a second. As a result, the filenames would be the same, and a the end, the file would be combined with itself, sounding like it was doubled. I changed it so filenames will be unique, and also no longer treating a change from 0 to a Source ID as being a new transmission.
6b6342d7e7c84a0cbcd2f85b7aecb685b6aef4e6

- For the filenames, I changed it so that the .wav, .m4a, and .json file all had the same base. I also changed it so that the .m4a filename is now passed to the upload script.
55a6ace15287b1ecccd7569fe65abd2f60348a0e

leee, 2021-07-31:
@robotastic I see, nice! I'm sorry I haven't had more time to spare lately to give the latest work the attention it deserves, but at least let me at least kick off a reinstall so you can see the latest changes in action.

Looks like `maxDuration` doesn't get adhered to in some way, [example](https://openmhz.com/system/fdnyfire?call-id=6105543a6a953be419b8dcae&time=1627739147001) - I have a `maxDuration` of `30` seconds on this system, but this file broke that limit and went to 45 seconds.

This also could be something to do with how the MDC decoder is handling post-transmission IDs.

jliedtke6014, 2021-07-31:
@robotastic   FYI for my issue its been running for days without crashing.

robotastic, 2021-07-31:
@leee No worries! I may have to go add Max Duration back in again. It may take a little finesse because the whole re-design was to optimize for allow the recorders / calls keep going until they stopped getting audio.

@jliedtke6014 Awesome! that is great news

jliedtke6014, 2021-08-01:
@robotastic  During my rebuild of the beta to see if I can get rid of the duplicated call. This happened on two machines I was rebuilding on:

[ 84%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/unit_tag.cc.o
[ 85%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/unit_tags.cc.o
[ 86%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o
[ 86%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/call_concluder/call_concluder.cc.o
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc: In function ‘Call_Data_t upload_call_worker(Call_Data_t)’:
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:124:44: warning: ‘-call_’ directive output may be truncated writing 6 bytes into a region of size between 1 and 255 [-Wformat-truncation=]
  124 |         snprintf(call_info.filename,255,"%s-call_%lu.wav",t.base_filename,call_info.call_num);
      |                                            ^~~~~~
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:124:41: note: using the range [0, 18446744073709551615] for directive argument
  124 |         snprintf(call_info.filename,255,"%s-call_%lu.wav",t.base_filename,call_info.call_num);
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /usr/include/c++/9/ext/string_conversions.h:43,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
                 from /usr/include/c++/9/memory:80,
                 from /usr/include/c++/9/thread:39,
                 from /home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.h:6,
                 from /home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:1:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:35: note: ‘__builtin___snprintf_chk’ output between 12 and 285 bytes into a destination of size 255
   67 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   68 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:125:51: warning: ‘-call_’ directive output may be truncated writing 6 bytes into a region of size between 1 and 255 [-Wformat-truncation=]
  125 |         snprintf(call_info.status_filename,255,"%s-call_%lu.json",t.base_filename,call_info.call_num);
      |                                                   ^~~~~~
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:125:48: note: using the range [0, 18446744073709551615] for directive argument
  125 |         snprintf(call_info.status_filename,255,"%s-call_%lu.json",t.base_filename,call_info.call_num);
      |                                                ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /usr/include/c++/9/ext/string_conversions.h:43,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
                 from /usr/include/c++/9/memory:80,
                 from /usr/include/c++/9/thread:39,
                 from /home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.h:6,
                 from /home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:1:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:35: note: ‘__builtin___snprintf_chk’ output between 13 and 286 bytes into a destination of size 255
   67 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   68 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:126:45: warning: ‘-call_’ directive output may be truncated writing 6 bytes into a region of size between 1 and 255 [-Wformat-truncation=]
  126 |         snprintf(call_info.converted,255,"%s-call_%lu.m4a",t.base_filename,call_info.call_num);
      |                                             ^~~~~~
/home/administrator/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:126:42: note: using the range [0, 18446744073709551615] for directive argument
  126 |         snprintf(call_info.converted,255,"%s-call_%lu.m4a",t.base_filename,call_info.call_num);
      |                                          ^~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/c++/9/cstdio:42,
                 from /usr/include/c++/9/ext/string_conversions.h:43,
[... cut ...]
[ 91%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/decoders/star_decode.cc.o
[ 91%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/decoders/signal_decoder_sink_impl.cc.o
[ 92%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/decoders/tps_decoder_sink_impl.cc.o
[ 93%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/decoder_wrapper_impl.cc.o
[ 93%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/plugin_wrapper_impl.cc.o
[ 94%] Building CXX object CMakeFiles/recorder.dir/lib/gr_blocks/selector_impl.cc.o

robotastic, 2021-08-02:
@JayManCanada6014 - that is a lot of ugly warnings! I just made some changes to fix it. It looks like it only happens with GCC and not Clang, so I didn't spot it on my Mac.

robotastic, 2021-08-02:
@leee I added support for `maxDuration` into v4.0

leee, 2021-08-02:
Running this now, just a quick eyeball shows [this 34s transmission](https://openmhz.com/system/fdnyems?call-id=6107bc476a953be419f6b1ca&time=1627896866001).

robotastic, 2021-08-02:
@leee yea... it looks like I was using an approach that doesn't stop things as quickly as possible. Just made a small change that should fix that.

leee, 2021-08-02:
Running the latest now, not a lot of examples to look at and check yet but so far so good: [[1]](https://openmhz.com/system/fdnyems?call-id=6107d71b6a953be419f8529b&time=1627903739001) [[2]](https://openmhz.com/system/fdnyfire?call-id=6107d79c6a953be419f85cc3&time=1627903869001)

In the second example that MDC ID doesn't get taken in though, and it looks like a lot of the calls on that analog conventional system (which uses post-transmission/trailing IDs) don't get IDs.

robotastic, 2021-08-02:
Nice! I will dig into the MDC ID stuff. The whole path for processing it and associating with a call is pretty convoluted.

leee, 2021-08-30:
Just for tracking, I'm still getting calls that exceed maxDuration. (I don't have an analysis or anything for you, sorry.)

But more interesting is that I'm seeing (for a conventional system) individual channel recorders drop out and never come back without process reboot. Something to dig into eventually. 🙂

robotastic, 2021-09-02:
@leee Just to check, those are for an Analog conventional system and not a P25 conventional system? I can try doing some stress testing with my FRS radio this weekend. There could be some bugs in my squelch start/stop code. 

I will also put fresh eyes on the maxDuration to see why things are slipping through. Is that for analog recordings or digital?

robotastic, 2021-09-10:
@leee I am not able to recreate the problem with Conventional recorders getting dropping and not returning. Could you share the config file for the system this is happening on? It could be that there are some options that are contributing... Perhaps maxDuration. Do you know how often it happens? Any chance it could be a squelch issue? I am just going to be using a FRS radio to try and recreate it.

robotastic, 2021-09-13:
@lee - Something to try. Could you make sure you have all Signaling decoding turned off? Remove MDC decoding from the config. There is a chance that the crash could be coming from there.

robotastic, 2021-10-21:
Closing this for now - there is a lot of different things happening in this issue and I think most of them have been addressed. Please open a new issue if things re-emerge.

closed 2021-10-21 by robotastic