TrunkRecorder/trunk-recorder #499: v4.0-beta - P25 Phase 2  - No Recordings

JayManCanada6014, 2021-08-01:
[2021-08-01 17:22:36.400444] (info)   [trunk-recorder-p25]      11C     TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 17:22:48.418849] (info)   [trunk-recorder-p25]      13C     TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 17:22:57.920702] (info)   [trunk-recorder-p25]      14C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 Mod: true
[2021-08-01 17:22:58.244378] (info)   [trunk-recorder-p25]      15C     TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 17:23:01.004509] (info)   [trunk-recorder-p25]      14C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Stopping Recorded Call, setting call state to COMPLETED - Last Update: 2s
[2021-08-01 17:23:01.004577] (info)   [trunk-recorder-p25]      14C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Concluding Recorded Call - Last Update: 2s      Call Elapsed: 4
[2021-08-01 17:23:01.004615] (error)   [trunk-recorder-p25]     14C     TG:       1528   Freq: 8.67538e+08      No Transmission were recorded!
[2021-08-01 17:23:01.035021] (info)   [trunk-recorder-p25]      16C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 Mod: true
[2021-08-01 17:23:01.358776] (info)   [trunk-recorder-p25]      17C     TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 17:23:06.710612] (info)   [trunk-recorder-p25]      20C     TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 17:23:08.006817] (info)   [trunk-recorder-p25]      16C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Stopping Recorded Call, setting call state to COMPLETED - Last Update: 2s
[2021-08-01 17:23:08.007059] (info)   [trunk-recorder-p25]      16C     TG:           ParaTranspo 1 (      1528)        Freq: 867.537500 MHz     Concluding Recorded Call - Last Update: 2s      Call Elapsed: 7
[2021-08-01 17:23:08.007364] (error)   [trunk-recorder-p25]     16C     TG:       1528   Freq: 8.67538e+08      No Transmission were recorded!
[2021-08-01 17:23:09.666570] (info)   [trunk-recorder-p25]      22C     TG:           ParaTranspo 1 (      1528)        Freq: 867.387500 MHz     Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 Mod: true
[2021-08-01 17:23:10.425340] (info)   [trunk-recorder-p25]      23C     TG:          Operators Ch 2 (      1506)        Freq: 867.137500 MHz     Starting P25 Recorder Num [1]   TDMA: false     Slot: 0 Mod: true

Everything is No Transmission were recorded!  since I installed Beta 4.0 on the 21st of July.  Even the latest build.
It is this system  https://www.radioreference.com/apps/db/?sid=8161

robotastic, 2021-08-02:
hmm... I just double checked and tuned in my local P25 P2 system, and things were recording fine. There weren't many changes that touched the core P25 stuff. The fact that it is a Phase 2 system could be a bit of a red herring. There might be some other bug. It is really weird that the freq for some of these is 0.0 MHz... It does look like the Phase 1 calls are getting recorded though, so maybe there is some sort Phase 2 issue.

And just to double check, if you switch back to the Master branch, everything records correctly?

JayManCanada6014, 2021-08-02:
The current master I get lines and lines of this, and then no recording just the json files.

[2021-08-01 22:24:54.898371] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:55.181924] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:55.505901] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:55.779613] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:56.103623] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:56.376938] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:56.711181] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:56.984738] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:57.258441] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:59.385959] (info)   [trunk-recorder-p25]      TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-01 22:24:59.710286] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:24:59.983881] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:25:00.307284] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-01 22:25:00.529801] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0

robotastic, 2021-08-02:
Hmm... Did things ever work with this system? It looks there is some underlying issue with decoding the P25 Phase 2 channel assignments. It can't get the TG # or the Frequency. In your config.json file, if you set `logLevel: debug` it should print out a lot more messages on the P25 processing. It might help spot where things are going wrong.

JayManCanada6014, 2021-08-02:
This did work until the recent changes to the master.

INSTANCE
-------------------------------------
[2021-08-02 08:40:52.190149] (info)   Capture Directory: /recordings
[2021-08-02 08:40:52.190161] (info)   Upload Server:
[2021-08-02 08:40:52.190172] (info)   Broadcastify Calls Server: https://api.broadcastify.com/call-upload
[2021-08-02 08:40:52.190183] (info)   Status Server:
[2021-08-02 08:40:52.190193] (info)   Instance Key:
[2021-08-02 08:40:52.190203] (info)   Instance Id:
[2021-08-02 08:40:52.190214] (info)   Broadcast Signals: false
[2021-08-02 08:40:52.190224] (info)   Default Mode: digital
[2021-08-02 08:40:52.190236] (info)   Call Timeout (seconds): 1
[2021-08-02 08:40:52.190248] (info)   Log to File: false
[2021-08-02 08:40:52.190261] (info)   Log Directory: logs
[2021-08-02 08:40:52.190272] (info)   Control channel decoding warnings (< messages/second): 10
[2021-08-02 08:40:52.190283] (info)   Control channel decoding updates (seconds): 3
[2021-08-02 08:40:52.190296] (info)   Control channel retune limit: 0
[2021-08-02 08:40:52.190306] (info)   Frequency format: 1
[2021-08-02 08:40:52.190316] (info)   Status as String: true
[2021-08-02 08:40:52.190326] (info)   Log Level: debug
[2021-08-02 08:40:52.190343] (info)   [trunk-recorder-p25]      Started with Control Channel: 866.637500 MHz
[2021-08-02 08:40:52.190584] (debug)   P25 trunking Decim: 25 Decim2:  4
[2021-08-02 08:40:52.190613] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 96000 Second decimated rate: 24000 FA: 6250 FB: 12000 System Rate: 2400000
[2021-08-02 08:40:52.194637] (info)      P25 Trunking ARB - Initial Rate: 2400000 Resampled Rate: 24000 Initial Decimation: 25 System Rate: 24000 ARB Rate: 1
Allocating 15 zero-copy buffers
[2021-08-02 08:40:52.489741] (info)   [trunk-recorder-p25]      TG:                OPS West (      3003)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-02 08:40:52.763674] (info)   [trunk-recorder-p25]      Decoding System ID 2E WACN: 16B7E NAC: 2E
[2021-08-02 08:40:52.764152] (info)     - Starting P25 Recorder Num [0] TG:                  OFSD71 (      3539)        Freq: 867.387500 MHz     TDMA: false     Slot: 0 Mod: true
[2021-08-02 08:40:52.764544] (info)   [trunk-recorder-p25]      TG:                  OFSD71 (      3539)        Freq: 867.387500 MHz     Starting Recorder on Src: rtl=0
[2021-08-02 08:40:53.688651] (info)   [trunk-recorder-p25]      TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-02 08:40:53.972301] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:40:54.296538] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:40:54.569757] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:40:54.782336] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    3539      TDMA -1  sa 36408        Encrypt 0       Bandwidth: 12.5
[2021-08-02 08:40:54.843052] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    3539      TDMA -1  sa 36408        Encrypt 0       Bandwidth: 12.5
[2021-08-02 08:40:57.514254] (debug)   tsbk00   Chan Grant      Channel ID:  3684       Freq: 867.137500 MHz    ga    3524      TDMA -1  sa 36107        Encrypt 0       Bandwidth: 12.5
[2021-08-02 08:40:57.514326] (info)     - Starting P25 Recorder Num [1] TG:                  OFSD41 (      3524)        Freq: 867.137500 MHz     TDMA: false     Slot: 0 Mod: true
[2021-08-02 08:40:57.514364] (info)   [trunk-recorder-p25]      TG:                  OFSD41 (      3524)        Freq: 867.137500 MHz     Starting Recorder on Src: rtl=0
[2021-08-02 08:40:57.514391] (debug)   tsbk00   Chan Grant      Channel ID:  3684       Freq: 867.137500 MHz    ga    3524      TDMA -1  sa 36107        Encrypt 0       Bandwidth: 12.5
[2021-08-02 08:40:59.478576] (debug)   tsbk14: SNDCP Data Channel Grant
[2021-08-02 08:40:59.975057] (info)   [trunk-recorder-p25]      TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-02 08:41:00.247982] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:41:00.571602] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:41:00.896073] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 08:41:01.007534] (trace)   [trunk-recorder-p25]     TG:                  OFSD71 (      3539)        Freq: 867.387500 MHz     Ending Recorded Call - Last Update: 2s  Call Elapsed: 9
[2021-08-02 08:41:01.007964] (info)     - Stopping P25 Recorder Num [0] TG:                  OFSD71 (      3539)        Freq: 867.387500 MHz     TDMA: false     Slot: 0
[2021-08-02 08:41:01.008095] (error)   wav error closing file
[2021-08-02 08:41:01.008107] (info)   [trunk-recorder-p25]      Deleting this call as it has a duration less than minimum duration of 0  TG:                  OFSD71 (      3539)        Freq: 867.387500 MHz    Call Duration: 0s
[... cut ...]
[2021-08-02 08:41:11.013739] (info)   [trunk-recorder-p25]      Deleting this call as it has a duration less than minimum duration of 0  TG:                  OFSD41 (      3524)        Freq: 867.387500 MHz    Call Duration: 0s
[2021-08-02 08:41:11.013779] (error)   Could not delete file /recordings/trunk-recorder-p25/2021/8/2/3524-1627908068_867387500.wav
[2021-08-02 08:41:12.481128] (info)   [trunk-recorder-p25]      TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq

leee, 2021-08-02:
> This did work until the recent changes to the master.

Just to make sure, can you post what branch you're on and the last commit? Something like:

```
$ git symbolic-ref --short HEAD
...
```
and
```
$ git --no-pager log -1 --stat
...
```

JayManCanada6014, 2021-08-02:
root@trunk-recorder-p25:/home/administrator/trunk-recorder# git symbolic-ref --short HEAD
master

root@trunk-recorder-p25:/home/administrator/trunk-recorder# git --no-pager log -1 --stat
commit 3b269bd70877f63793c08582a7ed29b20f2a7dbb (HEAD -> master, origin/master, origin/HEAD)
Author: Luke Berndt <lukekb@gmail.com>
Date:   Thu Jul 22 21:02:15 2021 -0400

    Fixed debug recorder tuning issue

 trunk-recorder/main.cc | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

leee, 2021-08-02:
Oh, okay. It seems like you're still running a beta build. Where are you running `recorder` from, e.g. from the build directory or from `/usr/local/bin`?

JayManCanada6014, 2021-08-02:
I've been deleting the trunk-recorder directory and re-running this each time.    I did the same to switch to the betas, and repull the betas.  Sorry not a big github user couldn't figure out a better way.

git clone https://github.com/robotastic/trunk-recorder.git
cd trunk-build
cmake ../trunk-recorder
make

leee, 2021-08-02:
In this example, do you also delete the `trunk-build` directory? Are you running the built recorder from somewhere like `/home/administrator/trunk-build/recorder` or from `/usr/local/bin/recorder`?

JayManCanada6014, 2021-08-02:
I do not delete the trunk-build and I am running it right from /home/administrator/trunk-build  with just ./recorder

leee, 2021-08-02:
Just to humor me, you may wish to create a new directory, say `/home/administrator/trunk-build-test` and `cmake/make` from there, and see if that built recorder (which should be what's in `master`) behaves like the beta.

JayManCanada6014, 2021-08-02:
> Just to humor me, you may wish to create a new directory, say `/home/administrator/trunk-build-test` and `cmake/make` from there, and see if that built recorder (which should be what's in `master`) behaves like the beta.

Same thing.  

DMA -1  sa 25006        Encrypt 1       Bandwidth: 12.5
[2021-08-02 15:21:21.049130] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3051 New src: 0
[2021-08-02 15:21:21.049272] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 2502 New src: 0
[2021-08-02 15:21:21.322753] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 2502 New src: 0
[2021-08-02 15:21:21.323002] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:21.647063] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 2502 New src: 0
[2021-08-02 15:21:21.647313] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:21.930583] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 2502 New src: 0
[2021-08-02 15:21:21.930769] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:22.001675] (trace)   [trunk-recorder-p25]     TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz     Ending Recorded Call - Last Update: 2s  Call Elapsed: 4
[2021-08-02 15:21:22.002180] (info)     - Stopping P25 Recorder Num [0] TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz     TDMA: false     Slot: 0
[2021-08-02 15:21:22.002449] (error)   wav error closing file
[2021-08-02 15:21:22.002542] (info)   [trunk-recorder-p25]      Deleting this call as it has a duration less than minimum duration of 0  TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz    Call Duration: 0s
[2021-08-02 15:21:22.002639] (error)   Could not delete file /recordings/trunk-recorder-p25/2021/8/2/2039-1627932078_867387500.wav
[2021-08-02 15:21:22.245516] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 2502 New src: 0
[2021-08-02 15:21:22.245694] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:23.946289] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    2039      TDMA -1  sa 22699        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:21:23.946367] (info)     - Starting P25 Recorder Num [0] TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz     TDMA: false     Slot: 0 Mod: true
[2021-08-02 15:21:23.946407] (info)   [trunk-recorder-p25]      TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz     Starting Recorder on Src: rtl=0
[2021-08-02 15:21:23.997045] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    2039      TDMA -1  sa 22699        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:21:23.997094] (debug)   tsbk00   Chan Grant      Channel ID:  3716       Freq: 867.537500 MHz    ga    3051      TDMA -1  sa 31570        Encrypt 1       Bandwidth: 12.5
[2021-08-02 15:21:24.057958] (debug)   tsbk00   Chan Grant      Channel ID:  3716       Freq: 867.537500 MHz    ga    3051      TDMA -1  sa 31570        Encrypt 1       Bandwidth: 12.5
[2021-08-02 15:21:24.270453] (debug)   tsbk2f   Unit Group Affiliation  Source ID:   17026      Group Address: 1529     Anouncement Goup: 1535
[2021-08-02 15:21:24.816863] (info)   [trunk-recorder-p25]      TG:                       - (         0)        Freq:   0.000000 MHz     Not Recording: no source covering Freq
[2021-08-02 15:21:24.928158] (debug)   tsbk2f   Unit Group Affiliation  Source ID:   17026      Group Address: 1529     Anouncement Goup: 1535
[2021-08-02 15:21:25.140991] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:25.474861] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:25.798623] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:26.081442] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:21:26.728930] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    2039      TDMA -1  sa 22599        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:21:26.779592] (debug)   tsbk00   Chan Grant      Channel ID:  3704       Freq: 867.387500 MHz    ga    2039      TDMA -1  sa 22599        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:21:26.891298] (debug)   tsbk00   Chan Grant      Channel ID:  3716       Freq: 867.537500 MHz    ga    2502      TDMA -1  sa 25004        Encrypt 1       Bandwidth: 12.5
[... cut ...]
[2021-08-02 15:21:34.409605] (error)   wav error closing file
[2021-08-02 15:21:34.409613] (info)   [trunk-recorder-p25]      Deleting this call as it has a duration less than minimum duration of 0  TG:          BylawParking 1 (      2039)        Freq: 867.387500 MHz    Call Duration: 0s
[2021-08-02 15:21:34.409645] (error)   Could not delete file /recordings/trunk-recorder-p25/2021/8/2/2039-1627932092_867387500.wav
[2021-08-02 15:21:34.409660] (info)   Exiting...
[2021-08-02 15:21:34.409669] (info)   stopping flow graph

leee, 2021-08-02:
Interesting, can you go back one commit and build from that?

in `~/trunk-recorder`, you can do this by:

```
$ git reset --hard HEAD~1
```

Or specify:

```
$ git reset --hard 92037671cb45d8c81512ec03f7374c49be41b220
```

JayManCanada6014, 2021-08-02:
Same thing  and I did make sure it pulled the right one at the bottom of the log.

[2021-08-02 15:40:23.205439] (debug)   tsbk00   Chan Grant      Channel ID:  3716       Freq: 867.537500 MHz    ga    3519      TDMA -1  sa 35960        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:40:23.206039] (trace)   [trunk-recorder-p25]     TG:                  OFSD31 (      3519)        Freq: 867.537500 MHz     Update Retuning - New Freq: 867.537500 MHz      Elapsed: 9s     Since update: 1s
[2021-08-02 15:40:23.206378] (debug)   tsbk00   Chan Grant      Channel ID:  3716       Freq: 867.537500 MHz    ga    3519      TDMA -1  sa 35960        Encrypt 0       Bandwidth: 12.5
[2021-08-02 15:40:23.479968] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:23.480116] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:23.803778] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:23.803910] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:24.077473] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:24.077612] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:24.402024] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:24.403333] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:24.677469] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:24.677650] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:24.951174] (debug)   tsbk2f   Unit Deregistration ACK Source ID:   30352
[2021-08-02 15:40:25.002219] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:25.002661] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:25.276387] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:25.276789] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
[2021-08-02 15:40:25.601136] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 3519 New src: 0
[2021-08-02 15:40:25.601567] (error)   WARNING: Old call found w/this message.source. Old TG:                       - (         0)/0New TG: 0 New src: 0
^C[2021-08-02 15:40:26.199363] (info)   Signal caught! Closing calls...
[2021-08-02 15:40:26.199501] (trace)   [trunk-recorder-p25]     TG:                  OFSD31 (      3519)        Freq: 867.537500 MHz     Ending Recorded Call - Last Update: 1s  Call Elapsed: 12
[2021-08-02 15:40:26.199728] (info)     - Stopping P25 Recorder Num [1] TG:                  OFSD31 (      3519)        Freq: 867.537500 MHz     TDMA: false     Slot: 0
[2021-08-02 15:40:26.199789] (error)   wav error closing file
[2021-08-02 15:40:26.199825] (info)   [trunk-recorder-p25]      Deleting this call as it has a duration less than minimum duration of 0  TG:                  OFSD31 (      3519)        Freq: 867.537500 MHz    Call Duration: 0s
[2021-08-02 15:40:26.199881] (error)   Could not delete file /recordings/trunk-recorder-p25/2021/8/2/3519-1627933214_867387500.wav
[2021-08-02 15:40:26.199926] (info)   Exiting...
[2021-08-02 15:40:26.199967] (info)   stopping flow graph
root@trunk-recorder-p25:/home/administrator/trunk-build# ../trunk-recorder/ git --no-pager log -1 --stat
bash: ../trunk-recorder/: Is a directory
root@trunk-recorder-p25:/home/administrator/trunk-build# cd ..
root@trunk-recorder-p25:/home/administrator# cd trunk-recorder/
root@trunk-recorder-p25:/home/administrator/trunk-recorder# git --no-pager log -1 --stat
commit 92037671cb45d8c81512ec03f7374c49be41b220 (HEAD -> master)
Merge: cc6e295 c27fb30
Author: Luke Berndt <luke@robotastic.com>
Date:   Mon Jul 19 22:16:20 2021 -0400

    Merge pull request #494 from leee/leee/analog-deemph

    Adjust analog_recorder deemphasis filter

JayManCanada6014, 2021-08-02:
root@trunk-recorder-p25:/home/administrator/trunk-recorder# git --no-pager log -1 --stat
commit 92037671cb45d8c81512ec03f7374c49be41b220 (HEAD -> master)
Merge: cc6e295 c27fb30
Author: Luke Berndt <luke@robotastic.com>
Date:   Mon Jul 19 22:16:20 2021 -0400

    Merge pull request #494 from leee/leee/analog-deemph

    Adjust analog_recorder deemphasis filter

It didn't paste the commit in the message above..

leee, 2021-08-02:
Do you recall what version/commit you last had a working build of?

Can you try `75b8796a2c81039db0264cc5afd997df9eecf860`?

JayManCanada6014, 2021-08-02:
> Do you recall what version/commit you last had a working build of?
> 
> Can you try `75b8796a2c81039db0264cc5afd997df9eecf860`?

Still not working, can you give me something like 4 months back that you know is stable?

leee, 2021-08-02:
That was something 3 months back that I knew was stable 😅

You could try to start from `v3.3.0` and start your binary search of finding the broken commit from there:

```
$ git reset --hard v3.3.0
(or ee5158492f6967f27ff0f3f5f3f16ae42ff22883)
```

Could you also show:
```
$ which recorder
```

And maybe delete it, just so there's no way to accidentally run the beta from `/usr/local/bin/recorder` when you've been trying to run `./recorder` instead?

leee, 2021-08-02:
> Still not working, can you give me something like 4 months back that you know is stable?

And just to make a point... That commit (`75b8796`) shouldn't even have the `WARNING: Old call found w/this message.source.` warnings. Were you sitll seeing this?

JayManCanada6014, 2021-08-02:
root@trunk-recorder-p25:/home/administrator# which recorder
root@trunk-recorder-p25:/home/administrator#
Once I deleted the trunk-build and trunk-recorder there are no other recorders.

JayManCanada6014, 2021-08-02:
Hmmm ok, something is really wrong and I'll say it's not trunk recorder.   Sorry @robotastic 
I am going to install OpenWebRX and see whats up.

JayManCanada6014, 2021-08-02:
Nope strange, strong control channel but yeah no audio at all on any build.
Argh.

JayManCanada6014, 2021-08-03:
FML - Just started working, even on the beta 4.0.  No idea why.

closed 2021-08-03 by JayManCanada6014

robotastic, 2021-08-03:
@JayManCanada6014 This is so weird!! Let me know if it crops up again. I have a feeling there is some P25 feature on this system that is getting parsed incorrectly.

leee, 2021-08-03:
If the debug messages are to be believed, then i agree as well. At least you have a working build again, including with the beta, so that can be ruled out now.

ab0tj, 2021-09-07:
@JayManCanada6014 are you by chance running more than one source? I get this when trying to use two sources, but works fine with just one source.