TrunkRecorder/trunk-recorder #500: Settings API

blackwellj, 2021-08-23:
I would like to make a webpage to configure trunk Recorder. would it be possible to create a rest api to allow settings changes on the fly without the need to restart the docker container each time?

robotastic, 2021-09-02:
Totally! I am trying to keep as much added features out of the core of Trunk Recorder to keep things clean. Let me see if there is some way to make this work with the plugin system. Ultimately, the way you send in those changes should be flexible. It could be over a REST API or WebSocket or MQTT. It would be great to be able to adjust the error correction on the fly though.

blackwellj, 2021-09-02:
Error correction, channels and squelch would be amazing! I’m using the system to monitor analog marine vhf communications

JoeGilkey, 2021-11-28:
I have an idea for this. I have an application that I wrote that reads the config files, runs rtl_power on the control channels, then uses the results to calculate ppm. The ppm values are then written out to the config file. The config read/write is in a separate program that uses grpc to communicate with the ppm calculation program. I'll work on making something more formal. You could use a grpc -> rest gateway (see this example: [gRPC with REST and Open APIs](https://grpc.io/blog/coreos/)) to expose this.

robotastic, 2024-12-15:
This is still a good idea.... I think it could work for things around tuning, like Gain and Error