TrunkRecorder/trunk-recorder #501: Smartnet - Low Volume on Analog

JayManCanada6014, 2021-08-23:
NOT using the Beta but I am using the latest master.   Fairly sure it's not trunk but a gnuradio update that did it but analog comms on smartnet is very low.
I've reinstalled from scratch on two different OS's with the same result.  In the config the audio is turned up to 32 but that isn't high enough.
Also off two separate tower sites with different talkgroups.

leee, 2021-08-31:
This could possibly be because of the change I submitted in #494 (you can read my justification of that change in #478).

I only changed around the behavior of the deemphasis filter, but didn't bother to boost gain overall afterwards, because I was hoping that level normalization during the uploader stage (audio files getting processed with `sox`) would make that okay.

If you want to experiment, you could test some local changes, either reverting my change, changing this back to 75µs:

https://github.com/robotastic/trunk-recorder/blob/3b269bd70877f63793c08582a7ed29b20f2a7dbb/trunk-recorder/recorders/analog_recorder.cc#L148

Or maybe try bumping gain of the low pass filter here from 1:

https://github.com/robotastic/trunk-recorder/blob/3b269bd70877f63793c08582a7ed29b20f2a7dbb/trunk-recorder/recorders/analog_recorder.cc#L173

robotastic, 2021-09-02:
@JayManCanada6014 Let me know if reverting this change fixes it for you? We can always make this an config file parameter too.

JayManCanada6014, 2021-09-02:
> @JayManCanada6014 Let me know if reverting this change fixes it for you? We can always make this an config file parameter too.

Yeah changing it from  d_tau = 0.00075; // 750us   to  d_tau = 0.000075; // 750us 

Brought the audio levels back up on analog.

taclane, 2021-09-03:
> Let me know if reverting this change fixes it for you? We can always make this an config file parameter too.

I've been wondering why conventional analog was so quiet on v3.x, but loud and clear on the v4 beta.  Seem like this thread had the answer.  

Bumping d_tau back down to 0.000075 made it so the `analogLevels` setting worked again.