TrunkRecorder/trunk-recorder #503: Is there a way to get the Signal strength in DBZ of a TX

MaxwellDPS, 2021-08-30:
Is there a way to get the Signal strength in DBZ of a TX? Could this be added to the call json?

robotastic, 2021-09-02:
I had been interested in this... I have not found a good way to do Signal to Noise... but it looks like a basic Db could be possible:
https://www.tablix.org/~avian/blog/archives/2015/04/signal_power_in_gnu_radio/

Let me add this to the queue. I also have to revisit how the error measurements in the P25 Decoder is handled. It would be great to collect both metrics.

MaxwellDPS, 2021-09-07:
@robotastic That will help debugging 110%. Thanks for adding to the que!

robotastic, 2024-12-15:
Rough signal strength is available on conventional systems.... let me see if I can add it to all channels