TrunkRecorder/trunk-recorder #506: per system/talkgroup upload disable

ZeroChaos-, 2021-08-31:
There are lots of different uses for radio systems and talk groups. Sometimes, while I may want to personally monitor one, I don't feel comfortable uploading it.  While running multiple differently configured instances of trunk-recorder can already accomplish the goal, it would be a nice feature addition to add an upload=true/false option per system (and maybe even talkgroup?) to control the behavior.

Again, this can already be done on one system by running two trunk-recorders and two config files, but it would be nice to run just one.

leee, 2021-08-31:
You can disable uploads on a system basis by not populating `apiKey` and `broadcastifyApiKey`. No need for more instances of `trunk-recorder` which also consumes more SDRs that might already be covering the same spectrum.

Per-talkgroup management is a different story. In some view this is beyond the scope of `trunk-recorder`. If you're uploading somewhere else with `uploadScript`, you can handle cherrypicking uploads starting from there. If you're uploading to an OpenMHz instance or Broadcastify Calls, you can look at these services' documentation or the code for the uploaders, and in essence reimplement them in an `uploadScript`.

ZeroChaos-, 2021-08-31:
I didn't realize the keys were per system, so this bug probably invalid.  I'm re-implementing uploads myself in the upload script because I want to do some heavy filtering and light metadata modification :-)

Sorry for the noise, please carry on :-D

closed 2021-08-31 by ZeroChaos-

leee, 2021-08-31:
I understand, the documentation can also be kind of dense to fresh eyes which could use the documentation most, so if you're up for the task I'm sure they could be made easier to digest!