TrunkRecorder/trunk-recorder #507: Audio processing before upload

ab0tj, 2021-09-03:
The P25 system that I monitor has quite a wide range of audio levels, ranging from ear splitting to barely audible. I would like to run the .wav files through sox to normalize the audio levels before uploading. I have tried using uploadScript to process the .wav file, but the file that is uploaded to OpenMHz and Broadcastify is always the original (but encoded into AAC) instead of the modified version. Adding the ability to run a script that processes the audio before uploading would be helpful.

robotastic, 2021-09-08:
That is a pretty good idea - right now it is run as a system command, but it could be a small shell script so people can edit it easier. The command is here: https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/uploaders/call_uploader.cc#L20

I may have messed up the command, so let me know if there are better default sox settings.

ab0tj, 2021-09-08:
As I understand it, the 'norm' option operates on the peak level of the audio. So in that case a loud "blip" in the signal will cause the rest of the file to have negative gain applied. 'compand' seems to be needed, maybe in conjunction with 'norm'. There's some examples here: https://forum.doom9.org/showthread.php?t=165807