TrunkRecorder/trunk-recorder #510: Unable to build on Arch Linux

bijan588, 2021-09-07:
Hi,
I'm stuck here.

I am unable to build any of the trunk-recorder versions including the beta.

Initially I got an error and found a report on here about it being from a new version of GNURadio. Downgraded to 3.8 but still unable to build, this time further along in the make it crashes.  All dependencies are installed.

`                 from /home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.cc:1:
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.cc:1:
/home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.h: In member function ‘osmosdr::source::sptr Source::cast_to_osmo_sptr(gr::basic_block_sptr)’:
/home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.h:119:73: error: could not convert ‘boost::dynamic_pointer_cast(const boost::shared_ptr<U>&) [with T = osmosdr::source; U = gr::basic_block]()’ from ‘boost::shared_ptr<osmosdr::source>’ to ‘osmosdr::source::sptr’ {aka ‘std::shared_ptr<osmosdr::source>’}
  119 |     return boost::dynamic_pointer_cast<osmosdr::source, gr::basic_block>(p);
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
      |                                                                         |
      |                                                                         boost::shared_ptr<osmosdr::source>
/home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.cc: In constructor ‘Source::Source(double, double, double, std::string, std::string, Config*)’:
/home/bijan/Downloads/trunk-recorder-4.0-beta-3/trunk-recorder/source.cc:564:20: error: no match for ‘operator=’ (operand types are ‘gr::basic_block_sptr’ {aka ‘boost::shared_ptr<gr::basic_block>’} and ‘osmosdr::source::sptr’ {aka ‘std::shared_ptr<osmosdr::source>’})
  564 |     source_block = osmo_src;
      |                    ^~~~~~~~
In file included from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/format/alt_sstream.hpp:22,
                 from /usr/include/boost/format/internals.hpp:24,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/boost/property_tree/json_parser/detail/parser.hpp:8,
`

I've searched but I can't find anything. Any ideas?

robotastic, 2021-09-08:
I think I have some homework to do - This sounds related to #459 . I think I have to setup a few different Dockerfile to try out builds on different OSs.

robotastic, 2021-09-15:
I have had a chance to test with gnuradio 3.9 yet, but I did improve the compile process for Ubuntu 21.04. Can you give the branch `4.0-beta` a try?

toddejohnson, 2021-09-18:
There is an aur for this: https://aur.archlinux.org/packages/trunk-recorder-git

Although I haven't been able to get it working just yet it is building

robotastic, 2021-09-21:
This should be fixed on the `4.0-beta` branch. Going to close for now, but please re-open if new problems crop up.

closed 2021-09-21 by robotastic