TrunkRecorder/trunk-recorder #512: [Idea] Make a Community Driven Config Repo

MaxwellDPS, 2021-09-11:
@robotastic I'm curious what you would think of setting up a repo where people can share Trunk recorder configs for their local sites, to create a scanner type database?

robotastic, 2021-09-13:
This is a great idea! To keep things simple, I will just start a Github Repo along with a description document that has a table explaining each config, along with a link to the file, and the RR entry. I will start doing that with all of the examples included in Trunk Recorder. People can make a Pull Request to add additional entries. I will also create a folder for help/upload scripts too.

Dygear, 2022-05-16:
@MaxwellDPS was this ever completed?

MaxwellDPS, 2022-05-27:
@Dygear sorry It took me a sec, but yest indeed! https://github.com/robotastic/trunk-recorder-configs/pull/1/files

closed 2022-05-27 by MaxwellDPS