TrunkRecorder/trunk-recorder #517: Docker 4.0-beta libopenmhz_uploader.so needed in /app

toddejohnson, 2021-09-18:
For some reason docker 4.0-beta doesn't use the lib but look for ./libopenmhz_uploader.so which happens to be /app so you have to copy the library for the plugin into your mounted /app folder to get running. 

```sh
docker run -it --privileged -e TZ="America/Chicago" --user "$(id -u):$(id -g)" -v /var/run/dbus:/var/run/dbus -v $(pwd):/app --tmpfs /app/media robotastic/trunk-recorder:4.0-beta
```

```
[2021-09-18 11:20:32.745437] (info)   Setting up plugin -  Name: openmhz_uploader        Library file: ./libopenmhz_uploader.so
[2021-09-18 11:20:32.745672] (error)   Failed parsing Config: boost::dll::shared_library::load() failed (dlerror system message: ./libopenmhz_uploader.so: cannot open shared object file: No such file or directory): Bad file descriptor
```

If I copy the modules from /src to /app it starts working:
```sh
docker run -it --privileged -e TZ="America/Chicago" --user "$(id -u):$(id -g)" -v /var/run/dbus:/var/run/dbus -v $(pwd):/app --tmpfs /app/media robotastic/trunk-recorder:4.0-beta /bin/bash
cp /src/libopenmhz_uploader.so /app/
exit
docker run -it --privileged -e TZ="America/Chicago" --user "$(id -u):$(id -g)" -v /var/run/dbus:/var/run/dbus -v $(pwd):/app --tmpfs /app/media robotastic/trunk-recorder:4.0-beta
```

It worked! Are plugins supposed to be optional? Is there a reason libopenmhz_uploader.so needs to be in pwd?

toddejohnson, 2021-09-19:
@robotastic Just wanted to mention that this is on the robotastic/trunk-recorder:4.0-beta image if that was hidden in my commands above.

robotastic, 2021-09-21:
Good catch - I switched it to be a `make install` so that the files get copied... going to go test now.

toddejohnson, 2021-09-21:
Just finished a `docker build` on 4.0-beta and it is working now. Looks like 9d5710b18a0992eece17204cc89412fdcfae9b26 fixed it.

closed 2021-09-21 by toddejohnson