TrunkRecorder/trunk-recorder #521: Failing to open RTL-SDR

KyleKotowick, 2021-09-27:
This is an odd one that, while maybe not directly related to trunk-recorder, hopefully someone here has experience with.

I'm using an RTL-SDR dongle. I have trunk-recorder installed as a `systemd` service, and it's giving this error:
```
Sep 27 18:01:31 recorder-user start.sh[928764]: Using device #0 Generic RTL2832U OEM
Sep 27 18:01:31 recorder-user start.sh[928764]: usb_open error -3
Sep 27 18:01:31 recorder-user start.sh[928764]: Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules
Sep 27 18:01:31 recorder-user start.sh[928764]: [2021-09-27 18:01:31.030381] (error)   Failed parsing Config: Failed to open rtlsdr device.
Sep 27 18:01:31 mit-ems-radio systemd[1]: radio-recorder.service: Main process exited, code=exited, status=1/FAILURE
```

As you can see, it's failing to open the device. Note that the user I'm logged into the device with (`my-user`) is **different** than the user that is being used to run trunk-recorder (`recorder-user`).

Now here's the weird part. I usually only access this machine through an SSH terminal (as `my-user`). If I log into the computer physically (using a monitor, starting the desktop GUI), still as `my-user`, then all of a sudden it starts working, and it's able to successfully open the RTL-SDR USB device. Note that other than logging in, I have done nothing; trunk-recorder is just a service that is restarting itself because it failed previously (still restarting itself as `recorder-user`).

If I reboot the machine, the issue returns, and it cannot open the RTL-SDR USB device until I log in physically again.

Any thoughts?

hayden-t, 2021-09-27:
this is a permissions error, i had it when trying to open sdr with new non privileged user i created. i just ended up running it as root which fixed it, but to get your new user working you might have to edit udev permissions.
http://www.pclinuxos.com/forum/index.php?topic=135714.0
https://www.kevinhooke.com/2014/07/26/installing-rtl-sdr-and-dump1090-on-a-raspberry-pi-to-receive-ads-b-signals/

KyleKotowick, 2021-09-29:
Yep, changing the udev rules from "0660" to "0666" appears to have worked. Thanks!

closed 2021-09-29 by KyleKotowick

hayden-t, 2021-09-29:
success ! :)