TrunkRecorder/trunk-recorder #522: Had a hair pulling bug that could not tune control channel when ever changing config.json from working.

hayden-t, 2021-09-30:
So Ive had a working config that ive adapted and evolved over time. As well changing some hardware, but everything still works fine. But now I want to take out a system or change a dongle and no matter what I try I cant tune the control channel anymore. Even stripping it right back to one control channel system and one dongle doent work. nothing works but by my current config and i cant work out why. 

So in last attempt I tri changing docker images from latest to edge and it fixes it, no problems everything i try works. I notice a new message in the logs

`Detected Kernel usbfs mmap() bug, falling back to buffers in userspace`

I wonder if this is related to my problem ....

robotastic, 2021-10-07:
Googling that error, it looks like it is related to the Pi and perhaps the firmware. Could you try making sure the firmware is up to date?
https://forums.raspberrypi.com/viewtopic.php?t=244735

robotastic, 2021-10-21:
Going to close this for now - Please reopen if you discover more.

closed 2021-10-21 by robotastic

hayden-t, 2021-12-13:
So I just dug myself out of the same problem again, my system had been running fine for since last post (2 months ago) but it ran out of disk space causing me to look into it, and even after cleaning the disk space same problem, the container running edge from october that was fine was now not finding the control channel.... i tried everything and gave up and downloaded  another image 3.3.7, which i think might have even been that same version that i got with edge tag at the time, i configured this and ran it and it worked first go with exactly the same config.json . so happy times. but i thought i wonder if i can compare them to find out what the difference was, so i tried running the old container again, And what do you know, its working again all of a sudden !! 

With once again showing the same message:

```
Allocating 15 zero-copy buffers
Detected Kernel usbfs mmap() bug, falling back to buffers in userspace
```

So maybe its a bug with docker or my system, who knows !?!?! i tried many restarts, dongle unplugs and check in desktop sdrtrunk. 

I guess ill just wait and see again....

hayden-t, 2022-02-20:
@robotastic can you please reopen this issues as i cant

hayden-t, 2022-02-20:
so this time i have compose going to make it easier to spin up and down containers. and removing the 3.3.7 container and recreating it gives me the "Detected Kernel usbfs mmap() bug, falling back to buffers in userspace" that was missing but it still wont work, but if i spin up a 3.3.6 it works ... both use the same compose .yaml and config.json i just change the docker image version, yet 3.3.7 had worked for about a month

hayden-t, 2022-02-20:
still troubles, i can remove a docker container and image and then get it all going again, and it works, reboot and it fails...

hayden-t, 2022-02-21:
just for reference the message i mention is generated by rtl-sdr lib in debian patch https://sources.debian.org/patches/rtl-sdr/0.6-1/0002-lib-Add-workaround-for-Linux-usbfs-mmap-bug.patch/

hayden-t, 2022-02-21:
so ive compiled from src (didnt take as long as id feared, on N2+ Odroid) and it is working again, im guessing its a docker problem, i wonder if theres some way to trigger that conditional "Detected Kernel usbfs mmap() bug, falling back to buffers in userspace" code ???

hayden-t, 2022-02-21:
nope same problem, i reboot (via cli) and it wontn work again, missing the kernel usbfs error

hayden-t, 2022-02-22:
well, knock on wood, it seems ok now. i wiped and installed a new debian 11 image with a newer kernel than b4.