TrunkRecorder/trunk-recorder #527: Arch Linux AUR package Segfault with P25

toddejohnson, 2021-10-11:
I'm trying to get this packaged in the AUR to help simplify use in Arch Linux. I've made both [trunk-recorder](https://aur.archlinux.org/packages/trunk-recorder) and [trunk-recorder-git](https://aur.archlinux.org/packages/trunk-recorder-git).

The PKGBUILD is similar to a Dockerfile and are at https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=trunk-recorder

When I run this with robotastick/trunk-recorder:4.0 it works. When I run with your [Arch Dockerfile](https://github.com/robotastic/trunk-recorder/blob/master/utils/Dockerfile.arch-latest.dev) it also works.  When I run with [my AUR package](https://github.com/robotastic/trunk-recorder/blob/master/utils/Dockerfile.arch-latest-aur.dev) it doesn't work. When debugging with gdb I get:

```
[New Thread 0x7fffe7ebf640 (LWP 2818463)]
[New Thread 0x7fffe76be640 (LWP 2818464)]
[New Thread 0x7fffe6ebd640 (LWP 2818465)]
[New Thread 0x7fffe66bc640 (LWP 2818466)]
[New Thread 0x7fffe5ebb640 (LWP 2818467)]
Allocating 15 zero-copy buffers
[New Thread 0x7fffe56ba640 (LWP 2818468)]
[New Thread 0x7fffe4c79640 (LWP 2818469)]
[New Thread 0x7fffc7fff640 (LWP 2818470)]
[New Thread 0x7fffc77fe640 (LWP 2818471)]
[New Thread 0x7fffc6ffd640 (LWP 2818472)]
[New Thread 0x7fffc67fc640 (LWP 2818473)]
[New Thread 0x7fffc5ffb640 (LWP 2818474)]
[New Thread 0x7fffc57fa640 (LWP 2818475)]
[New Thread 0x7fffc4ff9640 (LWP 2818476)]
[New Thread 0x7fffa7fff640 (LWP 2818477)]
[New Thread 0x7fffa77fe640 (LWP 2818478)]
[New Thread 0x7fffa6ffd640 (LWP 2818479)]
[New Thread 0x7fffa67fc640 (LWP 2818480)]
[New Thread 0x7fffa5ffb640 (LWP 2818481)]
[New Thread 0x7fffa57fa640 (LWP 2818482)]
[New Thread 0x7fffa4ff9640 (LWP 2818483)]
... A bunch of these new threads...
[New Thread 0x7ffe1fd8f640 (LWP 2819122)] 
[New Thread 0x7ffe1f58e640 (LWP 2819123)]                                 
[New Thread 0x7ffe1ed8d640 (LWP 2819124)]             
[New Thread 0x7ffe1e58c640 (LWP 2819125)]
[New Thread 0x7ffe1dd8b640 (LWP 2819126)]
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/11.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::referen
ce = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Thread 196 "fsk4_demod_ff58" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff08f61640 (LWP 2818656)]
0x00007ffff6008d22 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff6008d22 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff5ff2862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6a920ba in ?? () from /usr/lib/trunk-recorder/libgnuradio-op25_repeater.so
#3  0x00007ffff6aa37f9 in ?? () from /usr/lib/trunk-recorder/libgnuradio-op25_repeater.so
#4  0x00007ffff6c231d6 in gr::block_executor::run_one_iteration() () from /usr/lib/libgnuradio-runtime.so.3.9.2
#5  0x00007ffff6c7f2d5 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /usr/lib/libgnuradio-runtime.so.3.9.2
#6  0x00007ffff6c72f6d in ?? () from /usr/lib/libgnuradio-runtime.so.3.9.2
#7  0x00007ffff6c73208 in ?? () from /usr/lib/libgnuradio-runtime.so.3.9.2
#8  0x00007ffff6580c07 in ?? () from /usr/lib/libboost_thread.so.1.76.0
#9  0x00007ffff655e259 in start_thread () from /usr/lib/libpthread.so.0
#10 0x00007ffff60ca5e3 in clone () from /usr/lib/libc.so.6
(gdb) 
```

When I remove my P25 systems it seems to work although the analog systems aren't often used.  I tried changing fsk4 to qpsk might help but the same thing happens. 

This can be reproduced with:
```sh
docker build -t tr-arch-aur --build-arg AUR_PACKAGE=trunk-recorder -f utils/Dockerfile.arch-latest-aur.dev .
```

toddejohnson, 2021-10-12:
Finally!!

```gdb
Thread 46 "fsk4_demod_ff58" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff55ff3640 (LWP 9022)]
0x00007ffff602ed22 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff602ed22 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff6018862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff6ab2f9a in std::__replacement_assert (__file=__file@entry=0x7ffff6b031b8 "/usr/include/c++/11.1.0/bits/stl_vector.h", __line=__line@entry=1045, 
    __function=__function@entry=0x7ffff6b030b0 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&;"..., __condition=__condition@entry=0x7ffff6b03040 "__n < this->size()") at /usr/include/c++/11.1.0/x86_64-pc-linux-gnu/bits/c++config.h:504
#3  0x00007ffff6ac2c09 in std::vector<int, std::allocator<int> >::operator[] (this=<optimized out>, __n=<optimized out>) at /usr/include/c++/11.1.0/bits/stl_vector.h:1045
#4  std::vector<int, std::allocator<int> >::operator[] (__n=<optimized out>, this=<optimized out>) at /usr/include/c++/11.1.0/bits/stl_vector.h:1043
#5  gr::op25_repeater::fsk4_demod_ff_impl::forecast (this=<optimized out>, noutput_items=<optimized out>, ninput_items_required=...) at /usr/src/debug/trunk-recorder-4.0/lib/op25_repeater/lib/fsk4_demod_ff_impl.cc:223
#6  0x00007ffff6c244f6 in gr::block_executor::run_one_iteration() () from /usr/lib/libgnuradio-runtime.so.3.9.3
#7  0x00007ffff6c80fa5 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /usr/lib/libgnuradio-runtime.so.3.9.3
#8  0x00007ffff6c7490d in ?? () from /usr/lib/libgnuradio-runtime.so.3.9.3
#9  0x00007ffff6c74ba8 in ?? () from /usr/lib/libgnuradio-runtime.so.3.9.3
#10 0x00007ffff65a6c07 in ?? () from /usr/lib/libboost_thread.so.1.76.0
#11 0x00007ffff6584259 in start_thread () from /usr/lib/libpthread.so.0
#12 0x00007ffff60f05e3 in clone () from /usr/lib/libc.so.6
```

robotastic, 2021-10-12:
It definitely has something to do with the P25_recorder. If I set the "digitalRecorders": 0, it will run fine and capture lots of trunking messages. However, it will crash right on startup, after the config has been read in, if digitalRecorders is greater than zero. This suggests it has something to do with the graph for the P25_recorder not being happy.

robotastic, 2021-10-12:
But... I am still baffled why this is only a problem in this one instance... and why it works fine when you use the Arch Docker image I built.

And just to confirm - I used the utils/Dockerfile.arch-latest.dev Dockerfile to create a container, then launched the container and built Trunk Recorder inside it, installed it and then ran it against the same config.json file and it ran fine, without crashing.

When I do it this way - the Gnuradio version is: 3.9.3.0, which looks to be the same version as the AUK image where it crashes.

At this point, I sort of want to say that it is probably not a gnuradio / Trunk Recorder issue. It must have something to do with the AUK based script for compiling / installing.

toddejohnson, 2021-10-12:
That is what has me puzzled too! You can use that same image and do:
```
git clone https://aur.archlinux.org/trunk-recorder.git # or https://aur.archlinux.org/trunk-recorder-git.git
cd trunk-recorder-git
makepkg -si
```
This will build and install the package. It doesn't like building as root so you might need to add a user or work with it a bit more. I'm confused how packaging this is causing these odd issues. I'd like to say I'm missing a dependency but haven't found it in ldd or other.

toddejohnson, 2021-10-15:
Interesting I can replace libgnuradio-op25_repeater.so with my non-packaged version and it starts working.

toddejohnson, 2021-10-15:
I think I found it!

> buildflags
> Allow the use of user-specific buildflags (CPPFLAGS, CFLAGS, CXXFLAGS, LDFLAGS) during build() as specified in makepkg.conf(5). More useful in its negative form !buildflags with select packages that have problems building with custom buildflags.

So I set !buildflags and it appears to be working.

robotastic, 2021-10-15:
Jackpot! 🏆
If you could put together a PR, that would be great. I would help make sure I don't screw it up. I think it is better to leave the debug flag off for the production version because performance is supposedly a little better because there is some optimization. I think it is possible to use a CMAKE flag to support different **debug** and **release** versions.

... wait, gotcha, there would be 2 different AURs? trunk-recorder-git & trunk-recorder ?  That sounds perfect.