TrunkRecorder/trunk-recorder #535: v4.0 - Duplicate Calls on Smartnet

JayManCanada6014, 2021-10-21:
Ever since I went to 4.0 last week, I am getting a lot of duplicate calls being uploaded to my Rdio-Scanner.
I even pulled the most recent copy of trunk this morning with the same issue.    When there is a lot of traffic on the system seems to be when it happens.

This is a Smartnet system with both Analog and Digital TG's.

Could this be the new system where chunks of calls are merged before upload?

JayManCanada6014, 2021-10-22:
I do everything from the GitHub. I am going to look at my archives, I keep everything and I am going to see if I can track down if any are dups in there.

JayManCanada6014, 2021-10-22:
So I was listening and found a spot when it happened. Now there is no duplicate .wav files and I can't really see that it did two of them in the logs.

Oct 21 22:14:05 trunknew Rdio Scanner[146]: NewCall: system=100 talkgroup=8704 file=8704-1634868829_141375000-call_125.m4a Success
Oct 21 22:14:07 trunknew Rdio Scanner[146]: NewCall: system=100 talkgroup=8800 file=17344-1634868836_142005000-call_128.m4a Success

8704 is patched into 17344
All the calls are showing up on trunk as 8704 but it got 17344 from someplace.   I don't save the .json files but is it possible it isn't doing the patches right?

And do you need me to do anything?

robotastic, 2021-10-24:
Ah - This looks like it is Phase 2 audio... it is possible that something could be screwed up for Phase 2.

devicenull, 2021-11-22:
I'm seeing this with the op25-update branch still - https://gist.githubusercontent.com/devicenull/0dc97a3dc9fae35f8c6a871895c27e4f/raw/6adf4f101a33709933086b2f410f654cfd53c7c0/gistfile1.txt

Overlapping audio files are here -https://www.dropbox.com/sh/d44586ydo5da5cq/AABv7rbZ5RM24Tea8QkgoBC8a?dl=0

closed 2024-12-14 by robotastic