TrunkRecorder/trunk-recorder #543: Phase 2 TDMA Control Channel support?

lwvmobile, 2021-11-03:
Hello,

just writing to inquire whether or not trunk-recorder currently supports Phase 2 TDMA control channels? I have adequate signal for the control channel, yet my terminal just scrolls 

[2021-11-03 15:31:07.003310] (error)   [duke]	There is only one control channel defined
[2021-11-03 15:31:07.003373] (error)   [duke]	 Control Channel Message Decode Rate: 1/sec, count:  3

System I am attempting to monitor is:
https://www.radioreference.com/apps/db/?sid=11618

Thanks.

gofaster, 2021-11-03:
Hi. Add the other 2 alternate control channels for that site you are monitoring.
i.e.: "control_channels": [855362500, 855937500, 858287500],

You are getting low decode rates.  Either PPM correction is not close enough, low signal, or too much signal swamping the front end (your gain setting is maxed out)
Also, modulation should be "qpsk"

robotastic, 2021-11-06:
It should work for P25 Phase 2 systems.... do you know if the control channel itself is broadcast using TDMA? My understanding that all P25 systems, Phase 1 and Phase 2 both used the full channel.

I would start by changing the modulation to `qpsk` and double check the error correction. 2.0 seems pretty high. It might be better to try using the `error` config setting instead and specify how far off the SDR is when you tune it to that freq.

robotastic, 2021-11-19:
Closing for now, feel free to reopen if you have more details.

closed 2021-11-19 by robotastic