TrunkRecorder/trunk-recorder #561: Build fails with error: could not convert 'gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::frame_assembler_impl]()' from 'std::shared_ptr<gr::op25_repeater::frame_assembler_impl>' to 'gr::op25_repeater::frame_assembler::sptr' {aka 'boost::shared_ptr<gr::op25_repeater::frame_assembler>'}

mldulaney, 2021-11-29:
/builddir/build/BUILD/trunk-recorder-4.1.0/lib/op25_repeater/lib/frame_assembler_impl.cc: In static member function 'static gr::op25_repeater::frame_assembler::sptr gr::op25_repeater::frame_assembler::make(int, const char*, int, int, gr::msg_queue::sptr)':
/builddir/build/BUILD/trunk-recorder-4.1.0/lib/op25_repeater/lib/frame_assembler_impl.cc:77:21: error: could not convert 'gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::frame_assembler_impl]()' from 'std::shared_ptr<gr::op25_repeater::frame_assembler_impl>' to 'gr::op25_repeater::frame_assembler::sptr' {aka 'boost::shared_ptr<gr::op25_repeater::frame_assembler>'}
   76 |                 return gnuradio::get_initial_sptr
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~
   77 |                     (new frame_assembler_impl(sys_num, options, debug, msgq_id, queue));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     std::shared_ptr<gr::op25_repeater::frame_assembler_impl>

JoeGilkey, 2021-11-30:
@mldulaney what version of GnuRadio is on your system?

mldulaney, 2021-11-30:
3.9.2.0-7.fc35

JoeGilkey, 2021-11-30:
Ah, try using 3.8.x instead. I don't think the OP25 code has been updated to support 3.9 yet.

robotastic, 2021-11-30:
@mldulaney I merged in a PR for Fedora 35 that has gnuradio 3.9 in it... give it a try and let me know if it works. Just `pull` the latest version of Trunk Recorder and rebuild.

toddejohnson, 2021-11-30:
I also got this when updating the AUR package for 4.1.0:
```
[ 30%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/frame_assembler_impl.cc.o
/build/trunk-recorder/src/trunk-recorder-4.1.0/lib/op25_repeater/lib/frame_assembler_impl.cc: In static member function ‘static gr::op25_repeater::frame_assembler::sptr gr::op25_repeater::frame_assembler::make(int, const char*, int, int, gr::msg_queue::sptr)’:
/build/trunk-recorder/src/trunk-recorder-4.1.0/lib/op25_repeater/lib/frame_assembler_impl.cc:77:21: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::op25_repeater::frame_assembler_impl]()’ from ‘std::shared_ptr<gr::op25_repeater::frame_assembler_impl>’ to ‘gr::op25_repeater::frame_assembler::sptr’ {aka ‘boost::shared_ptr<gr::op25_repeater::frame_assembler>’}
   76 |                 return gnuradio::get_initial_sptr
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~
   77 |                     (new frame_assembler_impl(sys_num, options, debug, msgq_id, queue));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     std::shared_ptr<gr::op25_repeater::frame_assembler_impl>
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:174: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/frame_assembler_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:260: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

Compiling AUR trunk-recorder-git r1706.8072483 was successful this morning. Any chance you would be interested in a new release or could you point me to the commit/PR so I can make a patch that would allow 4.1.0 to compile with GR3.9?

toddejohnson, 2021-11-30:
Doh! Looks like #560 is what I'm looking for. to fix this?

toddejohnson, 2021-11-30:
For the [AUR package](https://aur.archlinux.org/packages/trunk-recorder) I stole [a patch](https://aur.archlinux.org/cgit/aur.git/tree/trunk-recorder-pr560.patch?h=trunk-recorder) from #560 that has resolved this issue for me if it helps. Excited to have 4.1.0!!

robotastic, 2021-12-03:
Cool! I will close for now. Let me know if something else breaks.

closed 2021-12-03 by robotastic

mldulaney, 2021-12-04:
Confirmed all is good.