TrunkRecorder/trunk-recorder #566: add audio type info to json

ZeroChaos-, 2021-12-01:
This probably seems very silly, but it would help a lot of the json included information on if the audio was analog or digital.  I have a pretty extensive filter set that I use from encode-upload.sh which helps a lot for analog and makes no sense at all for digital.  it would be nice if I could check one flag in the json to determine which filter set to apply when converting from wav to m4a

robotastic, 2021-12-03:
This seems doable - let me see if there is a good way to check the recorder type and write that to the json.

robotastic, 2021-12-15:
I added support for this in the latest push. It will output `audio_type` to the json and the options are analog, digital tdma, and digital

tadscottsmith, 2021-12-15:
I believe there also was a request from you to add the system short name to the JSON? @robotastic it seems like the info is already being populated into the call_info and just not output in the JSON. 

`json_file << "\"system\": \"" << call_info.short_name << "\",\n";`

ZeroChaos-, 2022-01-10:
Looks like this is all set based on https://github.com/robotastic/trunk-recorder/commit/5798d993afbdabdd49fbf4298d1d6477fecdd51e

closed 2022-01-10 by ZeroChaos-