TrunkRecorder/trunk-recorder #577: Segmentation fault (core dumped) after running a few minutes

Cellphone70, 2021-12-13:
I upgraded a to v4.2.0 this weekend on one of my trunk-recorder instances.  This is a fresh OS install of Ubuntu 20.04 on a Raspberry Pi 4 Model B with 4GB RAM.  I am monitoring a P25 system that is a mix of Phase 1 and Phase 2 using an Airspy Mini.

Trunk-recorder runs for a few minutes and then terminates with message:
Segmentation fault (core dumped)

Below are the last few lines before it terminated:
[2021-12-13 16:30:07.007732] (trace)   [phxrwc] 17C     TG:       2485  Freq: 7.716062e+08      Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2021-12-13 16:30:07.007920] (trace)   [phxrwc] 17C     TG:       2485  Freq: 7.716062e+08      Concluding Recorded Call - Last Update: 4s      Call Elapsed: 10
[2021-12-13 16:30:07.008032] (info)   [phxrwc]  17C     TG:       2485  Freq: 7.716062e+08      - Transmission src: 8150787 pos: 0 length: 3.48
[2021-12-13 16:30:07.600835] (info)   [phxrwc]  17C     TG:       2485  Freq: 7.71606e+08       OpenMHz Upload Success - file size: 24112
[2021-12-13 16:30:08.463617] (info)   [phxrwc]  TG: 2485        Freq: 7.716062e+08      Broadcastify Upload Success - file size: 24112
[2021-12-13 16:30:08.974298] (info)   [phxrwc]  18C     TG:       2498  Freq: 7.711062e+08      Starting new Transmission       Src ID:  8150906
[2021-12-13 16:30:08.993125] (info)   [phxrwc]  23C     TG:       2963  Freq: 7.696062e+08      Not Recording: ENCRYPTED
[2021-12-13 16:30:09.899490] (info)   [phxrwc]  24C     TG:       2996  Freq: 7.716062e+08      Starting P25 Recorder Num [0]   TDMA: true      Slot: 1 QPSK: true
[2021-12-13 16:30:09.961866] (info)   [phxrwc]  25C     TG:       2599  Freq: 7.711062e+08      Not Recording: ENCRYPTED
[2021-12-13 16:30:10.300002] (info)   [phxrwc]  24C     TG:       2996  Freq: 7.716062e+08      Starting new Transmission       Src ID:  200004
[2021-12-13 16:30:12.007688] (trace)   [phxrwc] 19C     TG:       1795  Freq: 7.708562e+08      Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2021-12-13 16:30:12.008149] (trace)   [phxrwc] 19C     TG:       1795  Freq: 7.708562e+08      Concluding Recorded Call - Last Update: 4s      Call Elapsed: 10
[2021-12-13 16:30:12.008356] (info)   [phxrwc]  19C     TG:       1795  Freq: 7.708562e+08      - Transmission src: 100026 pos: 0 length: 5.22
[2021-12-13 16:30:12.651256] (info)   [phxrwc]  19C     TG:       1795  Freq: 7.70856e+08       OpenMHz Upload Success - file size: 34180
[2021-12-13 16:30:13.587218] (info)   [phxrwc]  TG: 1795        Freq: 7.708562e+08      Broadcastify Upload Success - file size: 34180
[2021-12-13 16:30:13.859446] (info)   [phxrwc]  24C     TG:       2996  Freq: 7.716062e+08      Starting new Transmission       Src ID:  200004
Segmentation fault (core dumped)

Please let me know if there is additional logging that I can enable, or if there are log files that I can post that will assist in determining cause.

robotastic, 2021-12-14:
It should be good now - can you try pulling that latest version?

Cellphone70, 2021-12-14:
Thanks for your quick response.  It has now been running for about 10 hours without issue.  Keep up the great work!

Cellphone70, 2021-12-15:
Ok, the issue is still occurring.  It stopped 3 times yesterday.  Please take another look.

Cellphone70, 2021-12-15:
It crashed again after running a few minutes today.  I'm not familiar with how to debug.  I have tried gdb based on other threads, but the core file is never created.  Please reply with some instructions on how to gather useful debug logs.  Thanks!

robotastic, 2021-12-15:
yea - if you are able to grab a core when it crashes, that would be really helpful. run `ulimit -c unlimited` then you can do `gdb recorder core` or maybe it is `gdb core recorder`. Once it starts, type in `bt`

Some timee things get messed up with the build though. Could you try create a different build directory, moving the config file there and then doing  cmake & make?

gonzoblue, 2021-12-16:
This is also happening to me after updating to the latest yesterday. Do you have better instructions on how to get the core dump? I've never done it before.

```
[2021-12-15 16:49:40.845972] (info)   [SD700]   TG: 91  Freq: 7.710562e+08      Broadcastify Upload Success - file size: 38820
[2021-12-15 16:49:40.923241] (info)   [SD700]   TG: 627 Freq: 7.697562e+08      Broadcastify Upload Success - file size: 49632
[2021-12-15 16:49:41.757506] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    Starting new Transmission       Src ID:  45809
[2021-12-15 16:49:48.008998] (trace)   [SD700]  16C     TG:        101 (             WES DISP 1)        Freq: 770.768750 MHz    Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2021-12-15 16:49:48.009057] (trace)   [SD700]  16C     TG:        101 (             WES DISP 1)        Freq: 770.768750 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 11
[2021-12-15 16:49:48.009097] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    - Transmission src: 45809 pos: 0 length: 2.16
[2021-12-15 16:49:48.009120] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    - Transmission src: 39016 pos: 2.16 length: 1.62
[2021-12-15 16:49:48.463151] (info)   [SD700]   18C     TG:         91 (             EAS DISP 1)        Freq: 770.518750 MHz    Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 QPSK: true
[2021-12-15 16:49:48.796735] (info)   [SD700]   19C     TG:        467 (                 CMD 7E)        Freq: 771.068750 MHz    Starting P25 Recorder Num [1]   TDMA: false     Slot: 0 QPSK: true
[2021-12-15 16:49:48.835140] (info)   [SD700]   TG: 101 Freq: 7.707688e+08      Broadcastify Upload Success - file size: 25672
[2021-12-15 16:49:49.047958] (info)   [SD700]   18C     TG:         91  Freq: 770.518750 MHz    Starting new Transmission       Src ID:  61944
[2021-12-15 16:49:49.162824] (info)   [SD700]   19C     TG:        467  Freq: 771.068750 MHz    Starting new Transmission       Src ID:  39046
Segmentation fault (core dumped)

```

robotastic, 2021-12-16:
Ok - I found atleast one place where it will Seg Fault. Basically, there was a spot where it was assuming a talkgroup was there without checking. The change should be in the latest version. Please give it a try and let me know. I will also work on getting documentation up on how to capture a core dump.

gonzoblue, 2021-12-16:
Testing now, been running a few minutes OK. Thanks for all the work! I'll try to rustle up a donation for ya...

Cellphone70, 2021-12-17:
I don't want to jinx it, but I have been running v4.2.2 for almost 22 hours without issue.  Thanks again for your quick help!

robotastic, 2021-12-17:
Awesome!! I am going to close it for now, but reopen if anything pops up

closed 2021-12-17 by robotastic