TrunkRecorder/trunk-recorder #578: --version option or alternative

hayden-t, 2021-12-13:
It would be create be able to tell what version or git commit your code/binary came from...?

robotastic, 2021-12-14:
This is a good idea - I feel like there is a way to bake the latest git commit hash into the CMake as an environment variable. I will have to look this up. I am pretty bad about remembering to rev the version number, so the git commit is prob better. I will try to look into it.

dreinhold, 2021-12-14:
@robotastic I had added this into dsd a long time ago https://github.com/szechyjs/dsd/commit/e40e89ae816beef82d4fcb5fe885b0c4c4f890b5
Might have some clues you can use...

hayden-t, 2021-12-14:
excellent ( )

taclane, 2025-11-13:
A `--version` option was added as of #688 to display the current version and latest git commit.