TrunkRecorder/trunk-recorder #581: Segfault in Talkgroup::get_priority

patrickfish, 2021-12-15:
```
Thread 1 "trunk-recorder" received signal SIGSEGV, Segmentation fault.
0x0000555555604b00 in Talkgroup::get_priority() ()
(gdb) bt
#0  0x0000555555604b00 in Talkgroup::get_priority() ()
#1  0x00005555555867ac in start_recorder(Call*, TrunkMessage, System*) ()
#2  0x0000555555587840 in handle_call(TrunkMessage, System*) ()
#3  0x0000555555587ba0 in handle_message(std::vector<TrunkMessage, std::allocator<TrunkMessage> >, System*) ()
#4  0x00005555555880f8 in monitor_messages() ()
#5  0x000055555557f7c4 in main ()
(gdb)
```

patrickfish, 2021-12-15:
On branch master
Your branch is up to date with 'origin/master'.

gonzoblue, 2021-12-16:
Unsure if related, but after updating to the latest yesterday I'm also gettin segfaults. Before the update this ran perfectly for weeks on end, now it won't stay up for a few minutes or hours.

```
[2021-12-15 16:49:40.845972] (info)   [SD700]   TG: 91  Freq: 7.710562e+08      Broadcastify Upload Success - file size: 38820
[2021-12-15 16:49:40.923241] (info)   [SD700]   TG: 627 Freq: 7.697562e+08      Broadcastify Upload Success - file size: 49632
[2021-12-15 16:49:41.757506] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    Starting new Transmission       Src ID:  45809
[2021-12-15 16:49:48.008998] (trace)   [SD700]  16C     TG:        101 (             WES DISP 1)        Freq: 770.768750 MHz    Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2021-12-15 16:49:48.009057] (trace)   [SD700]  16C     TG:        101 (             WES DISP 1)        Freq: 770.768750 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 11
[2021-12-15 16:49:48.009097] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    - Transmission src: 45809 pos: 0 length: 2.16
[2021-12-15 16:49:48.009120] (info)   [SD700]   16C     TG:        101  Freq: 770.768750 MHz    - Transmission src: 39016 pos: 2.16 length: 1.62
[2021-12-15 16:49:48.463151] (info)   [SD700]   18C     TG:         91 (             EAS DISP 1)        Freq: 770.518750 MHz    Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 QPSK: true
[2021-12-15 16:49:48.796735] (info)   [SD700]   19C     TG:        467 (                 CMD 7E)        Freq: 771.068750 MHz    Starting P25 Recorder Num [1]   TDMA: false     Slot: 0 QPSK: true
[2021-12-15 16:49:48.835140] (info)   [SD700]   TG: 101 Freq: 7.707688e+08      Broadcastify Upload Success - file size: 25672
[2021-12-15 16:49:49.047958] (info)   [SD700]   18C     TG:         91  Freq: 770.518750 MHz    Starting new Transmission       Src ID:  61944
[2021-12-15 16:49:49.162824] (info)   [SD700]   19C     TG:        467  Freq: 771.068750 MHz    Starting new Transmission       Src ID:  39046
Segmentation fault (core dumped)

```

robotastic, 2021-12-16:
Sorry about that - Just fixed it... should be good now.

patrickfish, 2021-12-16:
Confirmed - looking good now.

closed 2021-12-16 by patrickfish