TrunkRecorder/trunk-recorder #582: systems/system.cc: expected primary-expression before auto

toddejohnson, 2021-12-16:
I was starting to update to 4.2.2 for the AUR package and found that it wouldn't make:
```
[  7%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/systems/system.cc.o
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc: In member function ‘std::vector<long int> System::get_talkgroup_patch(long int)’:
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:458:18: error: expected primary-expression before ‘auto’
  458 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:458:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  458 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc: In member function ‘void System::update_active_talkgroup_patches(TrunkMessage)’:
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:472:18: error: expected primary-expression before ‘auto’
  472 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:472:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  472 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc: In member function ‘void System::clear_stale_talkgroup_patches()’:
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:496:18: error: expected primary-expression before ‘auto’
  496 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:496:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  496 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:513:17: error: expected primary-expression before ‘auto’
  513 |   BOOST_FOREACH(auto& stale_patch, stale_patches){
      |                 ^~~~
/build/trunk-recorder/src/trunk-recorder-4.2.0/trunk-recorder/systems/system.cc:520:18: error: expected primary-expression before ‘auto’
  520 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:216: CMakeFiles/trunk_recorder_library.dir/trunk-recorder/systems/system.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:243: CMakeFiles/trunk_recorder_library.dir/all] Error 2
```

So I tried trunk-recorder-git with similar results:
```
[  7%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/systems/system.cc.o
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc: In member function ‘std::vector<long int> System::get_talkgroup_patch(long int)’:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:459:18: error: expected primary-expression before ‘auto’
  459 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:459:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  459 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc: In member function ‘void System::update_active_talkgroup_patches(TrunkMessage)’:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:474:18: error: expected primary-expression before ‘auto’
  474 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:474:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  474 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc: In member function ‘void System::clear_stale_talkgroup_patches()’:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:499:18: error: expected primary-expression before ‘auto’
  499 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:499:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  499 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:516:17: error: expected primary-expression before ‘auto’
  516 |   BOOST_FOREACH(auto& stale_patch, stale_patches){
      |                 ^~~~
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/systems/system.cc:523:18: error: expected primary-expression before ‘auto’
  523 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:216: CMakeFiles/trunk_recorder_library.dir/trunk-recorder/systems/system.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:243: CMakeFiles/trunk_recorder_library.dir/all] Error 2
```

I also tried master(ae3b649f569842de22f42c825ca4618a3e1d79be) in a arch linux container via util steps:
```
[  7%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/systems/system.cc.o
/src/trunk-recorder/systems/system.cc: In member function ‘std::vector<long int> System::get_talkgroup_patch(long int)’:
/src/trunk-recorder/systems/system.cc:459:18: error: expected primary-expression before ‘auto’
  459 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/src/trunk-recorder/systems/system.cc:459:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  459 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/src/trunk-recorder/systems/system.cc: In member function ‘void System::update_active_talkgroup_patches(TrunkMessage)’:
/src/trunk-recorder/systems/system.cc:474:18: error: expected primary-expression before ‘auto’
  474 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/src/trunk-recorder/systems/system.cc:474:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  474 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/src/trunk-recorder/systems/system.cc: In member function ‘void System::clear_stale_talkgroup_patches()’:
/src/trunk-recorder/systems/system.cc:499:18: error: expected primary-expression before ‘auto’
  499 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
/src/trunk-recorder/systems/system.cc:499:3: error: ‘BOOST_FOREACH’ was not declared in this scope; did you mean ‘BOOST_NO_HASH’?
  499 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |   ^~~~~~~~~~~~~
      |   BOOST_NO_HASH
/src/trunk-recorder/systems/system.cc:516:17: error: expected primary-expression before ‘auto’
  516 |   BOOST_FOREACH(auto& stale_patch, stale_patches){
      |                 ^~~~
/src/trunk-recorder/systems/system.cc:523:18: error: expected primary-expression before ‘auto’
  523 |   BOOST_FOREACH (auto& patch, talkgroup_patches) {
      |                  ^~~~
```

I'll upload the logs for trunk-recorder-git as that will be the easiest to capture and retest.

[trunk-recorder-git-r1764.ae3b649-1-x86_64-build.txt](https://github.com/robotastic/trunk-recorder/files/7724276/trunk-recorder-git-r1764.ae3b649-1-x86_64-build.txt)

toddejohnson, 2021-12-16:
As suggested by @tadscottsmith I applied the below and it complied:

```diff
diff --git a/trunk-recorder/systems/system.h b/trunk-recorder/systems/system.h
index baed673..9598018 100644
--- a/trunk-recorder/systems/system.h
+++ b/trunk-recorder/systems/system.h
@@ -8,6 +8,7 @@
 #include "p25_trunking.h"
 #include "parser.h"
 #include "smartnet_trunking.h"
+#include <boost/foreach.hpp>
 
 #ifdef __GNUC__
 #pragma GCC diagnostic push
```

robotastic, 2021-12-16:
Cool - just pushed this change

closed 2021-12-16 by robotastic