TrunkRecorder/trunk-recorder #583: Segmentation fault when running rdio-scanner plugin

gisforgirard, 2021-12-18:
I've been trying to use the rdio-scanner plugin but have reverted to using the rdio-scanner dirwatch method, because after a very short period when running the rdio-scanner plugin, I get a "Segmentation fault (core dumped)" error, with no further details available to troubleshoot or help debug. I'm pretty much positive it's the rdio-scanner plugin, because I haven't had the error before enabling it, and after disabling it I haven't had a single error since. If there is some way I can try and locate the source of this error, I would be more than willing to... it seems to be pretty consistently occurring for me recently, so I should be able to easily reproduce it if someone lets me know how to determine any specific details...

robotastic, 2021-12-18:
Thanks for reporting it - I just put together a guide on how to capture info about a crash. Could you give it a try? Let me know if sections aren't clear or don't work. It should give us some info to help figure out what is broken:
https://github.com/robotastic/trunk-recorder/blob/master/docs/DEBUG.md

gisforgirard, 2021-12-18:
gotcha, thanks for the quick reply, just recompiled and have it running now, I'll let you know as soon as I have something to work with, shouldn't be too long if it's anything like the last couple of days of troubleshooting maybe a couple of hours tops...unless I'm wrong and it was something else, lol

gisforgirard, 2021-12-19:
clearly I must have not been running the newer version that this was already fixed in because it hasn't happened again, oops, totally uncorrelated to the changes I made in my config apparently, I can re-open it if this happens again but I'm guessing it was fixed and I was not running the version I thought I was, my bad...

closed 2021-12-19 by gisforgirard

robotastic, 2021-12-19:
No problem! Thanks for reporting it and let me know if you do find something else.

gisforgirard, 2021-12-22:
i just had another crash but apparently didn't run **ulimit -c unlimited** beforehand so the output got truncated, the "core" file wasn't generated anywhere that I could tell either, I had to run **coredumpctl list**, find the crash with trunk-recorder, copy the PID column, and then run **coredumpctl dump <PID> >> core**, after which I could run **gdb trunk-recorder core**, but the output looked like this:
```
...
[New LWP 6278]
[New LWP 6282]
[New LWP 5873]
[New LWP 5874]
[New LWP 5875]
[New LWP 5869]
[New LWP 6258]

warning: Section `.reg-xstate/6332' in core file too small.
Cannot access memory at address 0x7fada48a8168
Cannot access memory at address 0x7fada48a8160
Failed to read a valid object file image from memory.
```
so hopefully it happens again and i can actually get a useful dump...

gisforgirard, 2021-12-23:
so this is weird, it crashed and didn't actually save a core dump?

```       Message: Process 383390 (trunk-recorder) of user 1000 dumped core.
                
                Stack trace of thread 383788:
                #0  0x00007efd00ee305c _ZN2gr13op25_repeater24p25_frame_assembler_impl12general_workEiRSt6vectorIiSaIiEERS2_IPKvSaIS7_EERS2_IPvSaISB_EE (libgnuradio-op25_repeater.so + 0x6b05c)
                #1  0x00007efcff7bc483 _ZN2gr14block_executor17run_one_iterationEv (libgnuradio-runtime.so.3.8.4 + 0x6b483)
                #2  0x00007efcff80e47b _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd47b)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383417:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383427:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383424:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383437:
                #0  0x00007efcff6cf7b1 futex_abstimed_wait_cancelable (libpthread.so.0 + 0x107b1)
                #1  0x00007efcff812180 _ZN5boost18condition_variable10timed_waitERNS_11unique_lockINS_5mutexEEERKNS_10posix_time5ptimeE (libgnuradio-runtime.so.3.8.4 + 0xc1180)
                #2  0x00007efcff80e7fc _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd7fc)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
[... cut ...]
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
Coredump entry has no core attached (neither internally in the journal nor externally on disk).
```

gisforgirard, 2022-01-01:
still trying to figure out why this is happening, this is like the only program that refuses to save core dumps...

```
       Message: Process 383390 (trunk-recorder) of user 1000 dumped core.
                
                Stack trace of thread 383788:
                #0  0x00007efd00ee305c _ZN2gr13op25_repeater24p25_frame_assembler_impl12general_workEiRSt6vectorIiSaIiEERS2_IPKvSaIS7_EERS2_IPvSaISB_EE (libgnuradio-op25_repeater.so + 0x6b05c)
                #1  0x00007efcff7bc483 _ZN2gr14block_executor17run_one_iterationEv (libgnuradio-runtime.so.3.8.4 + 0x6b483)
                #2  0x00007efcff80e47b _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd47b)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383417:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383427:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383424:
                #0  0x00007efcff6cf376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007efcff7f2e33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007efcff80e62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 383437:
                #0  0x00007efcff6cf7b1 futex_abstimed_wait_cancelable (libpthread.so.0 + 0x107b1)
                #1  0x00007efcff812180 _ZN5boost18condition_variable10timed_waitERNS_11unique_lockINS_5mutexEEERKNS_10posix_time5ptimeE (libgnuradio-runtime.so.3.8.4 + 0xc1180)
                #2  0x00007efcff80e7fc _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd7fc)
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
[... cut ...]
                #3  0x00007efcff800eb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007efcff81a878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007efd00e2943b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007efcff6c8609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007efcff25d293 __clone (libc.so.6 + 0x122293)

Coredump entry has no core attached (neither internally in the journal nor externally on disk).
```

gisforgirard, 2022-01-05:
so despite having set `ulimit -c unlimited` it would appear that my `/etc/systemd/coredump.conf` file was causing the core files to not be generated:

> Jan 04 13:42:28 systemd-coredump[51553]: The core will not be stored: size 4389224448 is greater than 2147483648 (the configured maximum)

I even verified that the settings were correct via cat /proc/<PID>/limits:

> 
> Limit                     Soft Limit           Hard Limit           Units     
> Max cpu time              unlimited            unlimited            seconds   
> Max file size             unlimited            unlimited            bytes     
> Max data size             unlimited            unlimited            bytes     
> Max stack size            8388608              unlimited            bytes     
> Max core file size        unlimited            unlimited            bytes     
> Max resident set          unlimited            unlimited            bytes     
> Max processes             256443               256443               processes 
> Max open files            1024                 1048576              files     
> Max locked memory         67108864             67108864             bytes     
> Max address space         unlimited            unlimited            bytes     
> Max file locks            unlimited            unlimited            locks     
> Max pending signals       256443               256443               signals   
> Max msgqueue size         819200               819200               bytes     
> Max nice priority         0                    0                    
> Max realtime priority     0                    0                    
> Max realtime timeout      unlimited            unlimited            us        

changing the coredump.conf to use the following values appears to work when issuing `kill -s 11 <PID>` so I just have to wait for it to crash again and I can (hopefully) actually post something useful...

```
[Coredump]
ProcessSizeMax=8G
ExternalSizeMax=8G
JournalSizeMax=8G
```