TrunkRecorder/trunk-recorder #585: segfault in simplestream plugin

aaknitt, 2021-12-19:
I'm working on debugging this, just putting it here for awareness.  Something is messed up in the patch following in the simplestream plugin.  

```

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
--Type <RET> for more, q to quit, c to continue without paging--
Core was generated by `./trunk-recorder --config=config_waukesha.json'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb58a0a14 in std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/arm-linux-gnueabihf/libstdc++.so.6
[Current thread is 1 (Thread 0x648bdfc0 (LWP 22759))]
(gdb) bt full
#0  0xb58a0a14 in std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/arm-linux-gnueabihf/libstdc++.so.6
No symbol table info available.
#1  0xab945f30 in std::_Rb_tree<long, std::pair<long const, std::vector<long, std::allocator<long> > >, std::_Select1st<std::pair<long const, std::vector<long, std::allocator<long> > > >, std::less<long>, std::allocator<std::pair<long const, std::vector<long, std::allocator<long> > > > >::_M_erase_aux (
    this=0xaba52c7c <TGID_map>, __position=
    {first = 3, second = std::vector of length 1, capacity 1 = {58917}})
    at /usr/include/c++/10/bits/stl_tree.h:2515
        __y = 0x5faed78
#2  0xab943554 in std::_Rb_tree<long, std::pair<long const, std::vector<long, std::allocator<long> > >, std::_Select1st<std::pair<long const, std::vector<long, std::allocator<long> > > >, std::less<long>, std::allocator<std::pair<long const, std::vector<long, std::allocator<long> > > > >::_M_erase_aux (
    this=0xaba52c7c <TGID_map>, __first=
    {first = 4, second = std::vector of length 32, capacity 32 = {1, 99551932, 13, 775043377, 775435825, 825372209, 1931477049, 9123, -1557987326, -620648256, 0, 0, 0, 0, 0, 1, 2, 99551996, 13, 775043377, 775435825, 825372209, 57, 9124, -1541210110, -620648256, 0, 0, 0, 0, 0, 1}}, __last=
    {first = 4, second = std::vector of length 32, capacity 32 = {1, 99551932, 13, 775043377, 775435825, 825372209, 1931477049, 9123, -1557987326, -620648256, 0, 0, 0, 0, 0, 1, 2, 99551996, 13, 775043377, 775435825, 825372209, 57, 9124, -15--Type <RET> for more, q to quit, c to continue without paging--c
41210110, -620648256, 0, 0, 0, 0, 0, 1}}) at /usr/include/c++/10/bits/stl_tree.h:2531
No locals.
#3  0xab940444 in std::_Rb_tree<long, std::pair<long const, std::vector<long, std::allocator<long> > >, std::_Select1st<std::pair<long const, std::vector<long, std::allocator<long> > > >, std::less<long>, std::allocator<std::pair<long const, std::vector<long, std::allocator<long> > > > >::erase (this=0xaba52c7c <TGID_map>, __x=@0xa7a0ab00: 3) at /usr/include/c++/10/bits/stl_tree.h:2542
        __p = {first = {first = 3, second = std::vector of length 1, capacity 1 = {58917}}, second = {first = 4, second = std::vector of length 32, capacity 32 = {1, 99551932, 13, 775043377, 775435825, 825372209, 1931477049, 9123, -1557987326, -620648256, 0, 0, 0, 0, 0, 1, 2, 99551996, 13, 775043377, 775435825, 825372209, 57, 9124, -1541210110, -620648256, 0, 0, 0, 0, 0, 1}}}
        __old_size = 4
#4  0xab93ce9c in std::map<long, std::vector<long, std::allocator<long> >, std::less<long>, std::allocator<std::pair<long const, std::vector<long, std::allocator<long> > > > >::erase (this=0xaba52c7c <TGID_map>, __x=@0xa7a0ab00: 3) at /usr/include/c++/10/bits/stl_map.h:1069
No locals.
#5  0xab9385c0 in Simple_Stream::call_end (this=0x5eb8400, call_info=...) at /home/pi/trunk-recorder/plugins/simplestream/simplestream.cc:85
        recorder_id = @0xa7a0ab00: 3
        _foreach_continue83 = false
        _foreach_end83 = @0x648bb7f0: {<No data fields>}
        _foreach_cur83 = @0x648bb7f4: {<No data fields>}
        _foreach_col83 = @0x648bb7f8: {<No data fields>}
        talkgroup_num = 58917
        patched_talkgroups = std::vector of length 0, capacity 0
        recorders_to_erase = std::vector of length 1, capacity 1 = {3}
#6  0x0015971c in plugman_call_end (call_info=...) at /home/pi/trunk-recorder/trunk-recorder/plugin_manager/plugin_manager.cc:165
        plugin_error = 0
        plugin = 0x5eefe30
        it = 0x5eefe30
        total_error = 0
#7  0x001624b4 in upload_call_worker (call_info=...) at /home/pi/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:206
        result = 238
        error = 0
#8  0x0016fbdc in std::__invoke_impl<Call_Data_t, Call_Data_t (*)(Call_Data_t), Call_Data_t> (__f=@0x5f64548: 0x161d2c <upload_call_worker(Call_Data_t)>) at /usr/include/c++/10/bits/invoke.h:60
No locals.
#9  0x0016fac0 in std::__invoke<Call_Data_t (*)(Call_Data_t), Call_Data_t> (__fn=@0x5f64548: 0x161d2c <upload_call_worker(Call_Data_t)>) at /usr/include/c++/10/bits/invoke.h:96
No locals.
#10 0x0016f9c4 in std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >::_M_invoke<0u, 1u> (this=0x5f64100) at /usr/include/c++/10/thread:264
No locals.
#11 0x0016f854 in std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >::operator() (this=0x5f64100) at /usr/include/c++/10/thread:271
No locals.
#12 0x0016f580 in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Call_Data_t>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >, Call_Data_t>::operator() (this=0x648bd938) at /usr/include/c++/10/future:1350
No locals.
#13 0x0016f454 in std::__invoke_impl<std::unique_ptr<std::__future_base::_Result<Call_Data_t>, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Call_Data_t>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >, Call_Data_t>&> (__f=...) at /usr/include/c++/10/bits/invoke.h:60
No locals.
#14 0x0016f03c in std::__invoke_r<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Call_Data_t>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >, Call_Data_t>&> (__fn=...) at /usr/include/c++/10/bits/invoke.h:141
No locals.
#15 0x0016ed0c in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<Call_Data_t>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<Call_Data_t (*)(Call_Data_t), Call_Data_t> >, Call_Data_t> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/10/bits/std_function.h:292
No locals.
#16 0x001662cc in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const (this=0x648bd938) at /usr/include/c++/10/bits/std_function.h:622
No locals.
#17 0x001652b4 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (this=0x5f640e0, __f=0x648bd938, __did_set=0x648bd8e7) at /usr/include/c++/10/future:572
        __res = std::unique_ptr<std::__future_base::_Result_base> = {get() = 0x648bd900}
#18 0x0016a09c in std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__f=@0x648bd8e8: (void (std::__future_base::_State_baseV2::*)(std::__future_base::_State_baseV2 * const, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()> *, bool *)) 0x16528c <std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__futu [... cut ...]
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {-725848700, 95057544, 1686888384, -1094012692, 1686888384, 338, -1094012694, -1094012692, 1686887168, 1686887164, -1717986918, 1075157401, 0 <repeats 52 times>}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#35 0xb56fa208 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

aaknitt, 2021-12-19:
@robotastic I'm not certain yet, but I suspect that this could possibly be due to multiple threads calling the call_start() and call_end() functions in the plugin and therefore accessing the TGID_map in the plugin at the same time (not an issue in the main program since all modifications to the patch map take place in main and so shouldn't be conflicting):

https://stackoverflow.com/questions/8856604/c-map-erase-causes-unwanted-balancing

If two calls start and/or end at the same time I could see how this could happen.  If that does end up being the case, do you have any recommendations for how you handle this elsewhere in the program?  I'd like to keep a common approach and this isn't something I've dealt with in C++ before.

robotastic, 2021-12-20:
Hmm... yea, that could be a bit of a mess, because you don't want to be blocking in the call_end or call_start, because it will hold everything else up.

Do you need to track Patches? Also, you probably don't need to worry about using a Map. Using a vector or array should be fine. Should be easy to itterate since nothing will be too big.

aaknitt, 2021-12-20:
I'm not sure that blocking just for the very short amount of time it takes to modify the map (or vector, or array, or whatever) should be significant enough to impact much else.  

It's tracking patches to ensure that the audio from the desired talkgroup gets streamed even when that talkgroup is part of a patch (and not the supergroup).  But really it's not the patch tracking per se that's causing the issues, but just the tracking of which recorders should be streamed vs. which should be ignored.  That problem exists independent of the patch tracking.  It would be a little easier to manage if the recorder number were included in the call_info that gets passed to call_end(), but I don't think that solves the thread safety issue...I'd still need a vector to track which recorders should be streamed, and that vector could still be accessed my multiple threads at the same time.  

I'm still thinking a mutex may be the correct approach (testing it now) but open to alternatives.

robotastic, 2021-12-20:
Good point - that shouldn't take too long. A Mutex is probably a good call. That is what is used in the wav-file-sink.

Would it be possible to only record the Calls that have a recorder assigned? You could assume that anything that is being recorded is a useful call.

tadscottsmith, 2021-12-21:
I don't think it's related, but this plugin and and most of the patch group code has a type mismatch with talk group IDs. They're being written by the parser as unsigned longs and referenced everywhere else as signed longs. I've been testing a new branch for a few hours, so I'll put in a pull request in a bit.

tadscottsmith, 2021-12-21:
@aaknitt - would be curious if you're able to test the simplestream in #587? That's the only part I can't test on my system.

aaknitt, 2021-12-22:
@tadscottsmith  Yeah I had noticed that also and had planned on fixing in the PR I'll be submitting to fix the simplestream segfaults. I'll grab your changes and run it in my testing.

aaknitt, 2021-12-22:
> Would it be possible to only record the Calls that have a recorder assigned? You could assume that anything that is being recorded is a useful call.

For the purposes of this plugin I don't want to assume that everything being recorded by trunk-recorder should get streamed.  For example, I may want to send all traffic on the system to rdio-scanner or OpenMHz but only "live stream" a single talkgroup using the plugin.  I'm trying to support that kind of use case, which requires keeping track of which talkgroup is associated with each recorder.

Still playing whack-a-mole with this segfault...I thought the mutex would take care of it but still seeing issues.  I may have implemented the mutex incorrectly, but the line that the stack trace is now pointing to doesn't make sense...just a logging printout statement.  I'll keep after it.

closed 2022-02-17 by aaknitt