TrunkRecorder/trunk-recorder #586: Calling  recorder->get_num() on NULL recorder

tadscottsmith, 2021-12-20:
I'm running some static code analysis tools looking for errors causing the recent spike in segmentation faults. It appears this line is calling recorder->get_num() when we already know recorder is NULL.

https://github.com/robotastic/trunk-recorder/blob/e70ebc7f29c50aec005c4385dd8ba8362c72e36d/trunk-recorder/main.cc#L897

robotastic, 2021-12-20:
WHOAAA!!!! You are totally right. Let me go fix that. Also - what do you do for static analysis? I really have to upgrade my testing game.

robotastic, 2021-12-20:
Ok - I just patched that...

tadscottsmith, 2021-12-20:
I'm just diving in, but I've been testing out CLANG-TIDY against this codebase to try and find obvious errors. I've got a lot of output I could share, but a lot of it seems to just be "best practices" issues and I'm trying to weed out the highest priority issues first. Would be great to work through these and then have a test of the code run every time something is merged/updated.

robotastic, 2021-12-20:
Totally! It would be great to automate some testing as part of a release. If you get things going well enough, feel free to create a PR with a /test dir and some bash scripts with the correct commands to call.

It is probably best to ignore the OP25 code, unless there is a glaring bug. Keeping it close to the main branch makes it easier to merge in changes.

robotastic, 2021-12-20:
Also - I will close this for now, since this specific bug was fixed. Feel free to create new issues as you find things!

closed 2021-12-20 by robotastic