TrunkRecorder/trunk-recorder #589: Audio Issues with P25

JayManCanada6014, 2021-12-24:
Just putting this in here from the chat. I can confirmed I just downloaded and compiled mine and the same issue. I had to put my digital level from 12 to 1 to just get clear sound.

jhablitzel
@jhablitzel
Dec 23 11:04
Just upgraded from 4.0.3 to 4.2.2. Recording the Ohio MARCS system (P25 PH1). The newer version seems to have introduced chop or pulse into the audio effecting the overall voice quality in a negative way. This is especially noticeable when paging tones are playing, but you can hear the difference in the regular voice audio as well. Could this be a result of the change from float to int16? I can provide audio samples of both versions if needed.

Nick
@taclane
Dec 23 11:35
Are you using digitalLevels to increase the volume on your systems? Something about the math seems to have changed slightly during the recent update, and dropping digital levels a couple notches seems to solved audio quality issues for me and a few others around here.

jhablitzel
@jhablitzel
Dec 23 16:07
Yes I had to do that too. Was using 2, and started getting static/high levels. So dropped back to 1. Still getting the different sounding audio. Not super bad, but not as clear as it used to be.

JayMan
@JayManCanada_twitter
18:56
I normally had my digital set to 12 for my Smartnet system, I had to drop it to 1 to get clear audio.

taclane, 2021-12-25:
For what it's worth, the documentation for `digitalLevels` seems to imply it is an integer `number (1-16)`, but fractional values should be ok too.  If 1 is too quiet and 2 is starting to distort, 1.5 should be somewhere in-between.

JayManCanada6014, 2021-12-25:
> For what it's worth, the documentation for `digitalLevels` seems to imply it is an integer `number (1-16)`, but fractional values should be ok too. If 1 is too quiet and 2 is starting to distort, 1.5 should be somewhere in-between.

For me it starts to go sideways around 3,  even 2 sounds bad,  1.5 still too low the volume.

jhablitzel, 2021-12-25:
I am no longer getting distortion from the levels being too high once I reduced digitallevels to 1.  And I have good volume at that level.  However, that chop or pulse continues. 

It is almost like the levels on every other sample are slightly different, causing them to sound "bad" (not able to really describe it).

gisforgirard, 2021-12-25:
i use rdio-scanner and found the part where it executes ffmpeg and added audio normalization to fix the audio levels, maybe trunk-recorder could do something similar? https://ffmpeg.org/ffmpeg-filters.html#dynaudnorm

robotastic, 2021-12-28:
@gisforgirard - good point on having the file converter handle the autoleveling. It is already being run through `sox` which could also do the autoleveling. I will look and see if I can find  a good flag for that.

Also - with the shift to the latest version of OP25, the vocoder being used changed. I captured some of that here:
https://forums.radioreference.com/threads/op25-vocoder-difference.418063/

jhablitzel, 2021-12-30:
@robotastic I think you are correct on the change in the OP25 decoder being the issue.  The differences in the audio files in your post appear similar to what I'm seeing.

To clarify, I think the audio leveling issue is related, but not the entire problem.  I was able to get the audio level under control but changing digitalevels down to 1.  But at 1, it still peaks at 0db much of the time, so I could see where some systems could get distortion with digitallevels at minimum.  But to me the overarching problem is the difference in audio quality.  Just not as good as it used to be.

JayManCanada6014, 2021-12-30:
Is there a commit we can move back to that would revert this OP25 change?

chuot, 2022-01-07:
While this is not a real solution to the audio quality problem when recording from any type of P25 system, `sox --norm=-.01` seems to alleviate the problem. I also had to add it to combine_wav() since my instance runs with compressWav=false.

JayManCanada6014, 2022-01-14:
> While this is not a real solution to the audio quality problem when recording from any type of P25 system, `sox --norm=-.01` seems to alleviate the problem. I also had to add it to combine_wav() since my instance runs with compressWav=false.

Where exactly would I add  sox --norm=-.01

robotastic, 2022-01-21:
The `multi-vocoder` branch changes the Phase 1 Fullrate vocoder back to the original vocoder. It also switches the normalization early in the processing, so it is applied to the combined .wav file, and it uses --norm=-.01

Let me know if this sounds better and louder.

jhablitzel, 2022-01-21:
Sounds good as far as the audio quality.  Just like the old stuff I was
used to.

Not sure the normalization on the individual files is working.  See the
example files attached.  I looked at them with Adobe Audition (I do some
radio stuff and use that).  This is typical on the Sheriff TG I monitor.
Big difference between dispatcher and car, so this change will be big for
me.  The individual files didn't seem to normalize, and the final file is
louder than both of the individual files, so looks like it still did some
normalization when they were combined.  Not sure how that could happen, but
I'm about 99% positive I have the right code.  I compared the
call_concluder.cc code I downloaded and compiled against the code you had
shown as modified in the branch online and see the changes there. Are there
log files or something I can gather for you?

taclane, 2022-01-21:
> Big difference between dispatcher and car, so this change will be big for me. The individual files didn't seem to normalize, and the final file is louder than both of the individual files

Sox applies the normalization effect after it concatenates all the individual file, and it's a peak normalization, so the entire call will be normalized against the loudest transmission it captured.  I also noticed the same thing where there's always a big difference between some unit shouting into the mic on a windy day and the dispatcher in their quiet office.

There's nothing in the new or old code that normalizes each transmission individually, but if you want a quick preview of what it could sound like, add something like this
```
      char shell_command[4000];
      snprintf(shell_command, 4000, "sox %1$s %1$s.n.wav --norm=-.01; mv %1$s.n.wav %1$s", t.filename);
      int rc = system(shell_command);
```
to the bottom of the loop here, and it'll preprocess all the files before they are added together.

https://github.com/robotastic/trunk-recorder/blob/53a50fb3496544c80287b73cbab698d3cca6f1e4/trunk-recorder/call_concluder/call_concluder.cc#L165-L170

There's been some talk about adding normalization options to the config file someday.  It's possible that in the future, we could skip sox normalization and do a dynamic normalization through FFmpeg instead.

jhablitzel, 2022-01-21:
Thanks for the update.  I guess I misunderstood the change to
normalization.  I'll give the other code a try this weekend.

Would love to see some of this handled in configuration.  I'm doing a
combination of equalization/leveling in my upload script and liquidsoap to
try to level them on the two streams I generate to Broadcastify.  Still
have some transmissions that are so low you can't hear.  And those things
don't help at all with my Broadcastify upload.  Have considered not using
the built-in uploader and doing the upload through my script.  But not that
committed yet.

jhablitzel, 2022-01-28:
I tried putting the above code in to normalize the individual files before combining them.  It works, however, it seems that I get some very small files (.3 seconds) of very low level audio between nearly every transmission.  When it normalized these, they just became loud noise bursts between the transmissions.  Not very pleasing after awhile.  I'm not sure if these very small files are a by-product of the system I'm monitoring or something in trunk-recorder. 

I reverted the change and have been running the "stock" multi-vocoder branch for a week.  Sounds much better than the one running the newer P25.  But I still have the occasional transmission that is very low in volume.  There is one dispatcher in particular who is always 15db lower than everyone else.  No idea why.

robotastic, 2022-01-30:
Thanks for checking out normalizing individual files. That is a great find - files with noise in the will be jacked super high... I wonder if changing your squelch would help remove those small files. I will merge this in because it is atleast better than the current version.

robotastic, 2022-01-31:
I am going to close this for now

closed 2022-01-31 by robotastic