TrunkRecorder/trunk-recorder #590: Segfault in Talkgroup::get_priority

patrickfish, 2021-12-26:
Not sure if related to #585 since I'm not using simplestream, but I've been getting this every so often and finally caught it.

```
[Thread 0x7ffdba604700 (LWP 21387) exited]
[2021-12-25 17:40:48.009935] (error)   [system]       49784C Failed to conclude call - TG:       3847
        Sat Dec 25 17:37:26 2021 PST
[New Thread 0x7ffdba604700 (LWP 21446)]
[New Thread 0x7ffdb8600700 (LWP 21447)]
[Thread 0x7ffdb8600700 (LWP 21447) exited]

Thread 1 "trunk-recorder" received signal SIGSEGV, Segmentation fault.
0x000055555569afd2 in Talkgroup::get_priority (this=0x0)
    at /home/pfish/trunk-recorder4/trunk-recorder/talkgroup.cc:29
29        return priority;
(gdb)
(gdb)
(gdb)
(gdb)
(gdb) bt full
#0  0x000055555569afd2 in Talkgroup::get_priority (this=0x0) at /home/pfish/trunk-recorder4/trunk-recorder/talkgroup.cc:29
No locals.
#1  0x00005555555e0f66 in start_recorder (call=0x55555b251d10, message=..., sys=0x555559f99a80) at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:642
        TGID = @0x7fff18004ec0: 3009
        _foreach_continue641 = false
        _foreach_end641 = @0x7fffffffd098: {<No data fields>}
        _foreach_cur641 = @0x7fffffffd090: {<No data fields>}
        _foreach_col641 = @0x7fffffffd120: {<No data fields>}
        priority = 1
        source = 0x555559fb3270
        it = 0x555559fb3270
        talkgroup = 0x555559faa400
        source_found = true
        recorder_found = false
        recorder = 0x2d
        debug_recorder = 0x55555b251d10
        sigmf_recorder = 0x55555563629a <boost::algorithm::detail::is_classifiedF::~is_classifiedF()+28>
#2  0x00005555555e3b68 in handle_call (message=..., sys=0x555559f99a80) at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:962
        call = 0x55555b251d10
        call_found = false
        call_retune = false
        recording_started = false
#3  0x00005555555e3d94 in handle_message (messages=std::vector of length 1, capacity 1 = {...}, sys=0x555559f99a80) at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:977
        message = {message_type = GRANT, meta = "tbsk02\tMoto Patch Grant\tChannel ID:   433\tFreq: 853.712500 MHz\tsg    3857\tTDMA -1\tsa 9180352", freq = 853712500, talkgroup = 3857,
          encrypted = false, emergency = false, duplex = false, mode = false, priority = 0, tdma_slot = 0, phase2_tdma = false, source = 9180352, sys_num = 0, sys_id = 0, nac = 1532, wacn = 0,
          moto_patch_data = {sg = 93824993415160, ga1 = 140737488343600, ga2 = 140732724425392, ga3 = 0}}
        it =
    {message_type = GRANT, meta = "tbsk02\tMoto Patch Grant\tChannel ID:   433\tFreq: 853.712500 MHz\tsg    3857\tTDMA -1\tsa 9180352", freq = 853712500, talkgroup = 3857, encrypted = false, emergency = false, duplex = false, mode = false, priority = 0, tdma_slot = 0, phase2_tdma = false, source = 9180352, sys_num = 0, sys_id = 0, nac = 1532, wacn = 0, moto_patch_data = {sg = 93824993415160, ga1 = 140737488343600, ga2 = 140732724425392, ga3 = 0}}
#4  0x00005555555e5b33 in monitor_messages () at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:1177
        timeDiff = 1
        statusTimeDiff = 82
        msg = {px = 0x7ffee4090720, pn = {pi_ = 0x7ffee4090770}}
        sys_num = 0
        sys = 0x555559f99a80
        lastStatusTime = 1640482766
        lastMsgCountTime = 1640482847
        management_timestamp = 1640482848
        current_time = 1640482848
        trunk_messages = std::vector of length 1, capacity 1 = {{message_type = GRANT,
            meta = "tbsk02\tMoto Patch Grant\tChannel ID:   433\tFreq: 853.712500 MHz\tsg    3857\tTDMA -1\tsa 9180352", freq = 853712500, talkgroup = 3857, encrypted = false,
            emergency = false, duplex = false, mode = false, priority = 0, tdma_slot = 0, phase2_tdma = false, source = 9180352, sys_num = 0, sys_id = 0, nac = 1532, wacn = 0,
            moto_patch_data = {sg = 93824993415160, ga1 = 140737488343600, ga2 = 140732724425392, ga3 = 0}}}
#5  0x00005555555e844d in main (argc=1, argv=0x7fffffffde48) at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:1405
        desc = {static m_default_line_length = 80, m_caption = "Options", m_line_length = 80, m_min_description_length = 40, m_options = std::vector of length 2, capacity 2 = {{
              px = 0x555559f966a0, pn = {pi_ = 0x555559f962d0}}, {px = 0x555559f96850, pn = {pi_ = 0x555559f96e60}}}, belong_to_group = std::vector<bool> of length 2, capacity 64 = {0, 0},
          groups = std::vector of length 0, capacity 0}
        vm = <incomplete type>
        config_file = "./config.json"
        uri = "ws://localhost:3005"
(gdb)
```

patrickfish, 2021-12-26:
```#1  0x00005555555e0f66 in start_recorder (call=0x55555b251d10, message=..., sys=0x555559f99a80) at /home/pfish/trunk-recorder4/trunk-recorder/main.cc:642
        TGID = @0x7fff18004ec0: 3009```

3009 is not a talkgroup defined in my talkgroups CSV, however it is likely part of a patch with a TGID that is. Therefore...is it looking for the priority of the TGID if that was the "referenced" TGID? @aaknitt could this be related to your issue?

aaknitt, 2022-01-02:
@patrickfish this should be fixed in latest master branch code now.

patrickfish, 2022-01-02:
I've been running your PR for about a week now with no issues. Going to call this fixed via #591

closed 2022-01-02 by patrickfish