TrunkRecorder/trunk-recorder #592: Adjacent Channel Interference on Conventinal Channels

melbournefan, 2021-12-27:
There's a few 5w analog allocations I've got setup on my trunk-recorder instance that get 'bombed out' by a nearby 50w DMR system that's on a neighboring frequency which cause noise issues as well as 'birdies' to be recorded and shown in Rdio Scanner. 

I was wondering if there's any sort of bandwidth filtering that can be done in the config.json file so the receiving channel only captures that 12KHz that it's supposed to since I don't have the same issues in SDR# if I carefully tune the bandwidth of the receiving SDR, a feature such as that would help in receiving those lower powered channels that would otherwise be drowned out by the DMR system that's basically TX'ing 24/7.

Thanks

dtnetnt, 2023-03-15:
Did you ever get anywhere with this? I get the same thing with marine band VHF. Everything works perfectly until I get boat nearby transmit. Trunk-recorder picks up their transmission perfectly on the correct frequency but also picks up a false transmission which is just noise on the frequencies either side of the one they transmitted on. I've tried the gain on all sorts of different settings but to no avail. As you say if you could just capture the bandwidth either side of the required frequency it would be great.

frameshift18, 2023-03-15:
You could try playing with the maxDev setting. But it sounds more like your front end is getting overloaded.

ZeroChaos-, 2023-03-15:
I know we all want a software solution to this, but this isn't a terrible idea either: https://www.minicircuits.com/WebStore/dashboard.html?model=VLM-63-2W-S%2B

dtnetnt, 2023-03-15:
> You could try playing with the maxDev setting. But it sounds more like your front end is getting overloaded.

Thanks for the suggestion, I have looked at maxDev but I'm not really sure what it does, could you explain?

I think you're right about it being overloaded. A 25w transmission less thaan 0.25 miles away is bound to cause issues.

dtnetnt, 2023-03-15:
> I know we all want a software solution to this, but this isn't a terrible idea either: https://www.minicircuits.com/WebStore/dashboard.html?model=VLM-63-2W-S%2B

That's a really interesting suggestion, thank you.

So in simples terms would it limit the strength of the signal once it goes over a certain level? Would it affect weaker signals or would they pass through relatively unharmed? Thanks for your time.

ZeroChaos-, 2023-03-15:
Exact details on the page, but yes, and you only lose like 0.6 dBm from the normal signals

dtnetnt, 2023-03-15:
> Exact details on the page, but yes, and you only lose like 0.6 dBm from the normal signals

Brilliant, thank you for the speedy reply.