TrunkRecorder/trunk-recorder #596: duplicate call rejected sending to rdio-scanner

rabarar, 2022-01-04:
I just setup trunk-recorder and rdio-scanner this evening - but can't seem to get files to go in without being duplicated:

I basically get this for all files generated:
```

NewCall: system=411 talkgroup=11221 file=11221-1641272133_851687500-call_5.wav Duplicate call rejected
warning
01/03
23:55
NewCall: system=411 talkgroup=11221 file=11221-1641272125_858237500-call_4.wav Duplicate call rejected
warning
01/03
23:55
NewCall: system=411 talkgroup=11221 file=11221-1641272117_851962500-call_3.wav Duplicate call rejected
warning
01/03
23:55
NewCall: system=411 talkgroup=11221 file=11221-1641272109_851062500-call_2.wav Duplicate call rejected
warning
01/03
23:55
NewCall: system=411 talkgroup=11221 file=11221-1641272100_852800000-call_1.wav Duplicate call rejected
warning
01/03
23:49
NewCall: system=411 talkgroup=10901 file=10901-1641271771_853062500-call_28.wav Duplicate call rejected
warning
01/03
23:49
NewCall: system=411 talkgroup=10901 file=10901-1641271764_851575000-call_27.wav Duplicate call rejected
```

Not sure if it's a configuration issue - but when I manually run my upload script to push the files to rdio-scanner i goes up successfully.

rabarar, 2022-01-04:
here's the config.json file:

```
{
    "ver": 2,
    "captureDir":"/home/vagrant/audio",
    "frequencyFormat":"mhz",
    "logLevel":"info",
    "logFile":true,
    "audioStreaming":true,
    "controlWarnRate":0,
    "sources": [{
        "center": 855500000.0,
        "rate": 12500000,
        "error": 0,
        "gain": 16,
        "antenna": "RX2",
        "digitalRecorders": 4,
        "driver": "usrp"
    }],
    "systems": [{
        "shortName":"USRP-N210",
        "control_channels": [860237500, 858737500,859237500,859737500],
        "type": "p25",
        "modulation": "qpsk",
        "audioArchive": false,
        "transmissionArchive": false,
        "callLog":false,
        "compressWav":false
    }],
    "plugins":[{
        "name": "rdioscanner_uploader",
        "library":"librdioscanner_uploader.so",
        "uploadScript": "upload-call.sh 411",
        "server": "http://192.168.1.158:3001",
        "systems": [{
                "shortName":"USRP-N210",
                "apiKey":"XXXX",
                "systemId":411
        }]
    }]
}
```

rabarar, 2022-01-04:
UPDATE:

I see i've made an error in the config.json file on the key being in the wrong object: moving it correctly calls the upload-script - but the filename passed to it is incorrect:

```
[2022-01-04 10:51:09.452335] (info)   [USRP-N210]	2C	TG:      10901	Freq: 851.962500 MHz	Starting P25 Recorder Num [1]	TDMA: false	Slot: 0	QPSK: true
D[2022-01-04 10:51:09.605658] (info)   [USRP-N210]	2C	TG:      10901	Freq: 851.962500 MHz	Starting new Transmission 	Src ID:  2470508
[2022-01-04 10:51:12.859024] (info)   [USRP-N210]	0C	TG:      11203	Freq: 852.800000 MHz	Starting new Transmission 	Src ID:  2475529
[2022-01-04 10:51:16.337247] (info)   [USRP-N210]	3C	TG:      10901	Freq: 851.387500 MHz	TG not in Talkgroup File 
[2022-01-04 10:51:16.337316] (info)   [USRP-N210]	3C	TG:      10901	Freq: 851.387500 MHz	Starting P25 Recorder Num [2]	TDMA: false	Slot: 0	QPSK: true
Detected an UnSquelch, resetting the Costas Clock![2022-01-04 10:51:16.820346] (info)   [USRP-N210]	3C	TG:      10901	Freq: 851.387500 MHz	Starting new Transmission 	Src ID:  112
[2022-01-04 10:51:17.006589] (trace)   [USRP-N210]	2C	TG:      10901	Freq: 851.962500 MHz	Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2022-01-04 10:51:17.006679] (trace)   [USRP-N210]	2C	TG:      10901	Freq: 851.962500 MHz	Concluding Recorded Call - Last Update: 4s	Call Elapsed: 8
[2022-01-04 10:51:17.006702] (info)   [USRP-N210]	2C	TG:      10901	Freq: 851.962500 MHz	- Transmission src: 2470508 pos: 0 length: 2.52
[2022-01-04 10:51:17.014035] (info)   [USRP-N210]	2C 	 Running upload script: ./upload-call.sh 411 /home/vagrant/audio/USRP-N210/2022/1/4/10901-1641311469_851962500-call_2.wav /home/vagrant/audio/USRP-N210/2022/1/4/10901-1641311469_851962500-call_2.json /home/vagrant/audio/USRP-N210/2022/1/4/10901-1641311469_851962500-call_2.m4a
Call imported successfully.
[2022-01-04 10:51:17.111838] (error)   [USRP-N210]	2C	TG:      10901	Freq: 8.519625e+08	Rdio Scanner Upload Error: 
```

The filename has a suffice `-call_X` that is not in the generated file

rabarar, 2022-01-04:
UPDATE:

so I guess I misunderstood the plugin and the uploadScript for rdio-scanner - you only need one or the other? I removed the plugin, kept the uploadScrippt and it works - but i don't understand why the plugin fails (see above)

taclane, 2022-01-04:
It is correct that you use the rdio-scanner plugin or upload script, not both.  I'm suspecting your `rdioscanner_uploader failed` is due to the upload script is running first and then deleting the files before the plugin has a chance to go.  If for some reason you did want to do both, comment out the last line of the upload script, and then `audioArchive` in the config should clean up the leftover audio files after the plugin runs.

The plugin itself *mostly* works, but only with rdio-scanner 6.x.  Even then, it still has some quirks right now.  Rdio-scanner 6.1 should be on the horizon soonish; it's possible it may carry some upload fixes.

rabarar, 2022-01-04:
That sounds great… i threw in a sed rewrite on the filename and removed the plug-in config - works great!

closed 2022-01-04 by rabarar