TrunkRecorder/trunk-recorder #599: Installation Error

Dewey3, 2022-01-06:
I'm hoping this is just a burp in the internet connect, but multiple attempts to install the libraries for the Raspberry Pi are resulting in:

E: Unable to locate package libusb-1.0-0.dev
E: Couldn't find any package by glob 'libusb-1.0-0.dev'

promyvion, 2022-01-08:
Encountering the same issue! Did your period/hyphen swap idea pan out?

Dewey3, 2022-01-08:
> Encountering the same issue! Did your period/hyphen swap idea pan out?
> 
> EDIT: I tried it myself. Swapping a period for a hyphen let me install it just fine, so it appears the instructions are indeed out of date.

Yes, replace the period with a hyphen and the install will continue.

bctrainers, 2022-01-09:
> Encountering the same issue! Did your period/hyphen swap idea pan out?
> 
> EDIT: I tried it myself. Swapping a period for a hyphen let me install it just fine, so it appears the instructions are indeed out of date.

Which page did you use for install/compile guidance?

promyvion, 2022-01-09:
> > Encountering the same issue! Did your period/hyphen swap idea pan out?
> > EDIT: I tried it myself. Swapping a period for a hyphen let me install it just fine, so it appears the instructions are indeed out of date.
> 
> Which page did you use for install/compile guidance?

I was using the [Raspberry Pi instructions.](https://github.com/robotastic/trunk-recorder/blob/master/docs/INSTALL-PI.md)

robotastic, 2022-01-26:
Just fixed this instructions... sorry for the confusion!

closed 2022-01-26 by robotastic