TrunkRecorder/trunk-recorder #604: Segmentation fault

gisforgirard, 2022-01-10:
```
       Message: Process 157211 (trunk-recorder) of user 1000 dumped core.
                
                Stack trace of thread 157668:
                #0  0x00007fa43839b05c _ZN2gr13op25_repeater24p25_frame_assembler_impl12general_workEiRSt6vectorIiSaIiEERS2_IPKvSaIS7_EERS2_IPvSaISB_EE (libgnuradio-op25_repeater.so + 0x6b05c)
                #1  0x00007fa436c69483 _ZN2gr14block_executor17run_one_iterationEv (libgnuradio-runtime.so.3.8.4 + 0x6b483)
                #2  0x00007fa436cbb47b _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd47b)
                #3  0x00007fa436cadeb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007fa436cc7878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007fa4382e143b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007fa436b75609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007fa43670a293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 157211:
                #0  0x00007fa436b7c7b1 futex_abstimed_wait_cancelable (libpthread.so.0 + 0x107b1)
                #1  0x000055574c2ab4e5 _ZN5boost5posix22pthread_cond_timedwaitEP14pthread_cond_tP15pthread_mutex_tPK8timespec (trunk-recorder + 0xa84e5)
                #2  0x000055574c2ba758 _ZN5boost18condition_variable10timed_waitINS_9date_time18subsecond_durationINS_10posix_time13time_durationELl1000EEEPFbvEEEbRNS_11unique_lockINS_5mutexEEERKT_T0_ (trunk-recorder + 0xb7758)
                #3  0x000055574c2b2556 _ZN5boost11this_thread5sleepINS_9date_time18subsecond_durationINS_10posix_time13time_durationELl1000EEEEEvRKT_ (trunk-recorder + 0xaf556)
                #4  0x000055574c2a1828 _Z16monitor_messagesv (trunk-recorder + 0x9e828)
                #5  0x000055574c2a4157 main (trunk-recorder + 0xa1157)
                #6  0x00007fa43660f0b3 __libc_start_main (libc.so.6 + 0x270b3)
                #7  0x000055574c28bbbe _start (trunk-recorder + 0x88bbe)
                
                Stack trace of thread 157234:
                #0  0x00007fa4366fdaff __GI___poll (libc.so.6 + 0x115aff)
                #1  0x00007fa43646340d n/a (libusb-1.0.so.0 + 0xa40d)
                #2  0x00007fa43646465c libusb_handle_events_timeout_completed (libusb-1.0.so.0 + 0xb65c)
                #3  0x00007fa437cc9352 _ZN19libusb_session_impl25libusb_event_handler_taskEP14libusb_context (libuhd.so.3.15.0 + 0x971352)
                #4  0x00007fa437c96d4a _ZN9task_impl9task_loopERKN5boost8functionIFvvEEE (libuhd.so.3.15.0 + 0x93ed4a)
                #5  0x00007fa436a1ade4 n/a (libstdc++.so.6 + 0xd6de4)
                #6  0x00007fa436b75609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007fa43670a293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 157235:
                #0  0x00007fa436b7c7b1 futex_abstimed_wait_cancelable (libpthread.so.0 + 0x107b1)
                #1  0x00007fa437cc44af _ZN5boost18condition_variable10timed_waitI20lut_result_completedEEbRNS_11unique_lockINS_5mutexEEERKNS_10posix_time5ptimeET_ (libuhd.so.3.15.0 + 0x96c4af)
                #2  0x00007fa437cc5d80 _ZN21libusb_zero_copy_impl13get_recv_buffEd (libuhd.so.3.15.0 + 0x96dd80)
                #3  0x00007fa437c131dd _ZN9b200_impl17handle_async_taskEN5boost10shared_ptrIN3uhd9transport12zero_copy_ifEEENS1_INS_13AsyncTaskDataEEE (libuhd.so.3.15.0 + 0x8bb1dd)
                #4  0x00007fa437c05960 _ZN5boost6detail8function21function_obj_invoker0INS_3_bi6bind_tINS_8optionalISt4pairIjSt6vectorIhSaIhEEEEENS_4_mfi3mf2ISB_9b200_implNS_10shared_ptrIN3uhd9transport12zero_copy_ifEEENSF_INSE_13AsyncTaskDataEEEEENS3_5list3INS3_5valueIPSE_EENSO_ISJ_EENSO_ISL_EEEEEESB_E6invokeERNS1_15function_bufferE (libuhd.so.3.15.0 + 0x8ad960)
                #5  0x00007fa437c97449 _ZN13msg_task_impl9task_loopERKN5boost8functionIFNS0_8optionalISt4pairIjSt6vectorIhSaIhEEEEEvEEE (libuhd.so.3.15.0 + 0x93f449)
                #6  0x00007fa4382e143b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #7  0x00007fa436b75609 start_thread (libpthread.so.0 + 0x9609)
                #8  0x00007fa43670a293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 157239:
                #0  0x00007fa436b7c376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007fa436c9fe33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007fa436cbb62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007fa436cadeb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007fa436cc7878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007fa4382e143b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007fa436b75609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007fa43670a293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 157246:
                #0  0x00007fa436b7c376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007fa436c9fe33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007fa436cbb62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007fa436cadeb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007fa436cc7878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
                #5  0x00007fa4382e143b n/a (libboost_thread.so.1.71.0 + 0x1143b)
                #6  0x00007fa436b75609 start_thread (libpthread.so.0 + 0x9609)
                #7  0x00007fa43670a293 __clone (libc.so.6 + 0x122293)
                
                Stack trace of thread 157241:
                #0  0x00007fa436b7c376 futex_wait_cancelable (libpthread.so.0 + 0x10376)
                #1  0x00007fa436c9fe33 _ZN5boost18condition_variable4waitERNS_11unique_lockINS_5mutexEEE (libgnuradio-runtime.so.3.8.4 + 0xa1e33)
                #2  0x00007fa436cbb62d _ZN2gr15tpb_thread_bodyC2EN5boost10shared_ptrINS_5blockEEENS2_INS1_7barrierEEEi (libgnuradio-runtime.so.3.8.4 + 0xbd62d)
                #3  0x00007fa436cadeb4 _ZN5boost6detail8function26void_function_obj_invoker0IN2gr6thread19thread_body_wrapperINS3_13tpb_containerEEEvE6invokeERNS1_15function_bufferE (libgnuradio-runtime.so.3.8.4 + 0xafeb4)
                #4  0x00007fa436cc7878 _ZN5boost6detail11thread_dataINS_9function0IvEEE3runEv (libgnuradio-runtime.so.3.8.4 + 0xc9878)
[... 577 lines cut ...]
        out = 0x7fa418da93ee
        in = 0x7fa418dcfe48 "\002"
        terminate_call = false
        amt_produce = 96480
#1  0x00007fa436c69483 in gr::block_executor::run_one_iteration() () from /usr/local/lib/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#2  0x00007fa436cbb47b in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) () from /usr/local/lib/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#3  0x00007fa436cadeb4 in boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke(boost::detail::function::function_buffer&) () from /usr/local/lib/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#4  0x00007fa436cc7878 in boost::detail::thread_data<boost::function0<void> >::run() () from /usr/local/lib/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#5  0x00007fa4382e143b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
No symbol table info available.
#6  0x00007fa436b75609 in start_thread (arg=<optimized out>) at pthread_create.c:477
[... cut ...]
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140326311393024, -3866730122583529235, 140729876664718, 140729876664719, 140729876664720, 140326311390016, 3825274149679081709, 3827760745261115629}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#7  0x00007fa43670a293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
No locals.
(gdb) 
```

is there anything else I can share to assist in tracking this down?

robotastic, 2022-01-14:
Thanks for capturing the core dump! it looks like it is happening at the same place... I will go check it out. My guess is that I messed up a size check and I am trying to reach beyond the end of an array.... I can go check it out this weekend.

robotastic, 2022-01-21:
For some reason I had commented out a check that should stop this. I just merged it into Master. This was the fix: https://github.com/robotastic/trunk-recorder/commit/fd2932541e49325ebd02b9f239731b83f2eee696

Give it a try and please reopen if this doesn't fix it.

closed 2022-01-21 by robotastic

gisforgirard, 2023-02-09:
this came up again recently for me for whatever reason, i'm currently testing this change but it appears to have stabilized a major crashing issue that began appearing for me the last couple of days and kept happening consistently until testing this out - see PR:
https://github.com/robotastic/trunk-recorder/pull/772