TrunkRecorder/trunk-recorder #605: Control Channel Source Warning

tadscottsmith, 2022-01-13:
On a trunked system, setup_systems() doesn't seem to throw any kind of error if the control channel cannot be found within the range of any of the sources.

robotastic, 2022-01-14:
Oh - that is interesting... let me go check that because I thought there was a check in there somewhere for that but it may have gotten lost.

tadscottsmith, 2022-01-14:
Looked like you were accounting for it on conventional systems, but it may have been lost on trunked systems. I can also write something up, but I might not have a lot of time in the next couple of days and I wanted to track it.

robotastic, 2022-01-21:
OK - this commit should help. It will now exit if any of the trunking systems can't be tuned. Previously, it would succeed if one of the systems could be tuned. The only remaining challange is that it only checks the first control channel in each system. I will try to make it check all channels this weekend:
https://github.com/robotastic/trunk-recorder/commit/53a50fb3496544c80287b73cbab698d3cca6f1e4

closed 2022-11-21 by tadscottsmith