TrunkRecorder/trunk-recorder #606: Bug: Emergency flag not toggling correctly

bctrainers, 2022-01-14:
The emergency flag on audio transmissions per the JSON is not being persistently set correctly.

When a radio user pokes the emergency button on their radio, _and they are the **first radio activation** to a talkgroup broadcast sequence recorded_, trunk-recorder will set emergency in the JSON to true.

If a talkgroup already has a recorder active, be it just due to the persistent use of other users on the TG or the `callTimeout` value not being met, and the radio user pokes the emergency button, the emergency flag will never be toggled on with TR.

Example:

'RadioEmerg' is the radio user poking the emerg button.
'Radio A', 'Radio B' are just users using a talk group.

Works as intended: 

> RadioEmerg -> RadioA -> RadioB

The emerg flag is then set to true in the json.

Does not work as intended:

> RadioA -> RadioB -> RadioEmerg
> RadioA -> RadioEmerg -> RadioB

The emerg flag is never set to true in the JSON.

robotastic, 2022-01-21:
Agreed! What would be the ideal flow? Should a single transmission that is part of a call mark the whole call as being emergency? This seems decent, so it wouldn't be missed. I could also try to track it at the transmission level... or do both. That way you could tell which radio had set it.

It is difficult for me to test this stuff out. I am not sure if I have ever seen it toggled to be honest.

I will go research some options.

ZeroChaos-, 2022-01-21:
I would vote for "both".  As you said, marking the whole file makes it harder to miss, but marking the specific transmission avoids lost clarity.

bctrainers, 2022-01-30:
@robotastic circuit breaker like. Once it is toggled on any given radio activation, it remains on for the call recorded.

rosecitytransit, 2022-04-11:
related to this, does only the control channel get monitored for emergency flags or also voice channel messages, which also have Service Options fields? I know doing both could get a little messy since it would involve getting the data from OP25.

closed 2024-12-15 by robotastic