TrunkRecorder/trunk-recorder #610: git master 2c32fa5e04101713bbce509b5bcdd32c72ef0379 terminates on startup

ZeroChaos-, 2022-01-16:
```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  itemsize mismatch: short_to_float0:0 using 4, nonstop_wavfile_sink13:0 using 2
```

ZeroChaos-, 2022-01-16:
here is gdb with bt, info threads, and thread apply all bt full
```
Starting program: /usr/bin/trunk-recorder --config config-moon-4.json
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff3d99640 (LWP 18139)]
[Thread 0x7ffff3d99640 (LWP 18139) exited]
[New Thread 0x7ffff3d99640 (LWP 18140)]
[Thread 0x7ffff3d99640 (LWP 18140) exited]
[New Thread 0x7ffff3d99640 (LWP 18141)]
[Thread 0x7ffff3d99640 (LWP 18141) exited]
[New Thread 0x7ffff3d99640 (LWP 18142)]
[Thread 0x7ffff3d99640 (LWP 18142) exited]
[New Thread 0x7ffff3d99640 (LWP 18143)]
[Thread 0x7ffff3d99640 (LWP 18143) exited]
[New Thread 0x7ffff3d99640 (LWP 18144)]
[Thread 0x7ffff3d99640 (LWP 18144) exited]
[New Thread 0x7ffff3d99640 (LWP 18145)]
[Thread 0x7ffff3d99640 (LWP 18145) exited]
[New Thread 0x7ffff3d99640 (LWP 18146)]
[Thread 0x7ffff3d99640 (LWP 18146) exited]
[New Thread 0x7ffff3d99640 (LWP 18147)]
[Thread 0x7ffff3d99640 (LWP 18147) exited]
[New Thread 0x7ffff3d99640 (LWP 18148)]
[Thread 0x7ffff3d99640 (LWP 18148) exited]
[New Thread 0x7ffff3d99640 (LWP 18149)]

Thread 1 "trunk-recorder" received signal SIGABRT, Aborted.
0x00007ffff68c07ae in raise () from /lib64/libc.so.6
#0  0x00007ffff68c07ae in raise () from /lib64/libc.so.6
#1  0x00007ffff68aa538 in abort () from /lib64/libc.so.6
#2  0x00007ffff6c2f8d0 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#3  0x00007ffff6c3b6d8 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#4  0x00007ffff6c3b743 in std::terminate() () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#5  0x00007ffff6c3b9c4 in __cxa_throw () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#6  0x00007ffff6efe1d3 in gr::flowgraph::check_type_match(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#7  0x00007ffff6efeec2 in gr::flowgraph::connect(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#8  0x00007ffff6f0757c in gr::hier_block2_detail::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#9  0x00007ffff6f02a42 in gr::hier_block2::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#10 0x00005555555cff4d in ?? ()
#11 0x00005555555d008f in ?? ()
#12 0x00005555555ac4e7 in ?? ()
#13 0x0000555555587942 in ?? ()
#14 0x0000555555581236 in ?? ()
#15 0x00007ffff68ab80d in __libc_start_main () from /lib64/libc.so.6
#16 0x00005555555853aa in _start ()
  Id   Target Id                                          Frame 
* 1    Thread 0x7ffff3e19c00 (LWP 18135) "trunk-recorder" 0x00007ffff68c07ae in raise () from /lib64/libc.so.6
  12   Thread 0x7ffff3d99640 (LWP 18149) "libusb_event"   0x00007ffff697607f in poll () from /lib64/libc.so.6

Thread 12 (Thread 0x7ffff3d99640 (LWP 18149) "libusb_event"):
#0  0x00007ffff697607f in poll () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007ffff4855bbb in ?? () from /lib64/libusb-1.0.so.0
No symbol table info available.
#2  0x00007ffff6de1cfe in start_thread () from /lib64/libpthread.so.0
No symbol table info available.
#3  0x00007ffff6980aaf in clone () from /lib64/libc.so.6
No symbol table info available.

Thread 1 (Thread 0x7ffff3e19c00 (LWP 18135) "trunk-recorder"):
#0  0x00007ffff68c07ae in raise () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007ffff68aa538 in abort () from /lib64/libc.so.6
No symbol table info available.
#2  0x00007ffff6c2f8d0 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#3  0x00007ffff6c3b6d8 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#4  0x00007ffff6c3b743 in std::terminate() () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#5  0x00007ffff6c3b9c4 in __cxa_throw () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#6  0x00007ffff6efe1d3 in gr::flowgraph::check_type_match(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#7  0x00007ffff6efeec2 in gr::flowgraph::connect(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#8  0x00007ffff6f0757c in gr::hier_block2_detail::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#9  0x00007ffff6f02a42 in gr::hier_block2::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
No symbol table info available.
#10 0x00005555555cff4d in ?? ()
No symbol table info available.
#11 0x00005555555d008f in ?? ()
No symbol table info available.
#12 0x00005555555ac4e7 in ?? ()
No symbol table info available.
#13 0x0000555555587942 in ?? ()
No symbol table info available.
#14 0x0000555555581236 in ?? ()
No symbol table info available.
#15 0x00007ffff68ab80d in __libc_start_main () from /lib64/libc.so.6
No symbol table info available.
#16 0x00005555555853aa in _start ()
No symbol table info available.
```

ZeroChaos-, 2022-01-16:
rebuilt TR with -ggdb
```
Starting program: /usr/bin/trunk-recorder --config config-moon-4.json
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff3cde640 (LWP 24233)]
[Thread 0x7ffff3cde640 (LWP 24233) exited]
[New Thread 0x7ffff3cde640 (LWP 24234)]
[Thread 0x7ffff3cde640 (LWP 24234) exited]
[New Thread 0x7ffff3cde640 (LWP 24235)]
[Thread 0x7ffff3cde640 (LWP 24235) exited]
[New Thread 0x7ffff3cde640 (LWP 24236)]
[Thread 0x7ffff3cde640 (LWP 24236) exited]
[New Thread 0x7ffff3cde640 (LWP 24237)]
[Thread 0x7ffff3cde640 (LWP 24237) exited]
[New Thread 0x7ffff3cde640 (LWP 24238)]
[Thread 0x7ffff3cde640 (LWP 24238) exited]
[New Thread 0x7ffff3cde640 (LWP 24239)]
[Thread 0x7ffff3cde640 (LWP 24239) exited]
[New Thread 0x7ffff3cde640 (LWP 24240)]
[Thread 0x7ffff3cde640 (LWP 24240) exited]
[New Thread 0x7ffff3cde640 (LWP 24241)]
[Thread 0x7ffff3cde640 (LWP 24241) exited]
[New Thread 0x7ffff3cde640 (LWP 24242)]
[Thread 0x7ffff3cde640 (LWP 24242) exited]
[New Thread 0x7ffff3cde640 (LWP 24243)]

Thread 1 "trunk-recorder" received signal SIGABRT, Aborted.
0x00007ffff68057ae in raise () from /lib64/libc.so.6
#0  0x00007ffff68057ae in raise () from /lib64/libc.so.6
#1  0x00007ffff67ef538 in abort () from /lib64/libc.so.6
#2  0x00007ffff6b748d0 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#3  0x00007ffff6b806d8 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#4  0x00007ffff6b80743 in std::terminate() () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#5  0x00007ffff6b809c4 in __cxa_throw () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
#6  0x00007ffff6e431d3 in gr::flowgraph::check_type_match(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#7  0x00007ffff6e43ec2 in gr::flowgraph::connect(gr::endpoint const&, gr::endpoint const&) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#8  0x00007ffff6e4c57c in gr::hier_block2_detail::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#9  0x00007ffff6e47a42 in gr::hier_block2::connect(boost::shared_ptr<gr::basic_block>, int, boost::shared_ptr<gr::basic_block>, int) () from /usr/lib64/libgnuradio-runtime.so.3.8.4
#10 0x00005555556968ac in dmr_recorder::initialize (this=0x5555561c3f40, src=0x55555592ae50) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/recorders/dmr_recorder.cc:262
#11 0x000055555569354b in make_dmr_recorder (src=0x55555592ae50) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/recorders/dmr_recorder.cc:10
#12 0x00005555556425cb in Source::create_dmr_conventional_recorder (this=0x55555592ae50, tb=...) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/source.cc:204
#13 0x00005555555e295f in setup_systems () at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/main.cc:1287
#14 0x00005555555e4a03 in main (argc=3, argv=0x7fffffffd938) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/main.cc:1426
  Id   Target Id                                          Frame 
* 1    Thread 0x7ffff3d5ec00 (LWP 24229) "trunk-recorder" 0x00007ffff68057ae in raise () from /lib64/libc.so.6
  12   Thread 0x7ffff3cde640 (LWP 24243) "libusb_event"   0x00007ffff68bb07f in poll () from /lib64/libc.so.6

Thread 12 (Thread 0x7ffff3cde640 (LWP 24243) "libusb_event"):
#0  0x00007ffff68bb07f in poll () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007ffff479abbb in ?? () from /lib64/libusb-1.0.so.0
No symbol table info available.
#2  0x00007ffff6d26cfe in start_thread () from /lib64/libpthread.so.0
No symbol table info available.
#3  0x00007ffff68c5aaf in clone () from /lib64/libc.so.6
No symbol table info available.

Thread 1 (Thread 0x7ffff3d5ec00 (LWP 24229) "trunk-recorder"):
#0  0x00007ffff68057ae in raise () from /lib64/libc.so.6
No symbol table info available.
#1  0x00007ffff67ef538 in abort () from /lib64/libc.so.6
No symbol table info available.
#2  0x00007ffff6b748d0 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#3  0x00007ffff6b806d8 in ?? () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#4  0x00007ffff6b80743 in std::terminate() () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
No symbol table info available.
#5  0x00007ffff6b809c4 in __cxa_throw () from /usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/libstdc++.so.6
[... 28 lines cut ...]
        do_nocrypt = true
#11 0x000055555569354b in make_dmr_recorder (src=0x55555592ae50) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/recorders/dmr_recorder.cc:10
        recorder = 0x5555561c3f40
#12 0x00005555556425cb in Source::create_dmr_conventional_recorder (this=0x55555592ae50, tb=...) at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/source.cc:204
        log = {px = 0x0, pn = {pi_ = 0x5555561bf9d0}}
#13 0x00005555555e295f in setup_systems () at /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999/trunk-recorder/main.cc:1287
        rec = {px = 0x0, pn = {pi_ = 0x0}}
        call = 0x5555561c2640
[... cut ...]
        desc = {static m_default_line_length = 80, m_caption = "Options", m_line_length = 80, m_min_description_length = 40, m_options = std::vector of length 2, capacity 2 = {{px = 0x555555848000, pn = {pi_ = 0x555555847b10}}, {px = 0x5555558481b0, pn = {pi_ = 0x5555558487a0}}}, belong_to_group = std::vector<bool> of length 2, capacity 64 = {false, false}, groups = std::vector of length 0, capacity 0}
        vm = <incomplete type>
        config_file = "config-moon-4.json"
        uri = "ws://localhost:3005"
```

ZeroChaos-, 2022-01-16:
as is fairly obvious from the gdb output, it doesn't fail if I remove my conventionaldmr channels from the config

robotastic, 2022-01-16:
OK - I think I fixed it. The problem was with the switch from Float to int16.

ZeroChaos-, 2022-01-16:
confirmed, thanks!

closed 2022-01-16 by ZeroChaos-