TrunkRecorder/trunk-recorder #611: Release 4.3.0 and master won't build on Arch

toddejohnson, 2022-01-16:
I'm working on updating the PKGBUILD for the aur package. 

```
[  1%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/recorders/recorder.cc.o
In file included from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../source.h:13,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.cc:2:
/build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../recorders/p25_recorder.h:66:41: error: conf
licting declaration ‘typedef class std::shared_ptr<p25_recorder> p25_recorder_sptr’
   66 |   typedef std::shared_ptr<p25_recorder> p25_recorder_sptr;
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../call.h:19,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.h:59,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.cc:1:
/build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../systems/system.h:38:43: note: previous decl
aration as ‘typedef class boost::shared_ptr<p25_recorder> p25_recorder_sptr’
   38 |   typedef boost::shared_ptr<p25_recorder> p25_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
In file included from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../source.h:14,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.cc:2:
/build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../recorders/dmr_recorder.h:88:41: error: conf
licting declaration ‘typedef class std::shared_ptr<dmr_recorder> dmr_recorder_sptr’
   88 |   typedef std::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../call.h:19,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.h:59,
                 from /build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/recorder.cc:1:
/build/trunk-recorder/src/trunk-recorder-4.3.0/trunk-recorder/recorders/../systems/system.h:39:43: note: previous decl
aration as ‘typedef class boost::shared_ptr<dmr_recorder> dmr_recorder_sptr’
   39 |   typedef boost::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:76: CMakeFiles/trunk_recorder_library.dir/trunk-recorde
r/recorders/recorder.cc.o] Error 1
```
For master:
```
[  1%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/recorders/recorder.cc.o
In file included from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../source.h:13,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.cc:2:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../recorders/p25_recorder.h:66:41: error: confli
cting declaration ‘typedef class std::shared_ptr<p25_recorder> p25_recorder_sptr’                         
   66 |   typedef std::shared_ptr<p25_recorder> p25_recorder_sptr;                                        
      |                                         ^~~~~~~~~~~~~~~~~                                                     
In file included from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../call.h:19,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.h:59,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.cc:1:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../systems/system.h:38:43: note: previous declar
ation as ‘typedef class boost::shared_ptr<p25_recorder> p25_recorder_sptr’
   38 |   typedef boost::shared_ptr<p25_recorder> p25_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
In file included from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../source.h:14,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.cc:2:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../recorders/dmr_recorder.h:88:41: error: confli
cting declaration ‘typedef class std::shared_ptr<dmr_recorder> dmr_recorder_sptr’
   88 |   typedef std::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../call.h:19,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.h:59,
                 from /build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/recorder.cc:1:
/build/trunk-recorder-git/src/trunk-recorder/trunk-recorder/recorders/../systems/system.h:39:43: note: previous decla$
ation as ‘typedef class boost::shared_ptr<dmr_recorder> dmr_recorder_sptr’
   39 |   typedef boost::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:76: CMakeFiles/trunk_recorder_library.dir/trunk-recorde
r/recorders/recorder.cc.o] Error 1
```

We're on Boost-1.78.0 and gnuradio-3.9.4.0-2 now if it matters.

toddejohnson, 2022-01-16:
Still an issue with the latest 4.3.0 and 7c65f00

robotastic, 2022-01-16:
weird -yea, it builds fine locally and is failing on the CI. I will go figure it out

robotastic, 2022-01-16:
actually I take that back. it looks like the latest version is still building... I will keep you posted.

PR #612 (2022-01-17): Should complie issue with boost vs std in system.h
This fixes #611 for me. Thanks to @leee in Discord.
closed 2022-01-17 by robotastic