TrunkRecorder/trunk-recorder #614: possibly unusual unit identification in TR json output

ZeroChaos-, 2022-01-18:
Now that trbo-support has been merged I'm back on master (from yesterdayish).  I have the following json.  I have modified it during encode-upload to change the talkgroup, the original talkgroup was 1, as this is an analog conventional station.  The srcList having two sources is a little confusing and I wanted to make sure this is the intended behavior, and not inaccurate.  The first src listed is "-1" indicating there is no identifier (makes sense, the first bit of the transmission didn't have any mdc1200), and then the second bit of the same wav file has an mdc1200 and thus is identified.  My question is this, is it expected that (more or less) every analog transmission is going to list -1 as the first source?  It looks like if the wav doesn't have mdc1200 (or some other supported identifier) in the first ~second that -1 will be specified?  Is that how this works? Is this how it's "supposed" to work?  I'm fine with the behavior, I am just checking if this was the intentional and desired behavior.
```
{
  "freq": 470437500,
  "start_time": 1642518389,
  "stop_time": 1642518415,
  "emergency": 0,
  "encrypted": 0,
  "call_length": 24,
  "talkgroup": 99,
  "talkgroup_tag": "West EMS Dispatch Identified",
  "audio_type": "digital",
  "short_name": "wems",
  "freqList": [
    {
      "freq": 470437500,
      "time": 1642518389,
      "pos": 0.0,
      "len": 24,
      "error_count": 0.0,
      "spike_count": 0.0
    }
  ],
  "srcList": [
    {
      "src": -1,
      "time": 1642518389,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 56212,
      "time": 1642518409,
      "pos": 17,
      "emergency": 0,
      "signal_system": "",
      "tag": "Northwest EMS Dispatch 191 (db94)"
    }
  ]
}
```

ZeroChaos-, 2022-01-18:
That and this one (same tg manipulation, it was originally 1) is even crazier as -1 is specified many times?
```
{
  "freq": 470437500,
  "start_time": 1642518972,
  "stop_time": 1642519004,
  "emergency": 0,
  "encrypted": 0,
  "call_length": 30,
  "talkgroup": 2,
  "talkgroup_tag": "West EMS Dispatch Unidentified",
  "audio_type": "digital",
  "short_name": "wems",
  "freqList": [
    {
      "freq": 470437500,
      "time": 1642518972,
      "pos": 0.0,
      "len": 30,
      "error_count": 0.0,
      "spike_count": 0.0
    }
  ],
  "srcList": [
    {
      "src": -1,
      "time": 1642518972,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": -1,
      "time": 1642518979,
      "pos": 7,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": -1,
      "time": 1642518992,
      "pos": 18,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": -1,
      "time": 1642518995,
      "pos": 21,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    }
  ]
}
```
Is this because squelch dropped that many times but the callTimeout kept the wav active?  Again, I'm mostly requesting clarification here, as this behavior seems very useful to me (it identifies that there were X number of transmissions even if we don't know the unit ids).  I just want to make sure this isn't breaking anyone else's assumptions when parsing this (and that it's working as "desired" by upstream)

robotastic, 2022-01-21:
So this is related to this change from @taclane - I was testing out have squelch breaks trigger a new Transmission/wav file. Except the GNU Radio docs didn't match the code, so it never worked. Taclane caught this PR'd a fix. I added that in and it gave the behavior you are seeing above. 

But it ended up getting reverted  ( #608 ) because it was messing up some of the Conventional P25. so right now, all of those separate -1 transmissions would probably just be one file.

closed 2022-01-21 by robotastic