TrunkRecorder/trunk-recorder #616: Audio Duplication through Rdio-Scanner OpenMHZ

Nokoa, 2022-01-19:
I seem to be having an issue, which from what I noticed on OpenMHZ others are perhaps also experiencing.

The issue I am having is with file uploading to Rdio-Scanner using upload script, or directory watching. 

On a regular basis a full transmission is uploaded, and then parts of the same transmissions are also uploaded again. The partial recordings may start in the middle and go through to the end. This can happen multiple times for the same transmission. An attached example was taken from a Feed from OpenMHZ.
[recordings.zip](https://github.com/robotastic/trunk-recorder/files/7897976/recordings.zip)

JayManCanada6014, 2022-01-19:
I suspect your seeing the same thing https://github.com/robotastic/trunk-recorder/issues/535  

I was able to reduce mine a lot by changing the call timeout to 4 seconds but still get them when the system is busy.

robotastic, 2022-01-20:
Can you give the `overlap` branch a try? It should help cut down on this.

robotastic, 2022-02-04:
@Nokoa If you are able to capture the output from Trunk Recorder the next time you see this, that would be really helpful. It would be good to have all the output covering when the calls start to when they end.

Nokoa, 2022-02-07:
Here are parts of the logs for duplicated transmission. One file contains the full transmission while the other contains only a part from middle to end. Just note that the issue I am describing here is different than this one #321 

`[2022-02-07 10:07:24.091309] (info)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	TG not in Talkgroup File 
[2022-02-07 10:07:24.091388] (info)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	[32mStarting P25 Recorder Num [0][0m	TDMA: true	Slot: 1	QPSK: true
[2022-02-07 10:07:24.464924] (info)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	Starting new Transmission 	Src ID:  -1
[2022-02-07 10:07:28.005370] (trace)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s
[2022-02-07 10:07:28.852888] (info)   [islps-74]	[0;34m105405C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	TG not in Talkgroup File 
[2022-02-07 10:07:28.852980] (info)   [islps-74]	[0;34m105405C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	[32mStarting P25 Recorder Num [1][0m	TDMA: true	Slot: 1	QPSK: true
[2022-02-07 10:07:29.872283] (info)   	 - Stopping call because of overlapping Freq
[2022-02-07 10:07:29.872366] (trace)   [islps-74]	[0;34m105405C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	Stopping Recorded Call, setting call state to COMPLETED - Last Update: 1s
[2022-02-07 10:07:29.872410] (info)   [islps-74]	[0;34m105406C[0m	TG: [35m       919[0m	Freq: 8.677000e+08	TG not in Talkgroup File 
[2022-02-07 10:07:29.872446] (info)   [islps-74]	[0;34m105406C[0m	TG: [35m       919[0m	Freq: 8.677000e+08	[32mStarting P25 Recorder Num [4][0m	TDMA: true	Slot: 1	QPSK: true
[2022-02-07 10:07:30.003811] (trace)   [islps-74]	[0;34m105403C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	[33mConcluding Recorded Call[0m - Last Update: 12s	Call Elapsed: 18
[2022-02-07 10:07:30.003908] (info)   [islps-74]	[0;34m105403C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	- Transmission src: 3190008 pos: 0 length: 16.24
[2022-02-07 10:07:30.003992] (trace)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	[33mConcluding Recorded Call[0m - Last Update: 6s	Call Elapsed: 6
[2022-02-07 10:07:30.004043] (info)   [islps-74]	[0;34m105404C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	- Transmission src: -1 pos: 0 length: 4.4
[2022-02-07 10:07:30.004099] (trace)   [islps-74]	[0;34m105405C[0m	TG: [35m      4051[0m	Freq: 8.677000e+08	[33mConcluding Recorded Call[0m - Last Update: 2s	Call Elapsed: 2
[2022-02-07 10:07:30.004137] (error)   [islps-74]	[0;34m105405C[0m	TG: [35m      4051[0m	 Freq: 8.677e+08	No Transmission were recorded!
[2022-02-07 10:07:30.007025] (info)   [islps-74]	[0;34m105404C[0m 	 Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.m4a
[2022-02-07 10:07:30.007592] (info)   [islps-74]	[0;34m105403C[0m 	 Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.m4a`

robotastic, 2022-02-07:
````

[2022-02-07 10:07:24.091309] (info) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 TG not in Talkgroup File 
[2022-02-07 10:07:24.091388] (info) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 �[32mStarting P25 Recorder Num [0]�[0m TDMA: true Slot: 1 QPSK: true 
[2022-02-07 10:07:24.464924] (info) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 Starting new Transmission Src ID: -1 
[2022-02-07 10:07:28.005370] (trace) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s 
[2022-02-07 10:07:28.852888] (info) [islps-74] �[0;34m105405C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 TG not in Talkgroup File 
[2022-02-07 10:07:28.852980] (info) [islps-74] �[0;34m105405C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 �[32mStarting P25 Recorder Num [1]�[0m TDMA: true Slot: 1 QPSK: true 
[2022-02-07 10:07:29.872283] (info) - Stopping call because of overlapping Freq 
[2022-02-07 10:07:29.872366] (trace) [islps-74] �[0;34m105405C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 Stopping Recorded Call, setting call state to COMPLETED - Last Update: 1s 
[2022-02-07 10:07:29.872410] (info) [islps-74] �[0;34m105406C�[0m TG: �[35m 919�[0m Freq: 8.677000e+08 TG not in Talkgroup File 
[2022-02-07 10:07:29.872446] (info) [islps-74] �[0;34m105406C�[0m TG: �[35m 919�[0m Freq: 8.677000e+08 �[32mStarting P25 Recorder Num [4]�[0m TDMA: true Slot: 1 QPSK: true 
[2022-02-07 10:07:30.003811] (trace) [islps-74] �[0;34m105403C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 �[33mConcluding Recorded Call�[0m - Last Update: 12s Call Elapsed: 18 
[2022-02-07 10:07:30.003908] (info) [islps-74] �[0;34m105403C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 - Transmission src: 3190008 pos: 0 length: 16.24 
[2022-02-07 10:07:30.003992] (trace) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 �[33mConcluding Recorded Call�[0m - Last Update: 6s Call Elapsed: 6 
[2022-02-07 10:07:30.004043] (info) [islps-74] �[0;34m105404C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 - Transmission src: -1 pos: 0 length: 4.4 
[2022-02-07 10:07:30.004099] (trace) [islps-74] �[0;34m105405C�[0m TG: �[35m 4051�[0m Freq: 8.677000e+08 �[33mConcluding Recorded Call�[0m - Last Update: 2s Call Elapsed: 2 
[2022-02-07 10:07:30.004137] (error) [islps-74] �[0;34m105405C�[0m TG: �[35m 4051�[0m Freq: 8.677e+08 No Transmission were recorded! 
[2022-02-07 10:07:30.007025] (info) [islps-74] �[0;34m105404C�[0m Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257244_867700000-call_105404.m4a \n
[2022-02-07 10:07:30.007592] (info) [islps-74] �[0;34m105403C�[0m Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/4051-1644257232_867700000-call_105403.m4a
````

Awesome- Thanks, I added in some formatting. I will see if I can spot anything

robotastic, 2022-02-08:
@Nokoa It looks like there are actually 3 calls in this section. Do you know which calls it was?
It looks like there is 105403C , 105404C , 105405C 

The interesting thing is that they are all the same TG on the same Freq.... I will go dig into that scenario

Nokoa, 2022-02-08:
> 

I will get you some more logs that may have more information. 

TG 4051 in this instance was duplicated partially as described before. It looks like in the logs I had 4051 and 919 both active on the same frequency in different TDMA slots. Could it be that 919 starting in the middle of the transmission of 4051 in the other slot started a new recorder for 919 and then again for 4051. Maybe that's why the duplicated transmission doesn't start at the beginning, but still ends at the end? But then I am not sure why 4051 had 2 recorders started before 919 began.

Nokoa, 2022-02-08:
Here is another instance, here TG 4087 is duplicated. One recording is the full recording. The second one starts a little bit after the first one and goes on until its end.

`[2022-02-07 22:24:00.688738] (info)   [islps-74]	[0;34m2816C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	Starting new Transmission 	Src ID:  2900945
[2022-02-07 22:24:12.243453] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	TG not in Talkgroup File 
[2022-02-07 22:24:12.243587] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	[32mStarting P25 Recorder Num [1][0m	TDMA: true	Slot: 1	QPSK: true
[2022-02-07 22:24:12.855116] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	Starting new Transmission 	Src ID:  6922592
[2022-02-07 22:24:15.696425] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	Starting new Transmission 	Src ID:  6922592
[2022-02-07 22:24:19.013459] (trace)   [islps-74]	[0;34m2816C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s
[2022-02-07 22:24:19.013573] (trace)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s
[2022-02-07 22:24:20.799516] (info)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	TG not in Talkgroup File 
[2022-02-07 22:24:20.799607] (info)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	[32mStarting P25 Recorder Num [3][0m	TDMA: true	Slot: 1	QPSK: true
[2022-02-07 22:24:21.265375] (info)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	Starting new Transmission 	Src ID:  2900451
[2022-02-07 22:24:21.344858] (info)   [islps-74]	[0;34m2820C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	TG not in Talkgroup File 
[2022-02-07 22:24:21.344970] (info)   [islps-74]	[0;34m2820C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	[32mStarting P25 Recorder Num [4][0m	TDMA: true	Slot: 0	QPSK: true
[2022-02-07 22:24:21.602840] (info)   [islps-74]	[0;34m2820C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	Starting new Transmission 	Src ID:  -1
[2022-02-07 22:24:26.016494] (info)   Recorder state: 1
[2022-02-07 22:24:26.016623] (error)   [islps-74]	[0;34m2818C[0m  Rec Num: 1	TG: [35m     51397[0m	Freq: 8.685250e+08	[36m Removing call with stuck recorder [0m
[2022-02-07 22:24:26.016702] (trace)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	[33mConcluding Recorded Call[0m - Last Update: 11s	Call Elapsed: 14
[2022-02-07 22:24:26.016860] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1
[2022-02-07 22:24:26.016927] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	- Transmission src: 6922592 pos: 0 length: 0.16
[2022-02-07 22:24:26.017006] (info)   [islps-74]	[0;34m2818C[0m	TG: [35m     51397[0m	Freq: 8.685250e+08	- Transmission src: 6922592 pos: 0.16 length: 0.08
[2022-02-07 22:24:26.020456] (info)   [islps-74]	[0;34m2818C[0m 	 Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.m4a
[2022-02-07 22:24:27.015512] (trace)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2022-02-07 22:24:27.015610] (trace)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	[33mConcluding Recorded Call[0m - Last Update: 4s	Call Elapsed: 7
[2022-02-07 22:24:27.015649] (info)   [islps-74]	[0;34m2819C[0m	TG: [35m       917[0m	Freq: 8.679625e+08	- Transmission src: 2900451 pos: 0 length: 0.16
[2022-02-07 22:24:27.015721] (trace)   [islps-74]	[0;34m2820C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s
[2022-02-07 22:24:27.018741] (info)   [islps-74]	[0;34m2819C[0m 	 Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.m4a
[2022-02-07 22:24:28.397873] (info)   [islps-74]	[0;34m2821C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	TG not in Talkgroup File 
[2022-02-07 22:24:28.397957] (info)   [islps-74]	[0;34m2821C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	[32mStarting P25 Recorder Num [1][0m	TDMA: true	Slot: 0	QPSK: true
[2022-02-07 22:24:28.673655] (info)   [islps-74]	[0;34m2821C[0m	TG: [35m      1097[0m	Freq: 8.682375e+08	Starting new Transmission 	Src ID:  3190012
[2022-02-07 22:24:32.010575] (trace)   [islps-74]	[0;34m2817C[0m	TG: [35m     11149[0m	Freq: 8.682375e+08	Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s
[2022-02-07 22:24:32.010670] (trace)   [islps-74]	[0;34m2817C[0m	TG: [35m     11149[0m	Freq: 8.682375e+08	[33mConcluding Recorded Call[0m - Last Update: 4s	Call Elapsed: 35
[... cut ...] islps-74/2022/2/7/51395-1644301555_867962500-call_2833.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51395-1644301555_867962500-call_2833.m4a
[2022-02-07 22:26:03.667518] (info)   [islps-74]	[0;34m2832C[0m	TG: [35m      1097[0m	Freq: 8.679625e+08	Starting new Transmission 	Src ID:  2900250
[2022-02-07 22:26:07.012781] (trace)   [islps-74]	[0;34m2832C[0m	TG: [35m      1097[0m	Freq: 8.679625e+08	Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s`

robotastic, 2022-02-08:
That would be great - I will double check the code around TDMA Slots. I think I check for that, but it could be a weird corner case. 

One thing that will definitely help is if you set `callTimeout` to be 2 secs or maybe 1. The callTimeout governs how long a "Call" sticks around after the last trunk update message is heard. If a new transmission starts on the control channel, it should stop the recorder... but if for some reason the recorder catches the start of the new transmission on the voice channel before the control channel message, it will start recording.

Also make sure you have good reception for your control channel. Set `controlWarnRate` to -1 and it will print out the number of messages decoded. It should be above 30. Both of these settings should be at the top level of the config file.

robotastic, 2022-02-08:
Formatted version:
````

[2022-02-07 22:24:00.688738] (info) [islps-74] 2816C TG:  1097 Freq: 8.682375e+08 Starting new Transmission Src ID: 2900945 
[2022-02-07 22:24:12.243453] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 TG not in Talkgroup File 
[2022-02-07 22:24:12.243587] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 Starting P25 Recorder Num [1] TDMA: true Slot: 1 QPSK: true 
[2022-02-07 22:24:12.855116] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 Starting new Transmission Src ID: 6922592 
[2022-02-07 22:24:15.696425] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 Starting new Transmission Src ID: 6922592 
[2022-02-07 22:24:19.013459] (trace) [islps-74] 2816C TG:  1097 Freq: 8.682375e+08 Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s 
[2022-02-07 22:24:19.013573] (trace) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s 
[2022-02-07 22:24:20.799516] (info) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 TG not in Talkgroup File 
[2022-02-07 22:24:20.799607] (info) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 Starting P25 Recorder Num [3] TDMA: true Slot: 1 QPSK: true 
[2022-02-07 22:24:21.265375] (info) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 Starting new Transmission Src ID: 2900451 
[2022-02-07 22:24:21.344858] (info) [islps-74] 2820C TG:  1097 Freq: 8.682375e+08 TG not in Talkgroup File 
[2022-02-07 22:24:21.344970] (info) [islps-74] 2820C TG:  1097 Freq: 8.682375e+08 Starting P25 Recorder Num [4] TDMA: true Slot: 0 QPSK: true 
[2022-02-07 22:24:21.602840] (info) [islps-74] 2820C TG:  1097 Freq: 8.682375e+08 Starting new Transmission Src ID: -1 
[2022-02-07 22:24:26.016494] (info) Recorder state: 1 
[2022-02-07 22:24:26.016623] (error) [islps-74] 2818C Rec Num: 1 TG:  51397 Freq: 8.685250e+08 �[36m Removing call with stuck recorder  
[2022-02-07 22:24:26.016702] (trace) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 Concluding Recorded Call - Last Update: 11s Call Elapsed: 14 
[2022-02-07 22:24:26.016860] (error) stop_recording() - stopping wavfile sink but recorder state is: 1 
[2022-02-07 22:24:26.016927] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 - Transmission src: 6922592 pos: 0 length: 0.16 
[2022-02-07 22:24:26.017006] (info) [islps-74] 2818C TG:  51397 Freq: 8.685250e+08 - Transmission src: 6922592 pos: 0.16 length: 0.08 
[2022-02-07 22:24:26.020456] (info) [islps-74] 2818C Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51397-1644301452_868525000-call_2818.m4a 
[2022-02-07 22:24:27.015512] (trace) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s 
[2022-02-07 22:24:27.015610] (trace) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 Concluding Recorded Call - Last Update: 4s Call Elapsed: 7 
[2022-02-07 22:24:27.015649] (info) [islps-74] 2819C TG:  917 Freq: 8.679625e+08 - Transmission src: 2900451 pos: 0 length: 0.16 
[2022-02-07 22:24:27.015721] (trace) [islps-74] 2820C TG:  1097 Freq: 8.682375e+08 Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s 
[2022-02-07 22:24:27.018741] (info) [islps-74] 2819C Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/917-1644301461_867962500-call_2819.m4a 
[2022-02-07 22:24:28.397873] (info) [islps-74] 2821C TG:  1097 Freq: 8.682375e+08 TG not in Talkgroup File 
[2022-02-07 22:24:28.397957] (info) [islps-74] 2821C TG:  1097 Freq: 8.682375e+08 Starting P25 Recorder Num [1] TDMA: true Slot: 0 QPSK: true 
[2022-02-07 22:24:28.673655] (info) [islps-74] 2821C TG:  1097 Freq: 8.682375e+08 Starting new Transmission Src ID: 3190012 
[2022-02-07 22:24:32.010575] (trace) [islps-74] 2817C TG:  11149 Freq: 8.682375e+08 Stopping Recorded Call, setting call state to COMPLETED - Last Update: 4s 
[2022-02-07 22:24:32.010670] (trace) [islps-74] 2817C TG:  11149 Freq: 8.682375e+08 Concluding Recorded Call - Last Update: 4s Call Elapsed: 35 
[2022-02-07 22:24:32.010741] (info) [islps-74] 2817C TG:  11149 Freq: 8.682375e+08 - Transmission src: 2002157 pos: 0 length: 18.96 
[2022-02-07 22:24:32.015008] (info) [islps-74] 2817C Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/11149-1644301437_868237500-call_2817.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/11149-1644301437_868237500-call_2817.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/11149-1644301437_868237500-call_2817.m4a 
[2022-02-07 22:24:37.983677] (info) [islps-74] 2822C TG:  917 Freq: 8.682375e+08 TG not in Talkgroup File 
[2022-02-07 22:24:37.983897] (info) [islps-74] 2822C TG:  917 Freq: 8.682375e+08 Starting P25 Recorder Num [2] TDMA: true Slot: 1 QPSK: true 
[2022-02-07 22:24:38.446090] (info) [islps-74] 2822C TG:  917 Freq: 8.682375e+08 Starting new Transmission Src ID: 2900407 
[... cut ...]
[2022-02-07 22:26:01.012432] (info) [islps-74] 2833C Running upload script: ./upload-call.sh 1 /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51395-1644301555_867962500-call_2833.wav /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51395-1644301555_867962500-call_2833.json /home/shahar/trunk-recorder2022/overlap-test/islps-74/2022/2/7/51395-1644301555_867962500-call_2833.m4a 
[2022-02-07 22:26:03.667518] (info) [islps-74] 2832C TG:  1097 Freq: 8.679625e+08 Starting new Transmission Src ID: 2900250 
[2022-02-07 22:26:07.012781] (trace) [islps-74] 2832C TG:  1097 Freq: 8.679625e+08 Stopping Recorded Call, setting call state to INACTIVE - Last Update: 4s
````

robotastic, 2022-02-14:
@Nokoa Can you give the `no-overlap` branch a try? It should do a better job of reducing duplication.

tadscottsmith, 2022-02-16:
@robotastic I couldn't  build without modifying this line. #6b8140d9af903a8c7d6abab7fd5e33aefd2482e1

robotastic, 2022-02-18:
You are totally right ... not sure what happened there ... just fixed it, thanks @tadscottsmith

Nokoa, 2022-02-18:
> @Nokoa Can you give the `no-overlap` branch a try? It should do a better job of reducing duplication.

I am away from the equipment for a couple weeks, will give it a try as soon as I return, thank you!

tadscottsmith, 2022-02-22:
[02-22-2022_0125_01.zip](https://github.com/robotastic/trunk-recorder/files/8113214/02-22-2022_0125_01.zip)

I am running this branch and it seems to be working pretty well. I am not on a SmartNet system where it seems like most of the overlap issues have been happening, but I have noticed that occasionally this branch will drop transmissions. There's pretty clearly a missed transmission in the attached example between calls 753 and 755. I'd actually expect them to be one call, but 753 seems to end prematurely.

1. Call 753: SRC 317654 - "9175 Traffic."
2. Call 753: SRC 312983 - "9175."
3. MISSED - But, I assume something like "Out with plate XXX-XXX at Century /County E".
4. Call 755: SRC 312983 "Copy. Traffic. Century / County Road E , Echo..."

tadscottsmith, 2022-02-22:
I originally attached the wrong log file, but that should be corrected in the zip above.

robotastic, 2022-02-22:
Nice find! and helpful feedback. Let me go check out the Log file

This is one of the tricky things... perviously Trunk Recorder failed permissively, where it would default to recording... now it only records when it gets explicit GRANT messages, so if the control is full or not working 100% (or I messed up some code) transmissions may not get recorded.

robotastic, 2022-02-22:
yea - so it looks like there were definitely some transmissions that were dropped from call 753. It looks like it never received another GRANT message, so it dropped the samples coming in. I wonder if the fact that it was a Patched TG would have anything to do with it? Any chance the GRANT would have come in with a different TG? I didn't see anything obvious... and it would have ended the existing call if it got a GRANT for a TG on the same channel.

Call 753:
````
[2022-02-22 02:03:47.484344] (trace)   tbsk02	Moto Patch Grant	Channel ID:   269	Freq: 8.526875e+08	sg   11518	TDMA -1	sa 317654
[2022-02-22 02:03:47.484566] (info)   [ramsey]	[0;34m753C[0m	TG: [35m            RAM SO CENT[0m (     11518)	Freq: 8.526875e+08	[32mStarting P25 Recorder Num [9][0m	TDMA: false	Slot: 0	QPSK: true
````

````
[2022-02-22 02:03:54.889841] (info)   TEMINATOR!!
[2022-02-22 02:03:54.890054] (info)   [ramsey]	[0;34m753C[0m	TG: [35m     11518[0m	Freq: 8.526875e+08	record_more_transmissions is false, setting recorder state to STOPPED
````

````
[2022-02-22 02:03:59.075975] (trace)   MOT_GRG_CN_GRANT_UPDT(0x03): 	Channel ID:   269	Freq: 8.526875e+08	sg   11518	TDMA -1
[2022-02-22 02:03:59.076089] (info)   [ramsey]	[0;34m753	TTG: [35m            RAM SO CENT[0m (     11518)	Freq: 8.526875e+08	[36m Updating Call [0m
[2022-02-22 02:03:59.126351] (trace)   P25 Frame Assembler - Amt Prod: 864 output_queue: 1440 noutput_items: 864
[2022-02-22 02:03:59.126609] (trace)   setting silence_frame_count 0 to d_silence_frames: 0
[2022-02-22 02:03:59.126701] (error)   [ramsey]	[0;34m753C[0m	TG: [35m     11518[0m	Freq: 8.526875e+08	Dropping samples - Recorder state is: stopped
----
[2022-02-22 02:03:59.728549] (trace)   P25 Frame Assembler - Amt Prod: 864 output_queue: 1440 noutput_items: 864
[2022-02-22 02:03:59.728917] (trace)   setting silence_frame_count 0 to d_silence_frames: 0
[2022-02-22 02:03:59.729183] (error)   [ramsey]	[0;34m753C[0m	TG: [35m     11518[0m	Freq: 8.526875e+08	Dropping samples - Recorder state is: stopped
[2022-02-22 02:03:59.779754] (trace)   nac 40d type 7 size 12 mesg len: 12
[2022-02-22 02:03:59.779845] (trace)   TSBK: opcode: $3
[2022-02-22 02:03:59.779884] (trace)   MOT_GRG_CN_GRANT_UPDT(0x03): 	Channel ID:   269	Freq: 8.526875e+08	sg   11518	TDMA -1
[2022-02-22 02:03:59.779905] (info)   [ramsey]	[0;34m753	TTG: [35m            RAM SO CENT[0m (     11518)	Freq: 8.526875e+08	[36m Updating Call [0m
[2022-02-22 02:03:59.779932] (trace)   nac 40d type 7 size 12 mesg len: 12
[2022-02-22 02:03:59.779975] (trace)   TSBK: opcode: $5
[2022-02-22 02:03:59.779991] (debug)   tsbk05: Unit To Unit Answer Request
[2022-02-22 02:03:59.780012] (trace)   nac 40d type 7 size 12 mesg len: 12
[2022-02-22 02:03:59.780030] (trace)   TSBK: opcode: $9
[2022-02-22 02:03:59.780044] (debug)   tsbk09: Telephone Interconnect Voice Channel Grant Update
[2022-02-22 02:03:59.830215] (trace)   P25 Frame Assembler - Amt Prod: 864 output_queue: 576 noutput_items: 864
[2022-02-22 02:03:59.830420] (trace)   setting silence_frame_count 0 to d_silence_frames: 0
[2022-02-22 02:03:59.830515] (error)   [ramsey]	[0;34m753C[0m	TG: [35m     11518[0m	Freq: 8.526875e+08	Dropping samples - Recorder state is: stopped
````

Call 755:
````
[2022-02-22 02:04:04.919457] (trace)   tbsk02	Moto Patch Grant	Channel ID:   209	Freq: 8.523125e+08	sg   11518	TDMA -1	sa 312983
[2022-02-22 02:04:04.919498] (info)   [ramsey]	[0;34m755C[0m	TG: [35m            RAM SO CENT[0m (     11518)	Freq: 8.523125e+08	[32mStarting P25 Recorder Num [0][0m	TDMA: false	Slot: 0	QPSK: true
````

Call 754:
````
[2022-02-22 02:03:49.504261] (debug)   tsbk00	Chan Grant	Channel ID:    89	Freq: 8.515625e+08	ga    3750	TDMA -1	sa 127071	Encrypt 0	Bandwidth: 12.5
[2022-02-22 02:03:49.504318] (info)   TG: 3750 sys num: 0 freq: 8.51562e+08 TDMA Slot0 TDMA: false
[2022-02-22 02:03:49.548597] (trace)   P25 Frame Assembler - Amt Prod: 864 output_queue: 1152 noutput_items: 864
[2022-02-22 02:03:49.548694] (trace)   setting silence_frame_count 0 to d_silence_frames: 0
[2022-02-22 02:03:49.593397] (trace)   nac 40d type 7 size 12 mesg len: 12
[2022-02-22 02:03:49.593511] (trace)   TSBK: opcode: $0
````

tadscottsmith, 2022-02-22:
It seems like maybe this grant didn't trigger anything?
```
[2022-02-22 02:03:54.643342] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
[2022-02-22 02:03:54.643931] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
```

tadscottsmith, 2022-02-22:
Trying to tie the recorded transmissions to the appropriate grants during the same time. Call 754 is a separate TG, so it can be safely ignored.

```
[2022-02-22 02:03:47.484138] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
[2022-02-22 02:03:47.484344] (trace)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
[2022-02-22 02:03:47.485100] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
```
Call 753: SRC 317654 - "9175 Traffic."

```
[2022-02-22 02:03:52.126097] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 312983
[2022-02-22 02:03:52.216005] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 312983
```
Call 753: SRC 312983 - "9175."

```
[2022-02-22 02:03:54.643342] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
[2022-02-22 02:03:54.643931] (debug)   tbsk02   Moto Patch Grant        Channel ID:   269       Freq: 8.526875e+08      sg   11518      TDMA -1 sa 317654
```
MISSED - But, I assume something like "Out with plate XXX-XXX at Century /County E".

```
[2022-02-22 02:04:04.919317] (debug)   tbsk02   Moto Patch Grant        Channel ID:   209       Freq: 8.523125e+08      sg   11518      TDMA -1 sa 312983
[2022-02-22 02:04:04.919457] (trace)   tbsk02   Moto Patch Grant        Channel ID:   209       Freq: 8.523125e+08      sg   11518      TDMA -1 sa 312983
[2022-02-22 02:04:05.024255] (debug)   tbsk02   Moto Patch Grant        Channel ID:   209       Freq: 8.523125e+08      sg   11518      TDMA -1 sa 312983
```
Call 755: SRC 312983 "Copy. Traffic. Century / County Road E , Echo..."

robotastic, 2022-02-22:
Interesting - Those Patch Grants come after when all of the dropped samples happen. I wonder if the Control Channel processing got delayed behind the Voice Channel. There may also not be any guarantees about when a grant message comes.

tadscottsmith, 2022-02-22:
They actually come after. I edited my previous comment and added the dropped samples. It could be some sort of race condition, though, as it's happening in a small percentage of the calls.

robotastic, 2022-02-23:
Soooo... turns out this has been happening on my local system too. A lot. I just hadn't been checking the logs. I should be able to iterate on this a bit quicker now.

Here the Grant Message comes in, then the Termination, then some voice samples. I think this can happen when there is less "processing delay" on the Trunk Channel vs the Voice Channel. So the GRANT message resets the flag to start a new call, but the Termination flag then clears. I will have to come up with something a little more nuanced.
````
Feb 22 21:21:03 radio auto-restart-wmata.sh[241845]: [2022-02-22 21:21:03.652016] (error)   [wmata]        2        TTG:      34480        Freq: 496462500 Hz         GRANT Message
Feb 22 21:21:03 radio auto-restart-wmata.sh[241845]: [2022-02-22 21:21:03.769577] (info)   TEMINATOR!!
Feb 22 21:21:03 radio auto-restart-wmata.sh[241845]: [2022-02-22 21:21:03.769705] (info)   [wmata]        2C        TG:      34480        Freq: 496462500 Hz        record_more_transmissions is false, set>
Feb 22 21:21:04 radio auto-restart-wmata.sh[241845]: [2022-02-22 21:21:04.126904] (error)   [wmata]        2C        TG:      34480        Freq: 496462500 Hz        Dropping samples - Recorder state is: >
````

robotastic, 2022-02-25:
OK - @tadscottsmith I think I have cleaned up quite a bit. I put in place a new approach that essentially has 2 timers. The first tracks when the last grant/update message and only allows new transmissions to be started within 1 sec of receiving one. The second is for the overall call and collects multiple transmissions if the same TG / Freq is used within 5 sec. If you have a chance, can you try out the `no-overlap` branch again?

tadscottsmith, 2022-02-25:
I just updated, so I'll give it a shot. Sounds like a good approach!

tadscottsmith, 2022-03-01:
This seems to be working really well for me. While I haven't had a ton of time to dedicate to listening, I haven't heard any noticeable missed transmissions. I also don't have any of the tell-tale chain of "Dropping samples - Recorder state is: stopped" lines being recorded in the logs.

Nokoa, 2022-03-04:
Been using this new version for a couple days, and it also appears to be working really well so far. Have not noticed any duplications where I previously had them all the time.

tadscottsmith, 2022-03-21:
@robotastic So, this version works great, but it definitely doesn't work for Unit-to-Unit calls. Unit-to-Unit calls seem to stay on the same frequency, but use the destination SUID as the TGID, so it essentially "flips" for each transmission. I'm not sure these have ever worked on 4.x, honestly, but I figured I could start digging in. I'm curious if you have any ideas on how you want to handle the TGIDs though? I've thought of a few different methods, but would love your input.

1. Always use the smaller or larger SUID as the TGID in trunk-recorder.
2. Combine the two SUID in an specified order as the TGID, i.e. SUID 111111 and SUID 222222 make a TGID of 111111222222.
3. Some function to uniquely create a TGID from the pair of SUIDs. Something like a cantor pairing function?

I've done some work to break the UUV Grants and Updates out from the group grants and updates, but I'd like to know if you have a preference on the TGID before I get too much further.

robotastic, 2022-03-22:
Great point! yea, it would not be happy with unit-to-unit calls. I don't think I have any of these one my the systems I monitor. 
With the way it currently is, does each Transmission get turned into its own unique call? If so, that may not be the worst outcome. That way you would be preserving the source-id and the target-id for each transmission. 

Currently the TG Num is stored as a Long. I like the idea of combing the 2 src IDs but the problem would be if it can fit in a Long. It is 32-bit, so I think it could. It would cause some serious UI problems though. Option 1 is probably the safest.

I am going to be working on merging this branch in - If you are able to add in better UUV grant/updates, that would be awesome!

Nokoa, 2022-04-23:
Thank you for fixing this issue! It has been working quite well. I did however notice that quite often certain recordings are cut short, seconds too early. I have several computers recordings some of the same transmissions on different sites.  I have noticed that on one site I would have the full recording for the transmission, and on the other I have that same recording but its cut way short. I was wondering if this is perhaps an issue with my decoding quality, which is mostly 40/sec for the CC so I am not sure. Or maybe an issue with how recordings are being stopped now?

tadscottsmith, 2022-04-29:
You're onto something here. I thought things were a little quiet because of the crap weather, but I just turned on the scanner and I'm definitely missing a fairly significant number of calls/transmissions. I think things were working fine on the #no-overlap branch, so I wonder if something got missed during the merge.

tadscottsmith, 2022-04-29:
Here's an example. Note that this call appears to be seen on two systems simultaneously (not sure if that comes into play yet).
There's a grant for 11558 at 10:09:54 on  852.687500 MHz.
```
[2022-04-28 22:09:54.574142] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
```

Then at least one more grant and a bunch of updates later the call finally starts recording at 10:01:01 (almost 7 seconds into the conversation).
```
[2022-04-28 22:10:01.571168] (trace)   tsbk00   Chan Grant      Channel ID:   209       Freq: 852.312500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:10:01.571226] (info)   [ramsey]  210C    TG:            SPPD CH 1  W (     11558)        Freq: 852.312500 MHz    Starting P25 Recorder Num [1]   TDMA: false     Slot: 0 QPSK: true
[2022-04-28 22:10:01.571326] (trace)   [ramsey] 210C    TG:      11558  Freq: 852.312500 MHz    Starting wavfile sink SRC ID: 321150
```

Here's the full output for 11558. 
```
[2022-04-28 22:09:54.352757] (debug)   tsbk00   Chan Grant      Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.411492] (debug)   tsbk00   Chan Grant      Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.467444] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:54.467512] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:54.574142] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.589235] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:54.589314] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:54.677608] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.678561] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.007978] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.008917] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.008982] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:55.328784] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.418559] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.418643] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:55.450107] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.560597] (trace)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   12030      TDMA -1tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.561543] (trace)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   12030      TDMA -1tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.601415] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.601496] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:55.671353] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.671411] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1tsbk02   Grant Update    Channel ID:  1429       Freq: 859.937500 MHz    ga    2522      TDMA -1
[2022-04-28 22:09:55.775356] (trace)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   12030      TDMA -1tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:56.072044] (trace)   tsbk02   Grant Update    Channel ID:   841       Freq: 856.262500 MHz    ga   11558      TDMA -1
[... cut ...]
[2022-04-28 22:10:07.787014] (trace)   tsbk02   Grant Update    Channel ID:   209       Freq: 852.312500 MHz    ga   11558      TDMA -1
[2022-04-28 22:10:11.011531] (trace)   [ramsey] 210C    TG:            SPPD CH 1  W (     11558)        Freq: 852.312500 MHz     Removing call that has been inactive for more than 3 Sec  Rec last write: 198 State: 4
[2022-04-28 22:10:11.011656] (trace)   [ramsey] 210C    TG:            SPPD CH 1  W (     11558)        Freq: 852.312500 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 10
```

tadscottsmith, 2022-04-29:
It had also been over a minute since Freq: 852.687500 MHz had been used for another call. 
```
[2022-04-28 22:08:09.820891] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga    3580      TDMA -1
[2022-04-28 22:08:10.041577] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga    3580      TDMA -1
[2022-04-28 22:08:10.374498] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga    3580      TDMA -1
[2022-04-28 22:09:54.574142] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.677608] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.678561] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
[2022-04-28 22:09:55.007978] (trace)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1
```
And just over 2 minutes since the previous call for talk group 11558 concluded:
```
[2022-04-28 22:07:36.005452] (trace)   [ramsey] 122C    TG:            SPPD CH 1  W (     11558)        Freq: 853.237500 MHz     Removing call that has been inactive for more than 3 Sec  Rec last write: 1 State: 4
[2022-04-28 22:07:36.005691] (trace)   [ramsey] 122C    TG:            SPPD CH 1  W (     11558)        Freq: 853.237500 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 6
[2022-04-28 22:07:36.006559] (info)   [ramsey]  122C    TG:      11558  Freq: 853.237500 MHz    - Transmission src: 312980 pos: 0 length: 0.9
[2022-04-28 22:07:36.236353] (info)   [ramsey]  122C    TG:      11558  Freq: 8.532375e+08      Rdio Scanner Upload Success - file size: 14444
[2022-04-28 22:07:40.847754] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320570       Value: 0
[2022-04-28 22:07:40.859748] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320570       Value: 0
[2022-04-28 22:07:43.083105] (debug)   tsbk2f   Unit Group Affiliation  Source ID:  321232      Group Address: 11558    Anouncement Goup: 0
[2022-04-28 22:07:43.187747] (debug)   tsbk2f   Unit Group Affiliation  Source ID:  321232      Group Address: 11558    Anouncement Goup: 0
[2022-04-28 22:07:45.810243] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320570       Value: 0
[2022-04-28 22:07:45.922127] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320570       Value: 0
[2022-04-28 22:08:19.115582] (trace)   tsbk39 secondary cc: rfid 2 stid 3 ch1 441(853.763) ch2 433(853.712)
[2022-04-28 22:09:36.876159] (debug)   tsbk2f   Unit Group Affiliation  Source ID:  321167      Group Address: 11558    Anouncement Goup: 0
[2022-04-28 22:09:36.992110] (debug)   tsbk2f   Unit Group Affiliation  Source ID:  321167      Group Address: 11558    Anouncement Goup: 0
[2022-04-28 22:09:51.963700] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320063       Value: 0
[2022-04-28 22:09:52.021837] (debug)   tsbk2b   Location Registration Response  ga 11558        sa 320063       Value: 0
[2022-04-28 22:09:54.574142] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
[2022-04-28 22:09:54.677608] (debug)   tsbk00   Chan Grant      Channel ID:   269       Freq: 852.687500 MHz    ga   11558      TDMA -1 sa 321150       Encrypt 0       Bandwidth: 12.5
```

tadscottsmith, 2022-04-29:
For some reason, on some calls, it is never starting a transmission. I know there were transmissions on 11562 during the period of this call.
```
[2022-04-29 14:47:00.005856] (info)   [ramsey]  194C TG:            SPPD CH 3  E (     11562) Freq: 851.287500 MHz Elapsed: 37 State: inactive
[2022-04-29 14:47:01.009302] (trace)   [ramsey] 194C    TG:            SPPD CH 3  E (     11562)        Freq: 851.287500 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 38
[2022-04-29 14:47:01.010470] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 312982 pos: 0 length: 6.12
[2022-04-29 14:47:01.010671] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 321743 pos: 6.12 length: 2.52
[2022-04-29 14:47:01.010793] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 320659 pos: 8.64 length: 1.08
[2022-04-29 14:47:01.010911] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 312982 pos: 9.72 length: 12.42
[2022-04-29 14:47:01.011030] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 321743 pos: 22.14 length: 2.7
[2022-04-29 14:47:01.011254] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 320659 pos: 24.84 length: 1.44
[2022-04-29 14:47:01.011397] (info)   [ramsey]  194C    TG:      11562  Freq: 851.287500 MHz    - Transmission src: 312982 pos: 26.28 length: 1.44
[2022-04-29 14:47:01.345830] (info)   [ramsey]  194C    TG:      11562  Freq: 8.512875e+08      Rdio Scanner Upload Success - file size: 443564
[2022-04-29 14:47:47.474517] (error)   tsbk00   Chan Grant      Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1 sa 320884       Encrypt 0       Bandwidth: 12.5
[2022-04-29 14:47:47.474773] (error)   Starting new call - TG: 11562 sys num: 0 freq: 8.51588e+08 TDMA Slot0 TDMA: false
[2022-04-29 14:47:47.475771] (info)   [ramsey]  272C    TG:            SPPD CH 3  E (     11562)        Freq: 851.587500 MHz    Recorder found.
[2022-04-29 14:47:47.475950] (info)   [ramsey]  272C    TG:            SPPD CH 3  E (     11562)        Freq: 851.587500 MHz    Starting P25 Recorder Num [3]   TDMA: false     Slot: 0 QPSK: true
[2022-04-29 14:47:47.487275] (error)   tsbk00   Chan Grant      Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1 sa 320884       Encrypt 0       Bandwidth: 12.5
[2022-04-29 14:47:47.487474] (error)   [ramsey] 272C    TG:            SPPD CH 3  E (     11562)        Freq: 851.587500 MHz     GRANT Message for existing Call Call State: 1
[2022-04-29 14:47:47.597384] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:47.597445] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1tsbk02   Grant Update    Channel ID:    45       Freq: 851.287500 MHz    ga     802      TDMA -1
[2022-04-29 14:47:47.709919] (error)   tsbk02   Grant Update    Channel ID:    89       Freq: 851.562500 MHz    ga   11522      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:47.819571] (error)   tsbk02   Grant Update    Channel ID:    89       Freq: 851.562500 MHz    ga   11522      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:48.135677] (error)   tsbk02   Grant Update    Channel ID:    89       Freq: 851.562500 MHz    ga   11522      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:48.479941] (error)   tsbk02   Grant Update    Channel ID:    89       Freq: 851.562500 MHz    ga   11522      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:48.784368] (error)   tsbk02   Grant Update    Channel ID:    89       Freq: 851.562500 MHz    ga   11522      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:49.137255] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:49.137319] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1tsbk02   Grant Update    Channel ID:    45       Freq: 851.287500 MHz    ga     802      TDMA -1
[2022-04-29 14:47:49.350535] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:49.350788] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1tsbk02   Grant Update    Channel ID:    45       Freq: 851.287500 MHz    ga     802      TDMA -1
[2022-04-29 14:47:49.558081] (error)   tsbk02   Grant Update    Channel ID:   269       Freq: 852.687500 MHz    ga   12190      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:50.006526] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:47:50.006596] (error)   tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1tsbk02   Grant Update    Channel ID:   337       Freq: 853.112500 MHz    ga   1026 [... cut ...]
[2022-04-29 14:47:58.189682] (error)   tsbk02   Grant Update    Channel ID:   129       Freq: 851.812500 MHz    ga     638      TDMA -1tsbk02   Grant Update    Channel ID:    93       Freq: 851.587500 MHz    ga   11562      TDMA -1
[2022-04-29 14:48:02.003580] (trace)   [ramsey] 272C    TG:            SPPD CH 3  E (     11562)        Freq: 851.587500 MHz    Concluding Recorded Call - Last Update: 4s      Call Elapsed: 15
[2022-04-29 14:48:02.004706] (error)   [ramsey] 272C    TG:      11562   Freq: 8.51588e+08      No Transmission were recorded!
```

rosecitytransit, 2023-04-02:
Regarding unit-unit calls, what about simply always using the first-seen order, e.g by checking to see if there's an active call with a talkgroup matching the the incoming source as in https://github.com/robotastic/trunk-recorder/pull/470/commits/0eb05cdeede051881f41adecbf063606734206fe and if so swapping the IDs? The only downside is that the ID of who's talking in a transmission wouldn't be preserved.

I record a system with a lot of unit-unit calls and would offer it up for debugging, but they don't show up correctly on the control channel--there's never any update messages for them. With my modifications to address that, 4.5.0 does seem to be working fine for them.

closed 2024-12-14 by robotastic