TrunkRecorder/trunk-recorder #617: (error)   stop_recording() - stopping wavfile sink but recorder state is: 1 after Ubuntu 20.04 upgrade

jcummingsg2, 2022-01-21:
Upgraded from Ubuntu 18.04 to 20.04 and have had a configured SmartNet system for over a year running with no issues, now after upgrade and new compile of latest Trunk-Recorder seeing this error frequently and recorders getting tied up until exhaustion and high CPU.

[2022-01-20 16:50:10.017608] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1

Have tried rebuilding prior version of trunk-recorder and seeing the same behavior so may be related to ubuntu upgrade.  Not sure where to go yet on next step troubleshooting but wanted to open issue to start.

robotastic, 2022-01-21:
Thanks for reporting - It is probably not related to Ubuntu and is more likely due to something I screwed up. 
That error may not actually be a real error. I think with Analog recordings, it might show up every time a recording stops. I have to clean that up.
A couple thoughts:
- it might be worth double checking the squelch value. it is possible it is getting stuck open and keep recording.
- Is the whole system analog?
- Maybe try a short callTimeout

jcummingsg2, 2022-01-21:
Thanks for the reply, yes I do think its not related to Ubuntu upgrade but rather the version bump on trunk-recorder.  Interestingly, I was seeing this on 4.3.0 but also backed down to prior running code 4.2.0 and still seeing the error in the logs but it  does not seem to be tying up all the analog recorders anymore and its been running for about an hour an a half now and looking better.  I did not check squelch, that's a good thought and had not looked into that yet, will do this evening.

ZL1LAC, 2022-02-08:
Can confirm I'm also getting the same issue with the stopping wavfile sink but recorder state.

I'm running this in Docker using a Pi3B+.

Here is my config file.
```
{
	"ver": 2,
"sources": [{
	"center": 141400000,
	"rate": 2048000,
	"gain": 16.6,
	"analogRecorders": 1,
	"driver": "osmosdr",
	"device": "rtl=0"
}],
"systems": [
{
	"type": "conventional",
	"defaultMode": "analog",
	"channels": [141400000],
	"shortName": "EAS",
	"minDuration": 3,
	"callLog": false,
	"audioArchive": true,
	"squelch": -50
	}],
	"captureDir": "/home/pi/recorder/radio/"
}
```

robotastic, 2024-12-14:
This is probably resolved in current versions, but please reopen if it is not

closed 2024-12-14 by robotastic