TrunkRecorder/trunk-recorder #622: Background Audio Playback in iOS/iPadOS 15

bhall7, 2022-01-27:
With iOS 15 on iPhone or iPad, the Safari web browser stops playing audio in Trunk Recorder as soon as the browser no longer has focus or the phone is locked. You used to be able to tap Play from Control Center to continue playing Safari browser audio and continue listening while the phone is locked or when switching to a different app. But, iOS 15 appears to have changed this behavior so that you must keep Safari open and in focus for browser audio in Trunk Recorder to continue playing.

This affects other web applications--not just Trunk Recorder--and streaming audio services as well. For example, I was able to replicate this issue on soundcloud.com and bandcamp.com, when playing back a playlist, the audio stops after one track completes and does not advance to the next track in the playlist. See my more [detailed post on the Apple Discussions Forum](https://discussions.apple.com/thread/253331448). I reported this to Apple (escalated through several senior engineering teams) and does not seem to care.

I discovered the Nora Sector app recently (https://app.norasector.com/), and learned to my delight that it **is** able to play continuous audio in the background in Safari in iOS when Safari is not in focus or when using another app.

I wonder if the same behavior could be replicated in Trunk Recorder. I'm not sure what exactly is being done in the application to keep the audio playing, but it looks like it can be done. Might be worth a look.

patrickfish, 2022-01-27:
Hey Brian,

Do you know if there is a radar for this issue?

bhall7, 2022-01-27:
@patrickfish, I'm sorry, I don't know.

bhall7, 2022-01-27:
I do know that NoraSector is using WebRTC, but, I don't know if that's the reason why iOS/iPad OS will continue to play audio in the background when the browser isn't in focus. I don't know enough to understand why NoraSector audio stays active in the background, but TR does not.

Currently, regardless of the browser I use on iOS/iPad OS 15 (the latest versions of Chrome, Firefox, Safari), I cannot get TR audio to survive a browser minimization. So, it must be more of a device/OS level change rather than a browser-specific capability. I suspect that Apple likely made this change to prevent websites from running in the background to preserve both user privacy and device battery life.

If we could figure out what NoraSector is doing to keep the audio going in the background, that would be useful to know, and whether or not TR could be updated for the latest version of iOS/iPad OS.

bhall7, 2022-01-27:
Perhaps it is an issue that should be addressed with [Trunk Player](https://github.com/ScanOC/trunk-player/issues)

Thoughts?

bctrainers, 2022-01-27:
NoraSector is using WebRTC, and utilizes a stun server, which seems to be contacting an Icecast server(?). 

Doesn't utilize jPlayer in the slightest like TrunkPlayer does.

bctrainers, 2022-01-27:
All of those details are on this repo: https://github.com/norasector

Dygear, 2022-01-27:
Nice catch! Didn’t realize they were open source! I’ll have a look once I get home from work.

robotastic, 2022-01-29:
If anyone figures out the trick, that would be great! I am not sure how well it will work because I think they maybe sending out the audio differently. OpenMHz is using websockets to send out new calls, so the webpage needs to be active to pull in new calls. NoraSector might be using WebRTC, which is more like a VoIP call. Also NoraSector looks like a great project!

robotastic, 2022-02-26:
Closing for now. If someone figures it out, please submit a PR on the OpenMHz repo: https://github.com/openmhz/trunk-server

closed 2022-02-26 by robotastic