TrunkRecorder/trunk-recorder #623: Assertion (internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0

aaknitt, 2022-01-27:
I've been seeing this on my Pi quite a bit recently.  It may be specific to my setup since no once else seems to be reporting it, so low priority, just putting it here for awareness in case anyone else sees it.  

First start of trunk-recorder after reboot works fine.  On subsequent attempts to start trunk-recorder I'll often get this error. If I keep attempting to restart and eventually it fires up and runs without issue.  This doesn't happen immediately at startup, but after it's started to run for a second or two.

```
[2022-01-26 22:29:06.128513] (info)   [oasiswauk]       Started with Control Channel: 853.712500 MHz
[2022-01-26 22:29:06.130493] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 1400000
[2022-01-26 22:29:06.151215] (info)      P25 Trunking ARB - Initial Rate: 1400000 Resampled Rate: 25000 Initial Decimation: 14 System Rate: 24000 ARB Rate: 0.96
Allocating 15 zero-copy buffers
Allocating 15 zero-copy buffers
[2022-01-26 22:29:06.796761] (info)   [oasiswauk]       0C      TG:      58918 (              WKLEDISP5)   Freq: 851.212500 MHz     Starting P25 Recorder Num [0]   TDMA: false     Slot: 0 QPSK: true
[2022-01-26 22:29:06.797736] (info)   call_start called in freqsend plugin for TGID 58918 on system oasiswauk from unit 4294967295 on freq 8.51212e+08
[2022-01-26 22:29:06.798050] (info)   system.TGIDs.count(58918) is 1
[2022-01-26 22:29:06.798650] (info)   adding 8.51212e+08 to current_TGID_freqs[58918]
[2022-01-26 22:29:06.799328] (info)   [oasiswauk]       Decoding System ID 692 WACN: BEE00 NAC: 691
trunk-recorder: /home/pi/trunk-recorder/lib/lfsr/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = int; int Size = 1936; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
```

GDB core dump output:
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
--Type <RET> for more, q to quit, c to continue without paging--
Core was generated by `./trunk-recorder --config=config_waukesha.json'.
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0xb297dd00 (LWP 7463))]
(gdb) bt full
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
        set = {__val = {0, 0, 3066380324, 97633816, 3066380336, 4, 3046711012, 94, 97634696, 20, 94,
            2780855296, 3045405520, 94, 97634364, 97634696, 3197753744, 2780855296, 3045661992,
            3197753624, 1643257553, 2459606, 2459607, 854277696, 49478849, 2206310400, 49478850, 0,
            97634364, 4294967295, 97634224, 3197753744}}
        pid = <optimized out>
        tid = <optimized out>
        ret = <optimized out>
#1  0xb55de804 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0xb297dd00, sa_sigaction = 0xb297dd00}, sa_mask = {
            __val = {3043988740, 0, 1871936, 1871728, 3069300736, 3197753964, 3043117400, 7302446,
              3044013232, 3069303700, 0, 3043988736, 1196582912, 0, 3044101224, 2996296960, 3043988740, 0,
              1871936, 1871728, 3069300736, 3197754116, 3044098048, 5, 3044098048, 1871872, 1196582912,
              3044098048, 3067498496, 1871872, 3043988740, 109}}, sa_flags = 1871936,
          sa_restorer = 0x1c8f70}
        sigs = {__val = {32, 0 <repeats 31 times>}}
#2  0xb55ee51c in __assert_fail_base (fmt=0x1 <error: Cannot access memory at address 0x1>,
    assertion=0x1c9040 "(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.htm"..., assertion@entry=0x0, file=0x0,
    file@entry=0x1 <error: Cannot access memory at address 0x1>, line=109, line@entry=1871872,
    function=0x1c8f70 "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = int; int Size = 1936; int MatrixOrArrayOptions = 0]", function@entry=0xb297dd00 "\001") at assert.c:92
        str = 0x5d8b750 ""
        total = 4096
#3  0xb55ee5d4 in __GI___assert_fail (assertion=0x0,
    file=0x1 <error: Cannot access memory at address 0x1>, line=1871872, function=0xb297dd00 "\001")
    at assert.c:101
--Type <RET> for more, q to quit, c to continue without paging--
No locals.
#4  0x0017deec in Eigen::internal::plain_array<int, 1936, 0, 16>::plain_array (this=0x5d89908)
    at /home/pi/trunk-recorder/lib/lfsr/Eigen/src/Core/DenseStorage.h:109
        __PRETTY_FUNCTION__ = {<optimized out> <repeats 141 times>}
#5  0x0017d62c in Eigen::DenseStorage<int, 1936, 44, 44, 0>::DenseStorage (this=0x5d89908)
    at /home/pi/trunk-recorder/lib/lfsr/Eigen/src/Core/DenseStorage.h:187
No locals.
#6  0x0017cc08 in Eigen::PlainObjectBase<Eigen::Matrix<int, 44, 44, 0, 44, 44> >::PlainObjectBase (
    this=0x5d89908) at /home/pi/trunk-recorder/lib/lfsr/Eigen/src/Core/PlainObjectBase.h:484
No locals.
#7  0x0017c390 in Eigen::Matrix<int, 44, 44, 0, 44, 44>::Matrix (this=0x5d89908)
    at /home/pi/trunk-recorder/lib/lfsr/Eigen/src/Core/Matrix.h:259
No locals.
#8  0x0017adec in p25p2_lfsr::p25p2_lfsr (this=0x5d898f8, nac=1681, sysid=1682, wacn=781824)
    at /home/pi/trunk-recorder/lib/lfsr/lfsr.cxx:19
        xorbits = <optimized out>
        i = <optimized out>
        j = <optimized out>
        i = <optimized out>
#9  0x0012e0d8 in System::update_status (this=0x5b6ac48, message=...)
    at /home/pi/trunk-recorder/trunk-recorder/systems/system.cc:117
No locals.
#10 0x00087f5c in current_system_status (message=..., sys=0x5b6ac48)
    at /home/pi/trunk-recorder/trunk-recorder/main.cc:931
No locals.
#11 0x00088548 in handle_message (messages=std::vector of length 1, capacity 1 = {...}, sys=0x5b6ac48)
    at /home/pi/trunk-recorder/trunk-recorder/main.cc:1020
        message = {message_type = STATUS, meta = "", freq = 853712500, talkgroup = 0, encrypted = false,
          emergency = false, duplex = false, mode = false, priority = 0, tdma_slot = 0,
--Type <RET> for more, q to quit, c to continue without paging--
          phase2_tdma = false, source = -1, sys_num = 0, sys_id = 1682, nac = 1681, wacn = 781824,
          moto_patch_data = {sg = 58920, ga1 = 58921, ga2 = 58921, ga3 = 58921}}
        it =
    {message_type = STATUS, meta = "", freq = 853712500, talkgroup = 0, encrypted = false, emergency = false, duplex = false, mode = false, priority = 0, tdma_slot = 0, phase2_tdma = false, source = -1, sys_num = 0, sys_id = 1682, nac = 1681, wacn = 781824, moto_patch_data = {sg = 58920, ga1 = 58921, ga2 = 58921, ga3 = 58921}}
#12 0x0008a0dc in monitor_messages () at /home/pi/trunk-recorder/trunk-recorder/main.cc:1197
        timeDiff = 1
        statusTimeDiff = 1
        msg = {px = 0xa5c07c88, pn = {pi_ = 0xa5c07e20}}
        sys_num = 0
        sys = 0x5b6ac48
        lastStatusTime = 1643257552
        lastMsgCountTime = 1643257552
        management_timestamp = 1643257553
        current_time = 1643257553
        trunk_messages = std::vector of length 1, capacity 1 = {{message_type = STATUS, meta = "",
            freq = 853712500, talkgroup = 0, encrypted = false, emergency = false, duplex = false,
            mode = false, priority = 0, tdma_slot = 0, phase2_tdma = false, source = -1, sys_num = 0,
            sys_id = 1682, nac = 1681, wacn = 781824, moto_patch_data = {sg = 58920, ga1 = 58921,
              ga2 = 58921, ga3 = 58921}}}
#13 0x0008ceb8 in main (argc=2, argv=0xbe99e654) at /home/pi/trunk-recorder/trunk-recorder/main.cc:1440
        desc = {static m_default_line_length = 80, m_caption = "Options", m_line_length = 80,
          m_min_description_length = 40, m_options = std::vector of length 2, capacity 2 = {{
              px = 0x5b687f8, pn = {pi_ = 0x5b68bf0}}, {px = 0x5b68900, pn = {pi_ = 0x5b68c20}}},
          belong_to_group = std::vector<bool> of length 2, capacity 32 = {false, false},
          groups = std::vector of length 0, capacity 0}
        vm = <incomplete type>
--Type <RET> for more, q to quit, c to continue without paging--
        config_file = "config_waukesha.json"
        uri = "ws://localhost:3005"
```

aaknitt, 2022-01-29:
Have not seen this error since I switched to Release build...only saw it when doing Debug builds.

robotastic, 2022-01-31:
That is weird! The Eigen stuff is math heavy. I wonder if there is something in there that is not Pi compatible.

Dewey3, 2022-02-01:
Since I'm running on a couple of 8GB RPis, I just checked my logs and not seeing this; however, my log level is set to error and I notice this is coming up as info.  I've changed the log level on both RPis to info and will see if this pops up in the next day or two.

Dewey3, 2022-02-01:
I have not been able to produce this problem with either of my RPis.  In my case, Trunk-Recorder starts from a cron job 1 minute after my RPis boot, so I will normally reboot anytime I want to restart Trunk-Recorder (they both run unattended and headless).  In trying to see if I could force this issue, I stopped and manually restarted Trunk-Recorder a few times to see if I would get the error, but I never did.

robotastic, 2022-02-22:
I am going to close this for now - happy to reopen if someone comes across this or there ends up being a great way to reproduce it.

closed 2022-02-22 by robotastic