TrunkRecorder/trunk-recorder #629: Build fails on Fedora 35:  /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/recorders/p25_recorder.h:66:41: error: conflicting declaration 'typedef class std::shared_ptr<p25_recorder> p25_recorder_sptr'

mldulaney, 2022-02-02:
trunk-recorder 4.3.0

[  4%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o
cd /builddir/build/BUILD/trunk-recorder-4.3.0/redhat-linux-build/lib/op25_repeater/lib && /usr/bin/g++ -DBOOST_ALL_DYN_LINK -DEZPWD_NO_EXCEPTS -DFFTW3F_THREADS -DGNURADIO_VERSION=198914 -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -Dgnuradio_op25_repeater_EXPORTS -I/builddir/build/BUILD/trunk-recorder-4.3.0/redhat-linux-build/lib -I/builddir/build/BUILD/trunk-recorder-4.3.0/redhat-linux-build/include -I/builddir/build/BUILD/trunk-recorder-4.3.0/lib/op25_repeater/lib -I/builddir/build/BUILD/trunk-recorder-4.3.0/lib/op25_repeater/include -I/builddir/build/BUILD/trunk-recorder-4.3.0/lib -isystem /usr/lib64/python3.10/site-packages/numpy/core/include -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -pthread -Wno-narrowing -fvisibility=hidden -fPIC -DNDEBUG -fPIC   -fvisibility=hidden -w -std=gnu++11 -MD -MT lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o -MF CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o.d -o CMakeFiles/gnuradio-op25_repeater.dir/dmr_bs_tx_bb_impl.cc.o -c /builddir/build/BUILD/trunk-recorder-4.3.0/lib/op25_repeater/lib/dmr_bs_tx_bb_impl.cc
In file included from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/source.h:13,
                 from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.cc:6:
/builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/recorders/p25_recorder.h:66:41: error: conflicting declaration 'typedef class std::shared_ptr<p25_recorder> p25_recorder_sptr'
   66 |   typedef std::shared_ptr<p25_recorder> p25_recorder_sptr;
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.h:19,
                 from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.cc:1:
/builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/systems/system.h:38:43: note: previous declaration as 'typedef class boost::shared_ptr<p25_recorder> p25_recorder_sptr'
   38 |   typedef boost::shared_ptr<p25_recorder> p25_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
In file included from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/source.h:14,
                 from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.cc:6:
/builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/recorders/dmr_recorder.h:88:41: error: conflicting declaration 'typedef class std::shared_ptr<dmr_recorder> dmr_recorder_sptr'
   88 |   typedef std::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                         ^~~~~~~~~~~~~~~~~
In file included from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.h:19,
                 from /builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/call.cc:1:
/builddir/build/BUILD/trunk-recorder-4.3.0/trunk-recorder/systems/system.h:39:43: note: previous declaration as 'typedef class boost::shared_ptr<dmr_recorder> dmr_recorder_sptr'
   39 |   typedef boost::shared_ptr<dmr_recorder> dmr_recorder_sptr;
      |                                           ^~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:79: CMakeFiles/trunk_recorder_library.dir/trunk-recorder/recorders/recorder.cc.o] Error 1

mldulaney, 2022-02-14:
Ahoy!

psycholance, 2022-02-16:
this was probably fixed in https://github.com/robotastic/trunk-recorder/commit/477252ab9361d07ca2663a1c026213022f2b3439 , there's just no release yet

robotastic, 2022-02-26:
This release probably fixed things. Let me know if it doesn't. I am not running Fedora on any systems, so I don't test against it often.... I have to get a docker container running to do that and I always forget the steps.
https://github.com/robotastic/trunk-recorder/releases/tag/v4.3.1

closed 2022-02-26 by robotastic