TrunkRecorder/trunk-recorder #634: Removing call with stuck recorder

Dewey3, 2022-02-08:
This could be on my end, but I don't recall seeing it before the recent "major simplification using Call data" update.  I am getting the errors every 2-3 minutes:

`[2022-02-08 05:52:47.006763] (error)   [2A8]	[0;34m8798C[0m  Rec Num: 7	TG:         55 ([35m                 PGPD C[0m)	Freq: 771.356250 MHz	[36m Removing call with stuck recorder [0m`

`[2022-02-08 05:52:47.007066] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1`

`[2022-02-08 05:53:19.007107] (error)   [2A8]	[0;34m8799C[0m  Rec Num: 0	TG:         55 ([35m                 PGPD C[0m)	Freq: 770.356250 MHz	[36m Removing call with stuck recorder [0m`

`[2022-02-08 05:53:19.007478] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1`

`[2022-02-08 05:54:14.011657] (error)   [2A8]	[0;34m8804C[0m  Rec Num: 7	TG:        181 ([35m            PGSO 1 Disp[0m)	Freq: 771.356250 MHz	[36m Removing call with stuck recorder [0m`

`[2022-02-08 05:54:14.011975] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1`

`[2022-02-08 05:54:35.013168] (error)   [2A8]	[0;34m8809C[0m  Rec Num: 0	TG:        181 ([35m            PGSO 1 Disp[0m)	Freq: 770.356250 MHz	[36m Removing call with stuck recorder [0m`

`[2022-02-08 05:54:35.016667] (error)   stop_recording() - stopping wavfile sink but recorder state is: 1`

robotastic, 2022-02-08:
This message means that a recorder is actively recording, but the Control Channel message for that Call has been updated. This might have been happening before, but this state wasn't captured well.

Could you add `controlWarnRate`: -1 to your config file? It will tell you how many messages are getting decoded. It should be above 30.

robotastic, 2022-02-08:
Is this a conventional system?

Dewey3, 2022-02-08:
P25 Phase II - TDMA (Prince George's Co, MD - 2A8).

Dewey3, 2022-02-08:
Thanks Luke.  Using the `controlWarnRate` switch, so far all of my decode rate messages have been 115 and higher.  On my secondary system which doesn't have the greatest reception to start with, Charles Co, MD - A42E, which is a trunked Type 2 SmartNet system, but giving the same stuck recorder messages, the decode rate messages have been 75 and higher.  Thanks again.

Dewey3, 2022-02-08:
Well, that didn't take long.  Screenshot of an a question mark on an inverted background included in the attached folder:
[https://drive.google.com/drive/folders/1gAUouj7xlJr8RLRx1Ztl6BOhNw4gWfrZ?usp=sharing](url)

taclane, 2022-02-08:
> Well, that didn't take long. Screenshot of an a question mark on an inverted background included in the attached folder:
> https://drive.google.com/drive/folders/1gAUouj7xlJr8RLRx1Ztl6BOhNw4gWfrZ?usp=sharing

That is actually a different problem, specifically with the rdio-scanner plugin.  It's feeding junk into the `talkgroupGroup` field, and that junk is slowly overwriting whatever you may have set in Rdio Scanner. Rdio Scanner wasn't auto-populating this field until somewhat recently, so it went unnoticed until around now that it was causing problems.

I found a way to fix it, and should have a PR soonish.  But for now, just turn off the "Auto Populate" options in radio scanner, and it should stop this from happening further.

Dewey3, 2022-02-08:
Thank you sir!  Given how good Rdio-Scanner works, believe me, this is NOT a problem for me... just something I wanted to bring to attention.

closed 2022-02-08 by Dewey3

robotastic, 2022-02-09:
Great to hear there was a happy ending!