TrunkRecorder/trunk-recorder #636: Can Anyone Help Me Understand What is Happening?

Dewey3, 2022-02-11:
Attached are PuTTY screenshots of two systems within a minute of each other, `df -h` first, then followed by `sudo ncdu /`.  The system on the left, PG Co, MD (2A8) is three or four times **busier** than the system on the right, Charles Co, MD (A42E).  If it makes any difference, I wouldn't think so, the left 2A8 system is a trunked P25 Phase II-TDMA system while the right A42E is a trunked Mot Type II-SmartNet system that does have some analog talkgroups.  Both systems are running from the exact same image that I freshly created on this past Monday (2/7).  Both systems send the recordings to Rdio-Scanner via plugin and do not retain the recordings afterwards.  What I just can't understand and have never noticed before is why the far less busy system on the right uses disk space exponentially faster than the busier system on the left.  I first noticed this yesterday morning when I noticed I was almost out of disk space on the system on the right, so I rebuilt the less busy system from the same 2/7 image and started it over.  Of course, when first started yesterday, the newly restored RPi was smaller than the RPi running the busier system.  However, this morning, I noticed that the less busier system takes almost two times the disk real estate as the busier system.  I even ran `sudo apt` - `autoremove --purge` `autoclean` and `clean` on the system on the right, but that only recovered about 10MB of space.  The Swap for both systems is right at 100MB.  I know I can just throw a larger MicroSD in the RPi that is eating up disk space, but I am trying to understand the "why" since it is less than half as busy as the other RPi, and both systems used to stay undet 50% disk usage.

[image]

Dewey3, 2022-02-11:
Just a follow up post to the earlier question, I noticed that I've lost another 3/4 GB of disk real estate in the last 3 hours.  The one thing I didn't mentioned about this system in this thread is that it does not have the "best" reception.  Looking over the logs, I see a great deal of `... Dropping samples - Recorder state is: stopped` and `... Dropping samples - current_call is null	 Rec State: available	Since close: 1`.  Is there a location other than the my own defined recordings directory I can check to see if it is possible that these samples that are being "dropped" are quickly accumulating and eating up disk space?

robotastic, 2022-02-13:
@Dewey3 - interesting, the dropped samples probably doesn't matter. Those samples are in memory and getting dropped instead of being written to disk. My guess is that it has something to do with how Linux handles deleted files. The files are still being written to disk, they just get deleted after Rdio uploads them.

Looks like `lsof` might help: https://serverfault.com/questions/232525/df-in-linux-not-showing-correct-free-space-after-file-removal/232526

robotastic, 2022-02-13:
> Thanks much Luke! This is very strange, I've now discovered that all I have to do to get all of the space back is to reboot. That explains why I can't find any files that are responsible for the loss of drive space. I considered that maybe it's something Linux is doing... BUT... I'm running two RPi's built from the exact same image. One RPi works as expected while the other RPi reports that the drive is filling at approx. 1/4 GB an hour, yet nothing on the drive reflects that growth and a simple reboot brings everything back to normal. Unless you feel otherwise, I will leave this open for a couple days while I continue to try to figure out what is going on (I'm also considering the ServerFault posts in your link). In a worst case scenario, all I have to do is write a cron job to either monitor "reported" drive usage and reboot when almost full, or just automatically reboot every couple days around 3:30 or 4:00 in the morning when things are quiet. By the way, I did restore an image that was freshly built on 1/29/2022, but that did the same thing, but just at a slower much more unnoticeable rate, so I went back to my 2/7 built image to keep things the same between the two RPis.

The only thing I can think of that might explain the difference is that one of the Pi occasionally crashes and restarts. When that happens all those files would be freed up. Let me know if you figure anything else out!

Dewey3, 2022-02-17:
Thanks Luke!  I know that the RPi that is not going through this is is running fine 24/7, because I have a cron job on a server that checks in with all of my RPis (currently 20) and sends me health reports that include `uptime`.  I still have no idea what is going on, but I do know a simple reboot solves it.  So for now, there is no reason to keep this discussion open since I cannot identify the cause.  Again, thanks for Trunk-Recorder and all of your help!

closed 2022-02-17 by Dewey3