TrunkRecorder/trunk-recorder #638: debian fdkaac missing for arm64

hayden-t, 2022-02-21:
mentioned in the dependencies, you build a arm64 image for docker, will try with out it but otherwise think im stuck

hayden-t, 2022-02-21:
ok, so it built but i get this error 
`[2022-02-21 15:29:33.035543] (error)   Failed to convert call recording, see above error. Make sure you have sox and fdkaac installed.
`

robotastic, 2022-02-22:
Can you provide more context, I am confused? Is it that the ARM64 image on Docker Hub does not have FDKAAC installed?

hayden-t, 2022-02-22:
no, this is not docker related. debian 11 repo does not have arm64 package for fdkaac 
im back to using docker again though as compiling didnt help my problem, just mentioning it

robotastic, 2022-02-26:
Sorry - I am still confused. Is there specific line in a documentation file that needs to be changed? Could you open a PR with the info that needs to be updated? Should I make a note that Debian 11 is not supported for ARM64 platforms?

hayden-t, 2022-02-28:
Hi, it seems there actually isnt any version of debian which has a fdkaac package in the official repository apart from amd64 , so it either has to be sourced from another repo though i had trouble finding on, compiled from source (if possible), or m4a disabling https://packages.debian.org/bullseye/fdkaac

hayden-t, 2022-02-28:
I did find it there https://www.deb-multimedia.org/dists/bullseye/main/binary-arm64/package/fdkaac just have to get around conflict of libfdkaac version between mm and official repo

hayden-t, 2022-02-28:
The following packages have unmet dependencies:
 fdkaac : Depends: libfdk-aac2 (>= 1:2.0.1) but 2.0.1-1 is to be installed

closed 2022-02-28 by hayden-t

daniel-gallagher, 2022-05-18:
Though this issue is closed, I wanted to post the full solution for anyone else who comes across this error. 

> I did find it there https://www.deb-multimedia.org/dists/bullseye/main/binary-arm64/package/fdkaac just have to get around conflict of libfdkaac version between mm and official repo

The correct binary can be located as referenced in the above quote. [DIRECT LINK to download](https://www.deb-multimedia.org/pool/main/f/fdkaac-dmo/fdkaac_1.0.2-dmo1_arm64.deb)

`wget https://www.deb-multimedia.org/pool/main/f/fdkaac-dmo/fdkaac_1.0.2-dmo1_arm64.deb`

You might also need to `sudo apt install libfdk-aac2`, which will present the error as mentioned, even though the correct version is installed.  Force installing the package will resolve the issue and trunk-recorder will operate as expected.

`sudo dpkg --force-all -i fdkaac_1.0.2-dmo1_arm64.deb`