TrunkRecorder/trunk-recorder #645: If an uploader fails, duplicate recordings get sent to the other uploaders

wx-jc, 2022-03-03:
[image]

Screenshot of the description issue from Discord.  If an uploader fails due to a server side problem, duplicate calls get sent to the other uploaders.

robotastic, 2022-03-04:
So, for better or worse this is sort of by design. The larger goal was to retry if an upload fails... and having track plugin, by plugin, would make things sort of complex... so it just retries them all if one fails. It might be possible to make it an option, where you could turn this off.

wx-jc, 2022-03-05:
Yeah I was afraid it may not be a quick fix...but making it an option could be nice!  Perhaps others will have different ideas.

aaknitt, 2022-03-12:
Would it make sense to move the retry functionality into the plugins themselves rather than via the manager?  As the number of plugins grow, having a failure in one impact all of the others is definitely not great.

robotastic, 2022-03-12:
hmm... something like that could work. Right now, the call concluder goes through the list of plugins. If they were all successful, it wraps up the call and removes the files. If not, it keeps the files around and re-runs all the plugins. 

I could change it so that it just runs through the list of plugins once and then deletes the files when it gets to the end. If a single plugin fails, it is on that plugin to retry before handing control back to call concluder so the next plugin could run. 

This seems like it could work.... let me check it out.

robotastic, 2024-12-14:
I believe this has been addressed

closed 2024-12-14 by robotastic