TrunkRecorder/trunk-recorder #657: clean exiting removed

rosecitytransit, 2022-03-28:
In #470, I made it so the program would cleanly exit by closing open calls and writing JSON files for them (see https://github.com/robotastic/trunk-recorder/pull/470/commits/6ad366a5feb6535523ced51e8323430ab13b9821 ) but it seems the feature got removed when you you did this commit: https://github.com/robotastic/trunk-recorder/commit/70698818065b24a494fc5488c82445609eab9ebc

Might want to check to see if some of the other changes I submitted in #470 got inadvertently reverted too.

robotastic, 2022-03-29:
Oh man - that was an ugly commit I made. Let me go see if I can figure out why it is not exiting cleanly. I think the switch was in how I handle calls in 4.0

robotastic, 2022-03-29:
Yep that is weird - This block definitely got dropped somehow when I merged that branch in... and then did some horrible Git stuff. Let me go add it back. I will also do a quick spot check on the other changes.

https://github.com/robotastic/trunk-recorder/pull/470/commits/6ad366a5feb6535523ced51e8323430ab13b9821

robotastic, 2022-03-29:
Ok - Merged that code back in. I did a spot check and it looks like all of the other changes made it in there. I must have done something done and deleted it by accident. I may need to add a 5 second delay before exiting to make sure everything shutdown correctly.

rosecitytransit, 2022-03-29:
Great! Wish I was following along so could have spotted it sooner.  I was thinking today how it might need changes to work with the new call concluder method.  IIRC, I just copied how calls were being closed at the time.

rosecitytransit, 2022-03-29:
Need this re-opened (and apparently I can't do that). I've updated to the current master, 9eaf7ce226ec51dd16ee52859f5eebfdfc218a17, and now I'm getting a Segmentation fault when I exit the program (pressing control+c),

Adding some logging shows that the loop
https://github.com/robotastic/trunk-recorder/blob/9eaf7ce226ec51dd16ee52859f5eebfdfc218a17/trunk-recorder/main.cc#L1172-L1179
appears to be running OK on the 2 open calls at the time, but then runs again for a 3rd time and crashes at https://github.com/robotastic/trunk-recorder/blob/9eaf7ce226ec51dd16ee52859f5eebfdfc218a17/trunk-recorder/call.cc#L105 or just beyond

rosecitytransit, 2022-03-29:
I think I see the problem now: the ```for``` loop is missing an increment for the iterator.

robotastic, 2022-03-30:
Hmm... let me go dig into this. The `for` loop is actually incrementing via the Delete() call. It gives you the next element in the vector.

robotastic, 2022-03-30:
ok - I fixed it. I messed up on the state change and was forcing monitored calls to be handled like calls that are recorded. Should be good now.

rosecitytransit, 2022-03-30:
Looks like that did it. I figured it could be something like that. Looks like you'll have to close this now.

closed 2022-04-03 by robotastic